jneilliii/OctoPrint-BedLevelVisualizer #173: Correct data collection, but no visualization

sstickel, 2019-12-30:
Hello everybody,
i recently installed this plugin as it is much more flexible than my own console based python script.
Retrieving data seems to work as it should. (CATMULL is active)
[image]

But there is no visualization being plotted. It seems to do "something" as the scale is correct with 3mm in the middle, but no graph is displayed. Is there a way to debug this? Or is this a known issue? 
[image]

System is Marlin 2 based, octoprint running on a Pi 3.

The gcode the plugin is executing is:
`M155 S0`
`@BEDLEVELVISUALIZER`
`M420 V`
`M155 S3`

If you need more infos just tell me.

jneilliii, 2019-12-30:
It's because the points measured are outside the default range. Enable the option for use relative z offsets.

sstickel, 2019-12-30:
aha, that worked.
Why are these values out of range? To be honest I never understood till today what the absolute values here mean.

My Z-Offset is fixed at -1,95mm, this is what i get then. Bed distance is quite accurate.

jneilliii, 2019-12-30:
The surface graph has a fixed +/-2mm on the z because of the way I programmed it. I'm not quite sure why the report from your firmware comes out that way, maybe it includes the offset in the reading.

closed 2019-12-30 by jneilliii