jneilliii/OctoPrint-BedLevelVisualizer #176: BedLevelVisualizer Timeout Error

FaintandFuzzies, 2020-01-08:
I suddenly started getting a timeout error during while performing a bed visualization on my CR10-S4.  After reading a few posts, I altered the timeout settings (both decreased & increased) which still resulted with the error.  I also altered the gcode command from G29 to G29 T as was suggested.  However, the error still resulted.  Can you please suggest what I should look at next?

Thank you

jneilliii, 2020-01-08:
Check logs and browser's developer console for errors. Provide output of the above commands and screenshot of settings.

FaintandFuzzies, 2020-01-09:
Will do... Thanks

FaintandFuzzies, 2020-01-10:
I reviewed the browser developer for any errors, but none were present.  I'm attaching the Octoprint log (which seems good w/o errors), Command Output and Bed Visualizer Settings as requested.
[octoprint.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/4043384/octoprint.log)
[Command_Output.txt](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/4043385/Command_Output.txt)
[image]

jneilliii, 2020-01-10:
Ok, so based on the command output it seems that your firmware is running the full probing process with `G29 T`. You need to set the timeout to a value longer than it takes for that process to complete. Toy could also try using the command `M420 V` in terminal to see if it will just give you the report, some firmware does.

PeteBurgess, 2020-01-10:
I'll stick my 2 cents in here as I'm having the same problem. The G29T completes fine, the actual processing afterwards doesn't.

It seems to have coincided with a "apt-get dist-upgrade" so maybe something's changed in one of the dependencies ?

jneilliii, 2020-01-10:
You didn't happen to notice what got upgraded did you?  The only dependency I can think of right now is numpy, but all of that is managed under the python virtual environment, not apt.

jneilliii, 2020-01-10:
Another user is having similar issue in #177 that indicates a potential problem with large offsets.

FaintandFuzzies, 2020-01-10:
I set the timeout up to 100 and it still produced the error... is that not enough?  How is it that everything worked fine with the default settings and using a G29 (instead of G29 T) command worked without error, and now it doesn't?

jneilliii, 2020-01-10:
Not sure why all of a sudden it stopped working, did anything else change in your environment?

PeteBurgess, 2020-01-10:
> You didn't happen to notice what got upgraded did you? The only dependency I can think of right now is numpy, but all of that is managed under the python virtual environment, not apt.

No, but I can pull a log file later. There's a ton of stuff changed on the Pi recently but I'm 99% certain it was after this started occurring.

FaintandFuzzies, 2020-01-10:
> Not sure why all of a sudden it stopped working, did anything else change in your environment?

This is what I see in the logs that recently changed:
The bed visualizer updated on 09/16/2019 from 0.1.8 to 0.1.10
The bed visualizer updated on 11/24/2019 from 0.1.10 to 0.1.11
OctoPrint updated on 11/24/2019 from 1.3.11 to 1.3.12  (which is the current version I'm on)

When I used command M420 V, this is what resulted:
Send: M420 V
Recv: echo:Bed Leveling Off
Recv: echo:Fade Height Off

Nothing else resulted from the command.

jneilliii, 2020-01-10:
So nothing just before the issue started happening, assuming you've ran a mesh update after each of the updates. If there aren't errors in the developer console on initial load of the web interface or during the process then I'm at a loss.

FaintandFuzzies, 2020-01-10:
Everything worked fine after each update.  Is there a way to install a prior version of the plugin to see if that does anything different?

FaintandFuzzies, 2020-01-10:
I still rec'd the same error on version 0.1.9.

I decided to time exactly how long it takes for my S4 to create a 4x4 mesh.  It took 130sec, so I set the timeout for 135sec for good measure and ran the mesh generation again.  It didn't error out this time.

I don't know why this is working because I've never touched the timeout setting before and just let the defaults remain (60sec) and never had problems.

I'll go ahead and upgrade to the 0.1.10 version and leave the timeout at 135sec and let it be :P

jneilliii, 2020-01-11:
Sounds like your probing speed changed where it's taking longer now, but seems it's fixed now.

closed 2020-01-11 by jneilliii

mark-hahn, 2020-01-29:
I'm having this problem and setting comm timeouts to 120 secs didn't fix it.  My G29 takes 50 secs. 

However, the G29 finishes and the results are shown so I am just ignoring the error message for now.

mark-hahn, 2020-01-30:
Never mind.  I was updating the timeout for octoprint serial.  I just now noticed the timeout in this plugin settings and bumping it up fixed my timeout.