jneilliii/OctoPrint-BedLevelVisualizer #177: Regular expression fails for grid with big numbers

Uinden, 2020-01-10:
I am trying to setup autolevel for my printer so my report for `G29 T` has big numbers.
```
Recv: Bilinear Leveling Grid:
Recv:       0      1      2
Recv:  0 +35.993 +36.038 +36.075
Recv:  1 +35.953 +36.077 +36.198
Recv:  2 +35.999 +36.047 +36.129
Recv: 
```
Probably I'm doing something wrong but its my first time.
And I tried to see mesh in BedLevelVisualizer but it just waits for response forever.
Looks like problem in `processGCODE()`
```python
if self.processing and "ok" not in line and re.match(r"^((G33.+)|(Bed.+)|(\d+\s)|(\|\s*)|(\[?\s?\+?\-?\d?\.\d+\]?\s*\,?)|(\s?\.\s*)|(NAN\,?))+$", line.strip()):
```
Looks like it can't process those numbers
[image]

After I got lower numbers it showed my mesh in BedLevelVisualizer and `.match` works.
[image]

Can we have fix for it or maybe some warning like "Your numbers are too big, check your leveling settings".

jneilliii, 2020-01-10:
The match probably worked fine the first time, the surface mesh was just out of view on the display because it only shows +/- 2mm from 0. Enabling the option to use relative offsets would fix that.

Uinden, 2020-01-10:
@jneilliii 
Tested again. Small numbers works without problem, but 30+ just waits forever (or until timeout)
[image]

PR #184 (2020-03-15): 0.1.12
#### Added
- Rotation options to mesh visualization, community forum request.
- Custom debug logging for future troubleshooting
- Custom commands that will give additional buttons on tab, resolves #166 
- Cancel button to stop processing when visualization gets stuck, resolves #182 
- Larger precision number processing, resolves #177
- Themify support, will now color the background and axis based on tab colors, resolves #181 
- Marlin Mesh Bed Leveling support, thanks to #186 
- Marlin OpenScad Output support, thanks to #186 
- Option to show webcam while processing, resolves #191 

### [... cut]
closed 2020-01-31 by stale[bot]