jneilliii/OctoPrint-BedLevelVisualizer #178: Timeout when refreshing data

PeteBurgess, 2020-01-14:
Referencing #176 as it's something of a follow on. 

When clicking on "Update" to pull new mesh data, I'd expect a G29 T, followed by an updated display almost immediately. The G29 T happens but the visualizer times out. Digging into the logs, I've found the following :

2020-01-13 23:54:01,451 - octoprint.util.comm - ERROR - Error while processing hook bedlevelvisualizer:
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 2814, in _readline
    ret = hook(self, ret)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_bedlevelvisualizer/__init__.py", line 170, in processGCODE
    self.mesh = np.subtract(self.mesh, self.mesh[len(self.mesh[0])/2,len(self.mesh)/2], dtype=np.float, casting='unsafe').tolist()

jneilliii, 2020-01-14:
Can you attach your log please?

PeteBurgess, 2020-01-14:
It's a big log. That's the only relevant piece. It's printing at the moment but when it's done I'll pull the G29 T output.

PeteBurgess, 2020-01-14:
Send: G29 T
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv:     ( 20,215)                                      (215,215)
Recv:         0       1       2       3       4       5       6
Recv:  6 | +0.104  +0.067  +0.064  +0.022  +0.027  -0.042    .
Recv:    |
Recv:  5 | +0.114  +0.059  +0.050  -0.002   0.000  -0.074    .
Recv:    |
Recv:  4 | +0.082  +0.030  +0.027  -0.005  +0.010  -0.064    .
Recv:    |
Recv:  3 | +0.116  +0.077  +0.067  +0.020  +0.010  -0.069    .
Recv:    |
Recv:  2 | +0.106  +0.077  +0.067  +0.030  +0.040  -0.030    .
Recv:    |
Recv:  1 | +0.136  +0.077  +0.067  +0.017  +0.012  -0.074    .
Recv:    |
Recv:  0 | +0.077  +0.027  +0.025  -0.032  -0.027  -0.101    .
Recv:         0       1       2       3       4       5       6
Recv:     ( 20, 20)                                      (215, 20)

PeteBurgess, 2020-01-14:
2020-01-14 01:16:42,714 - octoprint.startup - INFO - ******************************************************************************
2020-01-14 01:16:42,716 - octoprint.startup - INFO - Starting OctoPrint 1.3.12
2020-01-14 01:16:42,717 - octoprint.startup - INFO - ******************************************************************************
2020-01-14 01:16:43,403 - octoprint.startup - INFO - We don't appear to be online, not fetching plugin blacklist
2020-01-14 01:16:43,477 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2020-01-14 01:16:45,920 - octoprint.plugin.core - INFO - Found 22 plugin(s) providing 22 mixin implementations, 35 hook handlers
2020-01-14 01:16:46,221 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2020-01-14 01:16:46,325 - octoprint.server - INFO - Intermediary server started
2020-01-14 01:16:46,326 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2020-01-14 01:16:46,480 - octoprint.plugin.core - INFO - Found 22 plugin(s) providing 22 mixin implementations, 35 hook handlers
2020-01-14 01:16:46,515 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2020-01-14 01:16:46,953 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2020-01-14 01:16:47,103 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2020-01-14 01:16:48,222 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python2 -m pip" as command to invoke pip
2020-01-14 01:16:49,170 - octoprint.util.pip - INFO - Version of pip is 19.3.1
2020-01-14 01:16:49,172 - octoprint.util.pip - INFO - pip installs to /home/pi/oprint/lib/python2.7/site-packages (writable -> yes), --user flag needed -> no, virtual env -> yes
2020-01-14 01:16:49,172 - octoprint.util.pip - INFO - ==> pip ok -> yes
2020-01-14 01:16:49,177 - octoprint.plugin.core - INFO - Initialized 22 plugin implementation(s)
2020-01-14 01:16:49,192 - octoprint.plugin.core - INFO - 22 plugin(s) registered with the system:
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/appkeys
|  Backup & Restore (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/backup
|  Bed Visualizer (0.1.11) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_bedlevelvisualizer
|  Core Wizard (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/corewizard
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/discovery
|  Display ETA (1.0.2) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_eta
|  DisplayLayerProgress Plugin (1.16.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_DisplayLayerProgress
|  Error Tracking (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/errortracking
|  Firmware Updater (1.6.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_firmwareupdater
|  Force Login (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/forcelogin
|  Fullscreen Plugin (0.0.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_fullscreen
|  Logging (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/logging
|  Metadata Preprocessor (0.1.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_metadatapreprocessor
|  Navbar Temperature Plugin (0.11) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_navbartemp
|  Pi Support Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pi_support
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pluginmanager
|  Printer Safety Check (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/printer_safety_check
|  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/virtual_printer
2020-01-14 01:16:49,207 - octoprint.environment - INFO - Detected environment is Python 2.7.16 under Linux (linux2). Details:
|  hardware:
|    cores: 4
|    freq: 1400.0
|    ram: 917016576
|  os:
|    id: linux
|    platform: linux2
|  plugins:
|    pi_support:
|      model: Raspberry Pi 3 Model B Plus Rev 1.3
|      octopi_version: 0.17.0
|  python:
|    pip: 19.3.1
|    version: 2.7.16
|    virtualenv: /home/pi/oprint
2020-01-14 01:16:49,218 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/webassets...
2020-01-14 01:16:49,236 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/.webassets-cache...
2020-01-14 01:16:50,062 - octoprint.server - INFO - Shutting down intermediary server...
[... cut ...]
2020-01-14 01:18:05,875 - octoprint.util.comm - ERROR - Error while processing hook bedlevelvisualizer:
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 2814, in _readline
    ret = hook(self, ret)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_bedlevelvisualizer/__init__.py", line 170, in processGCODE
    self.mesh = np.subtract(self.mesh, self.mesh[len(self.mesh[0])/2,len(self.mesh)/2], dtype=np.float, casting='unsafe').tolist()
ValueError: could not convert string to float: .

jneilliii, 2020-01-14:
Thanks, that helps a lot.  It appears that it's failing on a section in the code that is specific to center origin option.  Do you have that option enabled?  If you don't mind testing something out for me that would help out as well in fixing this problem.  If you could install the plugin using the URL below in plugin manager and then in the plugin's settings enable the debug logging option and then restart OctoPrint.  Try an update and then grab the `plugin_bedlevelvisuliver_debug.log` file from the logging section of settings and upload it here.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.12.zip

jneilliii, 2020-01-14:
The underlying issue stems from your bed report.  You'll notice in the 6 column all the offsets are a period (.) not a number, so it's not able to convert properly to do the math on that data. I'm not sure why that is happening with your firmware.
```
Send: G29 T
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv:     ( 20,215)                                      (215,215)
Recv:         0       1       2       3       4       5       6
Recv:  6 | +0.104  +0.067  +0.064  +0.022  +0.027  -0.042    .
Recv:    |
Recv:  5 | +0.114  +0.059  +0.050  -0.002   0.000  -0.074    .
Recv:    |
Recv:  4 | +0.082  +0.030  +0.027  -0.005  +0.010  -0.064    .
Recv:    |
Recv:  3 | +0.116  +0.077  +0.067  +0.020  +0.010  -0.069    .
Recv:    |
Recv:  2 | +0.106  +0.077  +0.067  +0.030  +0.040  -0.030    .
Recv:    |
Recv:  1 | +0.136  +0.077  +0.067  +0.017  +0.012  -0.074    .
Recv:    |
Recv:  0 | +0.077  +0.027  +0.025  -0.032  -0.027  -0.101    .
Recv:         0       1       2       3       4       5       6
Recv:     ( 20, 20)                                      (215, 20)
```

PeteBurgess, 2020-01-16:
Yeah, that would do it... checked Marlin and my mesh_inset etc. was a bit out of whack. Thanks for the help !

closed 2020-01-17 by jneilliii