jneilliii/OctoPrint-BedLevelVisualizer #179: Wrong coordiates used for visualization/graph

Creat, 2020-01-19:
I have a printer that has a custom bounding box defined, because at the moment the printbed is significantly smaller than the possible movement of the hotend, and I do actually need that movement.

This leads to the visualization to use the bounding box, and not the printbed size, to label the extent of the graph. So the lower left corner is -48,0, the top right is 255,210. My actual bed size is defined as 190x175. So when I want to find a spot from the graph on the bed I have to basically eyeball it, which is particularly cumbersome for UBL and manually re-probing positions (currently went back to Bilinear 
Obviously if there was a way to specify the position of the min- and the max- probe positions that would be great, cause as far as I know there is no way to get that fromt he firmware directly (marlin).

[image]
[image]
[image]

jneilliii, 2020-01-19:
> min- and the max- probe positions that would be great

You may want to look into firmware settings here prior to building. There are probing options that would achieve this in marlin.

Creat, 2020-01-19:
I think you misunderstood me: The probe positions are fine, it's probing the bed in a 5x5 pattern. The levelling itself also works well. The only things that doesn't "fit" is the visualization. It's showing the right values but the wrong coordinates. Near as I can tell it comes from the bouding box defines in Octoprint. Those are necessary for me to be able to manually move outside of the bed area.

These are the relevant marlin defines (hope I got them all):
in config:
```#define X_BED_SIZE 190
#define Y_BED_SIZE 175
#define X_MIN_POS -48
#define Y_MIN_POS 0
#define X_MAX_POS (X_MIN_POS+273) // = 225 with -48 as min
#define Y_MAX_POS (Y_MIN_POS+210)

#define GRID_MAX_POINTS_X 5
#define GRID_MAX_POINTS_Y GRID_MAX_POINTS_X
#define Z_SAFE_HOMING
#define Z_SAFE_HOMING_X_POINT (X_BED_SIZE/2) // X point for Z homing when homing all axes (G28).
#define Z_SAFE_HOMING_Y_POINT (Y_BED_SIZE/2) // Y point for Z homing when homing all axes (G28).```

in config advanced:
```#define MIN_PROBE_EDGE_LEFT  5
#define MIN_PROBE_EDGE_RIGHT (X_BED_SIZE-165) // probe at 165 from left most x position
#define MIN_PROBE_EDGE_FRONT 12
#define MIN_PROBE_EDGE_BACK  5```

Creat, 2020-02-07:
I should clarify one more thing: The problem is specific to non-mesh leveling. When in UBL mode it (at least now) works fine, since the output of marlin actually includes coordinates. This isn't the case with "classic" bed leveling (Bilinear in this case). It only outputs the offsets, but not where those were measured. But of the two options to visualize based on bed size/position or the (larger) custom bounding box, the latter is going to lead to confusion (as it by definition extends beyond the bed).

jneilliii, 2020-08-01:
I've started working on your request and was curious if this is still an issue for you since you're using UBL now? If you still have access to the Bilinear option I would like to get some logs from you if possible. If you can, enable debug logging in settings and restart OctoPrint. Run the update mesh process once and then grab the file `plugin_bedlevelvisualizer_debug.log` and drag it here into a comment. That would help me drastically in moving forward.

Creat, 2020-08-07:
I'm sorry, I'd love to help and provide feedback, but I've now switched to UBL. Honestly I switched mainly because of this plugin and me wanting to use it to it's full potential, so I got over the hurdle and just learned UBL.

Even more so, I had a (frankly probably rare) setup that caused the movable head positiong to exceed the printable area (printbed) by quite a bit. Essentially, I had temporarily mounted a smaller heated bed cause I had it lying around. By now I've got the correct size and the head can no longer move outside the printable area (as the bed is obviously intended to cover the valid range of the kinetic system).

So I can't even really just recreate the circumstance by just switching leveling mode in firmware and testing it, I'd have to also change quite a few size/position values, and I don't think I have the configs from back then.

jneilliii, 2020-08-07:
ok, no worries. I would agree your use case with the smaller bed is probably rare. I'll go ahead and close this now and if someone else comes along with a similar situation may re-open then.

closed 2020-08-07 by jneilliii