jneilliii/OctoPrint-BedLevelVisualizer #180: iPad Pro 12.9" webgl not supported error

RaoulDuke-Esq, 2020-01-20:
iPad Pro 12.9" gives bed visualizer error:
TypeError: undefined is not an object (evaluating 't.glplot.canvas')
Graph visualization shows "Webgl is not supported by your browser"
However, the browser does support webgl per get.webgl.org

I see this was fixed in 0.1.1 but I'm suspecting the large browser is not properly detected as a mobile device and thus not handled properly.  If I shrink the browser area using split-screen, the graph renders properly and I get no errors.

jneilliii, 2020-01-20:
That's a very interesting deduction. I'm wondering if this is an iOS version issue. The visualization is all handled by the plotly. js library, so would need to check that side for potential updates.

jneilliii, 2020-01-21:
I just pushed the latest ployly library to my current development branch.  If you don't mind testing to see if it resolves the problem for you, that would be great.  I do believe it may have some fixes in it that will resolve this issue for the browser/device detection. Use the URL below in plugin manager to update.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.12.zip

RaoulDuke-Esq, 2020-01-21:
Thanks for working on that so quickly!  I gave the new version a test this morning.  It still displays “WebGL is not supported by your browser” in the graph frame, but now without the popup referring to t.gplot.canvas.  Shrinking the browser area and refreshing the page still allows the graph to load.

jneilliii, 2020-01-21:
There's a small chance another plugin is also implementing plotly that is loading instead of the updated one. The only ones I know of are tempsgraph, tasmota, and tplinksmartplug. In any case, this is an upstream issue with plotly.

jneilliii, 2020-01-21:
Just to verify if you go to the following site, do they report the same "WebGL is not supported by your browser" error?

https://plot.ly/javascript/3d-surface-plots/

If so, I will open an issue on the plotly repository [here](https://github.com/plotly/plotly.js/issues/280).

RaoulDuke-Esq, 2020-01-21:
Yes, that site gives the same error. Thanks for following up and raising the issue with the upstream!

 BTW I love your plugin! My bed has never been more level.

jneilliii, 2020-01-21:
Thanks, I'm glad you like it.  Ticket has been opened with plotly.

jneilliii, 2020-01-22:
@RaoulDuke-Esq, can you chime in over at the issue on plotly repo please? 

> Which browser (& version) you are using?

https://github.com/plotly/plotly.js/issues/4502

RaoulDuke-Esq, 2020-02-03:
Looks like they've fixed this on the upstream in master.  Should be in their next release.
https://github.com/plotly/plotly.js/pull/4546

jneilliii, 2020-02-03:
Yep, I just saw that as well.  I'll try to remember to check back for that release and integrate it when the build is finalized and made public.

jneilliii, 2020-02-03:
Looks like they just released the official build.  Will get that merged in shortly and you can run some tests for me.

jneilliii, 2020-02-03:
Just pushed the latest plotly to the 0.1.12 branch for testing.  Please install using the URL below in plugin manager and make sure to force a refresh (not sure how to do this on iPad) after OctoPrint restarts if the Reload dialog doesn't show.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.12.zip

RaoulDuke-Esq, 2020-02-04:
Works like a dream! Thanks for the quick action and all your help.

PR #184 (2020-03-15): 0.1.12
#### Added
- Rotation options to mesh visualization, community forum request.
- Custom debug logging for future troubleshooting
- Custom commands that will give additional buttons on tab, resolves #166 
- Cancel button to stop processing when visualization gets stuck, resolves #182 
- Larger precision number processing, resolves #177
- Themify support, will now color the background and axis based on tab colors, resolves #181 
- Marlin Mesh Bed Leveling support, thanks to #186 
- Marlin OpenScad Output support, thanks to #186 
- Option to show webcam while processing, resolves #191 

### [... cut]
closed 2020-02-04 by RaoulDuke-Esq