jneilliii/OctoPrint-BedLevelVisualizer #181: [Feature Request] Themes support

NovaViper, 2020-01-21:
Hey, I just got Themify and made everything on there dark for my eyes, but this plugin remains to be unaffected, and I can't find a working css code for changing all of the elements of the plugin. Is it possible for you to add a dark theme like how the [TempsGraph plugin](https://plugins.octoprint.org/plugins/tempsgraph/) has an option for themes? Thanks

jneilliii, 2020-01-21:
Pretty sure others have gotten this to work with Themify.  What specific part of the graph are you not able to adjust?

NovaViper, 2020-01-21:
The background and the grid
[image]

jneilliii, 2020-01-21:
I can actually make those transparent and it will work based on the Themify settings.  I'll add this to the next version update. To test it on your end, try running this in the developer console of your browser and verify it works for you.  I just tested and it seems to work as expected with a dark background.

```
Plotly.relayout('bedlevelvisualizergraph',{'plot_bgcolor': 'rgba(0, 0, 0, 0)','paper_bgcolor': 'rgba(0, 0, 0, 0)'});
```

[image]

NovaViper, 2020-01-21:
This works! Though the numbers are a tad bit hard to see

jneilliii, 2020-01-21:
I agree.  I'll look into updating the text color based on the rest of the text on the page versus making it an option.  I did just push an update to my current dev branch.  With the transparency option set. You can install and test by using the URL in plugin manager.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.12.zip

jneilliii, 2020-01-21:
Just pushed another update to that same branch that should dynamically pull these colors from other elements on the page. You should be good to go with that one.

NovaViper, 2020-01-21:
It looks perfect in the dark theme, but is really hard to see in the white theme 
[image]

jneilliii, 2020-01-21:
Did you switch tabs after changing themes?

NovaViper, 2020-01-21:
No, I didn't.

jneilliii, 2020-01-21:
That should reload the mesh area and the text will then be black, unless you have set overrides in themify.

NovaViper, 2020-01-21:
Ah, switching tabs fixed the coloring

PR #184 (2020-03-15): 0.1.12
#### Added
- Rotation options to mesh visualization, community forum request.
- Custom debug logging for future troubleshooting
- Custom commands that will give additional buttons on tab, resolves #166 
- Cancel button to stop processing when visualization gets stuck, resolves #182 
- Larger precision number processing, resolves #177
- Themify support, will now color the background and axis based on tab colors, resolves #181 
- Marlin Mesh Bed Leveling support, thanks to #186 
- Marlin OpenScad Output support, thanks to #186 
- Option to show webcam while processing, resolves #191 

### [... cut]
closed 2020-03-15 by jneilliii