jneilliii/OctoPrint-BedLevelVisualizer #182: [Feature Request] Cancel button

lmcbmai, 2020-01-24:
During printer maintenance it is often necessary to cancel ongoing update of the chart and start over. Right now the only 2 ways to "get out" are either to wait for timeout, or to restart Octoprint. Neither is convenient,
Would be nice to have Cancel button, which will interrupt the process and discard any remaining gcode commads that had to been executed.

Thanks

jneilliii, 2020-01-24:
Will add to next update.

lmcbmai, 2020-01-24:
After looking into the code - should be very easy to implement. 'Cancel' button onclick handler should do exactly the same as timeout handler currently does.

Thanks.

lmcbmai, 2020-01-25:
I was able to add Cancel button and it work, but please double and triple check the code below, because, while I worked as a programmer for many years, that was long long ago and I've neither written Octoprint plugins, nor am very familiar with Python.

So here are the changes:
**belvisualizer.js** - added the following lines right above self.updateMesh = function () {
```
self.CancelUpdate = function () {
           clearTimeout(self.timeout);  
           self.processing(false);
        }

```
**bedlevelvisualizer_tab.jinja2**  - added the following right after <b>Please wait, retrieving current mesh.</b>
`<br><br><button class="btn" data-bind="click: CancelUpdate, enable: controlViewModel.isOperational() && !controlViewModel.isPrinting() && loginStateViewModel.isUser() && settingsViewModel.settings.plugins.bedlevelvisualizer.command() !== ''">Cancel</button>`

Works for me with OctoPrint 1.3.12 running on OctoPi 0.15.1

Thanks

jneilliii, 2020-01-25:
Had to do a little more than just that, but yes that would have handled just the front end side.  I implemented a simpleapi mixi implementation into the plugin that will also flag the backend side to stop processing and made the update button dynamic to automatically switch from Update to Cancel instead of adding another button.  Incorporated some UI changes as well recommended in #166. You can test by using the URL below in plugin manager to install.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.12.zip

lmcbmai, 2020-01-26:
I run few tests. In case of timeout the data was not updated once it became available. That is correct.
In another test I clicked Cancel while GCode was still running, i.e. did not wait for timeout. The screen changed to the last graph and GCode continued to run. This is fine, you probably don't want to terminate GCode because it may contain bed or nozzle heating commands and if terminated, the following commands switching off heaters wont' be executed. The problem, however, was that after completing G33, the graph was still updated with new data. It is up-to you to make the final decision, but in my opinion, if we canceled update request, we should discard any data received after clicking Cancel.
I also tested Flip-X, but no change there - it still hangs and eventually times out.

Thanks

jneilliii, 2020-01-26:
> I also tested Flip-X, but no change there - it still hangs and eventually times out.

Can you please open a new issue for this one and include the debug logs for additional troubleshooting? Don't remember if I ever saw those on the community forum post.

> The problem, however, was that after completing G33, the graph was still updated with new data

That's interesting.  I need to go back and walk through the workflow again.  I thought the cancel method I implemented would have not acted that way. I probably need to clear the mesh data server-side on cancel as well so that it isn't returned properly on completion of the command.

jneilliii, 2020-01-26:
Just pushed an update to the same branch that should fix that cancel logic and not continue to return updated data to be graphed now.  Let me know if the functionality is as expected now.

lmcbmai, 2020-01-27:
> Let me know if the functionality is as expected now.
Was running 26 hour-long print over weekend and had no chance to test. Will try tonight and let you know.
Thanks

lmcbmai, 2020-01-28:
Tested, still got updated chart after clicking Cancel. Looks like I cannot attach file to this post, so here is [debug log on Dropbox](https://www.dropbox.com/s/qmkculw9eyi1ej2/plugin_bedlevelvisualizer_debug.zip?dl=0)
Thanks

jneilliii, 2020-01-28:
Thanks, I'll check it out. In the future, you can just drag a log file into the comment section on github, it doesn't allow zip I think is what you ran into.

jneilliii, 2020-02-02:
I have not been able to recreate this cancel issue.  I have however, added a little more debug logging to the 0.1.2 branch linked above to the api cancelling process. This hopefully will narrow down what's going on with the results still being returned and graphed. If you don't mind reinstalling, restart OctoPrint if not prompted, force refresh on your browser (ctrl+f5 or shift+f5 depending on browser) and give it another go, that would be great.  After cancelling something like below should be added to the debug log showing the mesh prior and after clearing it..

```
[2020-02-02 00:44:58,246] DEBUG: Canceling mesh collection per user request
[2020-02-02 00:44:58,247] DEBUG: Mesh data collected prior to cancel:
[2020-02-02 00:44:58,247] DEBUG: [['-12.00', '5.00', '0.085'], ['-12.00', '28.12', '0.122'], ['-12.00', '51.25', '0.108'], ['-12.00', '74.37', '0.107'], ['-12.00', '97.50', '0.089'], ['-12.00', '120.62', '0.074'], ['-12.00', '143.75', '0.047'], ['-12.00', '166.88', '0.030'], ['-12.00', '190.00', '-0.036'], ['10.63', '5.00', '0.050'], ['10.63', '28.12', '0.098'], ['10.63', '51.25', '0.071'], ['10.63', '74.37', '0.087'], ['10.63', '97.50', '0.067'], ['10.63', '120.62', '0.061'], ['10.63', '143.75', '0.040'], ['10.63', '166.88', '0.000'], ['10.63', '190.00', '-0.032'], ['33.25', '5.00', '0.022'], ['33.25', '28.12', '0.024'], ['33.25', '51.25', '0.071'], ['33.25', '74.37', '0.035'], ['33.25', '97.50', '0.050'], ['33.25', '120.62', '0.048'], ['33.25', '143.75', '0.011'], ['33.25', '166.88', '-0.019'], ['33.25', '190.00', '-0.050'], ['55.88', '5.00', '-0.038'], ['55.88', '28.12', '-0.001'], ['55.88', '51.25', '0.021'], ['55.88', '74.37', '0.009'], ['55.88', '97.50', '0.027'], ['55.88', '120.62', '0.001'], ['55.88', '143.75', '-0.000'], ['55.88', '166.88', '-0.030'], ['55.88', '190.00', '-0.053'], ['78.50', '5.00', '-0.021'], ['78.50', '28.12', '-0.023'], ['78.50', '51.25', '0.011'], ['78.50', '74.37', '0.018'], ['78.50', '97.50', '0.017'], ['78.50', '120.62', '0.009'], ['78.50', '143.75', '-0.006'], ['78.50', '166.88', '0.003'], ['78.50', '190.00', '-0.041'], ['101.12', '5.00', '-0.042'], ['101.12', '28.12', '-0.029'], ['101.12', '51.25', '0.041'], ['101.12', '74.37', '0.024'], ['101.12', '97.50', '0.019'], ['101.12', '120.62', '0.031'], ['101.12', '143.75', '0.043'], ['101.12', '166.88', '0.019'], ['101.12', '190.00', '-0.005'], ['123.75', '5.00', '-0.046'], ['123.75', '28.12', '-0.040'], ['123.75', '51.25', '0.017'], ['123.75', '74.37', '0.021'], ['123.75', '97.50', '0.003'], ['123.75', '120.62', '0.029'], ['123.75', '143.75', '0.010'], ['123.75', '166.88', '0.029'], ['123.75', '190.00', '0.019'], ['146.38', '5.00', '-0.071'], ['146.38', '28.12', '-0.038'], ['146.38', '51.25', '0.014']]
[2020-02-02 00:44:58,247] DEBUG: Mesh data after clearing:
[2020-02-02 00:44:58,247] DEBUG: []
```

lmcbmai, 2020-02-03:
> added a little more debug logging to the 0.1.2 branch
Assuming you are referring to 0.1.12 branch...
[plugin_bedlevelvisualizer_debug.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/4146292/plugin_bedlevelvisualizer_debug.log)
The log file is attached. I run the following commands:
`G28
M300 S392 P250
G1 Y0 S1
G1 Y100 S0
G1 Z40
M140 S80
M190 S80
G1 X0 Y0
M300 S392 P250
G32 S2
G1 X150 Y150
M300 S392 P250
@BEDLEVELVISUALIZER
G33
M500
M140 S0
M300 S392 P250
G1 X348
G1 Y2
G28 X Y`

I click Cancel when it bed was heating (M140 S80). 
Thanks

jneilliii, 2020-02-03:
That makes sense now.  Looking at your gcode the collection doesn't even start until the flag is seen, so canceling prior to @BEDLEVELVISUALIZER wouldn't work in this case.  I'll add a separate flag for cancelled and check against that.  That should get you squared away.  Will work on the necessary changes during the week as time permits.  Thanks for all the debugging assistance.

lmcbmai, 2020-02-03:
Sure, ready to help. Thanks for outstanding support!
I can move @BEDLEVELVISUALIZER to the top of the script if that will make difference.

jneilliii, 2020-02-03:
That probably won't work either I don't think because the first OK back from the printer to OctoPrint would trigger stopping collection.

jneilliii, 2020-02-03:
So I took a quick stab at a separate mesh_collection_cancelled flag and it's breaking the serial communication, at least the way in which I'm testing. I notice with the virtual printer when I send G33 it gives this prompt.
```
Recv: G33 Auto Calibrate
Recv: Will take ~60s
```
Then followed after with some temp and sd status message and then a serialtimeout exception
```
Serial timeout while writing to serial port, trying again.
Unexpected error while writing to serial port: SerialTimeoutException: '' @ comm.py:_do_send_without_checksum:3459
Please see https://faq.octoprint.org/serialerror for possible reasons of this.
Changing monitoring state from "Operational" to "Offline (Error: SerialTimeoutException: '' @ comm.py:_do_send_without_checksum:3459)"
Closing down send loop
Connection closed, closing down monitor
```
This could be solely a virtual printer issue, but I just pushed the changes to the 0.1.12 branch if you want to try it out.

lmcbmai, 2020-02-03:
No change. Here is the latest log
[plugin_bedlevelvisualizer_debug.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/4146740/plugin_bedlevelvisualizer_debug.log)

jneilliii, 2020-02-07:
I think I figured out what was wrong.  The cancel flag was named wrong in the api callback.  Should be fixed now and working as expected.

lmcbmai, 2020-02-07:
SUCCESS!! Just tested and no more updates after cancellation. :)

Thanks again for great support!!

PR #184 (2020-03-15): 0.1.12
#### Added
- Rotation options to mesh visualization, community forum request.
- Custom debug logging for future troubleshooting
- Custom commands that will give additional buttons on tab, resolves #166 
- Cancel button to stop processing when visualization gets stuck, resolves #182 
- Larger precision number processing, resolves #177
- Themify support, will now color the background and axis based on tab colors, resolves #181 
- Marlin Mesh Bed Leveling support, thanks to #186 
- Marlin OpenScad Output support, thanks to #186 
- Option to show webcam while processing, resolves #191 

### [... cut]
closed 2020-03-15 by jneilliii