jneilliii/OctoPrint-BedLevelVisualizer #183: Constantly scanning printer output?

lmcbmai, 2020-01-24:
I am under impression that the plugin scans printer output all the time, i.e. not only while doing update of the chart. The reason I am saying this is because I have had several cases when the chart was successfully updated few minutes after timing out. Again, this is just an impression, I did not look into the code and cannot be positive that that is actually what is happening by design.
One of the main features of any plugin should be to be as lightweight as possible and do not take extra resources (CPU, memory, etc.) when it is not doing anything. Thus, if BedLevelVisualizer really looking into the content of serial communication, parsing the strings and searching to the patterns it can recognize, I'd suggest to limit all this only to the period from when user clicks update until the update is completed if timeout occurs.

Thanks

jneilliii, 2020-01-24:
You are correct, the plugin utilizes the gcode.received hook which passes the received serial communication through a function.  The plugin doesn't do any extensive manipulation/gathering however until the command `@BEDLEVELVISUALIZER` is sent and I've never seen any performance degradation caused, but I'm not a professional programmer and wouldn't even know where to begin to profile that. It may be possible to not register the gcode.received hook until after that command is sent, but I would have to do some research to see if that is even possible.  Have you experienced an issue with performance with the plugin enabled versus disabled?  This plugin is one of the top 3 installed globally and you're the first to mention this.

lmcbmai, 2020-01-24:
That's fine. That's why I posted this as a suggestion and not as feature request or bug report :) 

Thanks for the great work!

lmcbmai, 2020-01-24:
I checked the code and actually you are doing everything right. You have **self.processing** flag that disables parsing of serial input when you are not collecting new data. I think you just need to add **self.processing(false)** to the **setTimeout** function in javascript file. When timeout happens you will stop looking into further input.

Also, from code optimization point of view, I'd suggest replacing all **if self.processing** in **def processGCODE** with single **if self.processing** at the very top of **processGCODE**.  Basically if you are not in processing mode, you want to get out of **processGCODE** right away. 

Thanks

lmcbmai, 2020-01-25:
> I think you just need to add self.processing(false) to setTimeout function in javascript file.
Sorry, I just mentioned that you do have self.processing(false)  in timeout handler. Not sure why the data is still collected after timeout occurs.

jneilliii, 2020-01-25:
Yeah, just realized that when I was looking at the cancel button implementation.  The self.process(true) only handles the front end and doesn't instruct the backend to change its self.processing to false.  I'll definitely fix that.

jneilliii, 2020-01-25:
So actually looking at the code again, the processing should have automatically stopped upon receiving the next `ok` from the printer, but I'm still incorporating forcing the processing flag on the server side to be set to false on timeout since I had to do that for the cancel button anyway.

jneilliii, 2020-01-25:
Added simplapiplugin mixin implementation in latest development branch to flag server side processing to false on timeout and cancel.

PR #184 (2020-03-15): 0.1.12
#### Added
- Rotation options to mesh visualization, community forum request.
- Custom debug logging for future troubleshooting
- Custom commands that will give additional buttons on tab, resolves #166 
- Cancel button to stop processing when visualization gets stuck, resolves #182 
- Larger precision number processing, resolves #177
- Themify support, will now color the background and axis based on tab colors, resolves #181 
- Marlin Mesh Bed Leveling support, thanks to #186 
- Marlin OpenScad Output support, thanks to #186 
- Option to show webcam while processing, resolves #191 

### [... cut]
closed 2020-03-15 by jneilliii