jneilliii/OctoPrint-BedLevelVisualizer #185: Cannot parse Repetier G33 output when Flip X option is selected

lmcbmai, 2020-01-28:
The subject says it all. Ran test on Repetier 1.0.4 firmware with Flip X selected and timeout set to 20min. G32 (5x5) followed by G33 (9x9) took about 15 minutes but the screen was still showing Please wait... until timeout. Debug log is available [from this link on Dropbox](https://www.dropbox.com/s/ygqtbokuv2vkldk/plugin_bedlevelvisualizer_debug_Flip_X.zip?dl=0).
When Flip X is not selected, all other combination of Flip Y and Rotate work just fine.

Thanks

jneilliii, 2020-01-28:
Can you please post your gcode command settings?

lmcbmai, 2020-01-28:
Here is my GCode. I added few comments for your convenience 
```
; @BEDLEVELVISUALIZER
; G33 L0
; ----------
G28 Z
G28 Y
G28 X
M300 S392 P250  ; beep
M140 S80 ;Start heating bed
M190 S80 ;Wait for bed to reach temp before proceeding
G1 Y2 S1  ; activate probe step 1
G1 Y100 S0 ; activate probe step 2
G1 Z40 
G1 X0 Y0
M300 S392 P250 ; beep
;------------
G32 S2
;-----------
G1 X150 Y150
M300 S392 P250 ; beep
@BEDLEVELVISUALIZER
G33
M500 ; save config in EEPROM
;-----------
M140 S0 ; Disable heated bed
M300 S392 P250 ; beep
G1 X348   ; deactivate probe step 1
G1 Y2    ; activate probe step 2
G28 X Y 
```
Please note that my z-probe is activated/deactivated by magnets when printhead move to at a special locations outside print area. 

Thanks

jneilliii, 2020-01-29:
Is the part shown below actually in the settings or added for convenience?

```
; @BEDLEVELVISUALIZER
; G33 L0
; ----------
```

I don't think it would cause the plugin to flag twice because I'm using the cmd.startswith("@BEDLEVELVISUALIZER") to check, but I remember those lines from our other debugging in the forum discussion. I hope to be able to dive deeper into this over the weekend.

lmcbmai, 2020-01-29:
Those 2 lines were left from my previous tests and I commented them out. 

To simplify things I just re-tested with only these 2 lines, so my GCode looks like this:
```
BEDLEVELVISUALIZER
G33 L0

```
With Flip X off it parses result right away.  With Flip X on - times out.

lmcbmai, 2020-02-01:
I think I found the problem. After adding 2 extra debug lines before and after Flip_X I got entries like this:
```
[2020-02-01 09:01:30,884] DEBUG: *** b4_Flip_X ***
[2020-02-01 09:01:30,884] DEBUG: ['355.00', '257.25', '-0.052']
[2020-02-01 09:01:30,885] DEBUG: *** After_Flip_X ***
[2020-02-01 09:01:30,885] DEBUG: ['-0.052', '257.25', '355.00']
```
**new_line** variable contains X, Y and Z coordinates of a single measurement point, not the coordinates of the entire row along X. So when **new_line.reverse()** is called we just get reverse order Z, Y, X which apparently breaks the logic down the row. Hopefully this helps.
Thanks

jneilliii, 2020-02-02:
Thanks for this.  That would actually make sense, because the plugin was originally programmed to reverse a line containing only z coordinates, like this.

```
-0.452	-0.319	-0.237	0.287	0.140	0.139	0.136	0.317	0.247	0.247
```

So the repetier processing needs to be accounted for to not flip the new_line in this case, and flip it at the end instead after it's been re-organized into z coordinates.  I'll try to get that worked out this week.

lmcbmai, 2020-02-03:
Just to give you heads up - I will be traveling starting this coming Friday until early March and won't be able to run tests while I am away. 
Thanks

lmcbmai, 2020-03-04:
I am back from my trip and ready to help with testing whenever the fix becomes available.
Thanks

jneilliii, 2020-08-04:
Thanks to @jstevej the above commit should fix the problem for you @lmcbmai and allow flipping the x in Repetier if that's still an issue/requirement for you.

commit 42323da5 (2020-08-04): Fix x axis flip issue. (#280)

* Flip x axis properly. This fixes #185
commit 3fb614e6 (2020-10-03): 0.1.15 (#304)

fix axis label colors, resolves #278
make colorscale limits allow option auto for the old method of coloring, resolves #275
add responsive option to plotly config, resolves #274
first stab at SD card support, resolves #223
fix little issue with reactive option where graph was getting overlaid on top of correction table
change home button press to a different strategy, resolves #265
update plotly to version 1.54.7
add option to download snapshots on graph rendering completion, #239
remove custom home button in preference of resetCameraLastSave3d, resolves #265
numpy versioning fo [... cut]
commit 7565c066 (2020-10-03): 0.1.15 (#304)

fix axis label colors, resolves #278
make colorscale limits allow option auto for the old method of coloring, resolves #275
add responsive option to plotly config, resolves #274
first stab at SD card support, resolves #223
fix little issue with reactive option where graph was getting overlaid on top of correction table
change home button press to a different strategy, resolves #265
update plotly to version 1.54.7
add option to download snapshots on graph rendering completion, #239
remove custom home button in preference of resetCameraLastSave3d, resolves #265
numpy versioning fo [... cut]
closed 2020-08-18 by jneilliii