jneilliii/OctoPrint-BedLevelVisualizer #190: Force loading of mesh not stored by BV

KenLucke, 2020-03-01:
Bed Visualizer always seems to show only the last mesh it stored from its own probe, despite having done probes at the beginning of a print (and store them) via my gcode header.  I also sometimes do bed level tests with other plugins or via the console or terminal, saving it with M500.  Can you add a button (or a toggle option) to force it to load the current mesh stored in EEPROM?

Or am I missing something that enables that already>?

jneilliii, 2020-03-01:
Have you tried switching the gcode commands to `M420 V` instead of G29 and disable thre stored option?

KenLucke, 2020-03-01:
Yes, that is a workaround for me, but I'd rather just have a button there so I can load the stored profile from other sources if I choose, or do the entire probe sequence over again directly in BV if I choose to do that.

jneilliii, 2020-03-01:
Next release actually will add custom button options, per #166 discussion.

closed 2020-03-01 by jneilliii