jneilliii/OctoPrint-BedLevelVisualizer #192: Temperature reset after Bed Leveling

pferlini63, 2020-03-09:
Hi all,
i don't know how to avoid the temperature reset afte bed leveling complete.

this is my log :

Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 -0.083 -0.071 -0.070 -0.161 -0.235
Recv:  1 -0.048 -0.045 -0.086 -0.134 -0.186
Recv:  2 -0.005 -0.002 -0.052 -0.116 -0.202
Recv:  3 +0.012 -0.015 -0.050 -0.113 -0.196
Recv:  4 +0.031 +0.020 +0.145 -0.132 -0.243
Recv: 
Recv: X:220.00 Y:195.00 Z:11.85 E:0.00 Count X:44000 Y:39000 Z:4644
Recv: ok
Send: M500
Recv: echo:Settings Stored (699 bytes; crc 56206)
Recv: ok
Send: M155 S3
Recv: ok
Send: M140 S0
Recv: ok
Send: M104 S0
Recv: ok

and this is the contents of command box:

M155 S30
G28 
G29 T; update mesh
M500 ;save values
M155 S3

can you help me ?
I don't want a M104/M140 after completion.

Thanks in advance

Paolo

jneilliii, 2020-03-12:
The plugin doesn't inject those unless they are in the plugin's settings. Do you have any GCODE in the GCODE scripts section of OctoPrint's settings that may be sending these commands?

pferlini63, 2020-03-12:
Dear jneilliii,
After a remove and reinstall the problem disappear.

So the case can be closed.

Thanks a lot for your courtesy.

Paolo

closed 2020-03-12 by jneilliii