jneilliii/OctoPrint-BedLevelVisualizer #195: Recent update Broke Bed Visualizer data update

sinuehT, 2020-03-15:
Since the most recent update my CR10 V2 bed update information does not feed through and the Bed Visualizer plugin cites a timeout shortly after issuing the command. Please can you help me correct this?

Mitnichten, 2020-03-15:
On my Ender3 the leveling  runs through normally but at the end of the process i get a timeout and the the gathered information is lost.

jneilliii, 2020-03-15:
Please enable debug logging, restart octoprint, try the process and then check the log for errors or upload here.

brendanstennett, 2020-03-15:
+1

Firmware: Marlin TH3D U1.R2.A7
Printer: Ender 3
Plugin Version: 0.1.12

Commands:
```
G28
@BEDLEVELVISUALIZER
G29 T
```

Log Output:
```
[2020-03-15 16:39:43,793] DEBUG: mesh collection started
[2020-03-15 16:39:45,801] DEBUG: echo:busy: processing
[2020-03-15 16:39:47,800] DEBUG: echo:busy: processing
[2020-03-15 16:39:49,800] DEBUG: echo:busy: processing
[2020-03-15 16:39:51,800] DEBUG: echo:busy: processing
[2020-03-15 16:39:53,800] DEBUG: echo:busy: processing
[2020-03-15 16:39:55,801] DEBUG: echo:busy: processing
[2020-03-15 16:39:57,801] DEBUG: echo:busy: processing
[2020-03-15 16:39:59,802] DEBUG: echo:busy: processing
[2020-03-15 16:40:01,800] DEBUG: echo:busy: processing
[2020-03-15 16:40:03,804] DEBUG: echo:busy: processing
[2020-03-15 16:40:04,983] DEBUG: Canceling mesh collection per user request
[2020-03-15 16:40:04,983] DEBUG: Mesh data collected prior to cancel:
[2020-03-15 16:40:04,984] DEBUG: []
[2020-03-15 16:40:04,984] DEBUG: Mesh data after clearing:
[2020-03-15 16:40:04,984] DEBUG: []
```

Steps to reproduce:
1. Click "Update Mesh"
- Screen turns to "Mesh Updating" display with spinner
- Printer starts the probing
- Shortly after, and at about probe 4/9, the spinner stops and the screen returns back to "Update Mesh" button
- After probe finishes, no display shows and clicking "Update Mesh" again starts the process all over again.

jneilliii, 2020-03-15:
Is your timeout value large enough?

brendanstennett, 2020-03-15:
@jneilliii That was the issue.  Timeout of 60 seconds was too short.

Thanks.

jneilliii, 2020-03-16:
Thanks @brendanstennett, what about you @sinuehT and @Mitnichten, is it a similar issue and you just need to increase the timeout?

phil303, 2020-03-16:
As another data point to the above discussion, I'm having the issue where the plugin is not doing the visualizing after the printer has run through the commands. I'm running an Ender 3 pro, Marlin 2.0.4.3, BLTouch 3.1, and tested this with both:

```
G28
G29 T
```
and
```
G28
@BEDLEVELVISUALIZER
G29 T
```
with a 1 minute timeout and a 5 minute timeout. In the 5 minute timeout case, the spinner keeps spinning for the entire duration before going back to the "Update Mesh Now" button (but no visualization).

When I check the terminal I see the ABL output but the plugin doesn't seem to be picking up on it.

Thanks for your work on this plugin @jneilliii. I'm excited to get this working.

phil303, 2020-03-16:
@jneilliii Not sure what changed but after I enabled debug logging and restarted Octoprint, it started working for me. From what I can tell there are no errors from that run. I turned off debug logging, restarted, and ran again to test if it was related to that but it still works.

Mitnichten, 2020-03-16:
Firmware: FIRMWARE_NAME:Marlin bugfix-1.1.x
Printer: Ender 3
Plugin Version: 0.1.12

### Commands:
G28
@BEDLEVELVISUALIZER
G29

### Log Output

> Send: G28
Recv:  T:19.22 /0.00 B:19.14 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:19.22 /0.00 B:19.10 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:19.22 /0.00 B:19.34 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:19.34 /0.00 B:19.26 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:19.18 /0.00 B:19.18 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:19.10 /0.00 B:19.14 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:19.10 /0.00 B:19.06 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:19.10 /0.00 B:19.22 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv: X:160.00 Y:126.00 Z:11.70 E:0.00 Count X:12800 Y:10080 Z:4680
Recv: ok
Send: G29
Recv:  T:19.18 /0.00 B:19.14 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:19.22 /0.00 B:19.02 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:19.26 /0.00 B:19.10 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:19.14 /0.00 B:19.18 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:19.18 /0.00 B:19.18 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:19.30 /0.00 B:19.18 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:19.30 /0.00 B:19.26 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv:  T:19.18 /0.00 B:19.30 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:19.53 /0.00 B:19.34 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:19.22 /0.00 B:19.26 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:19.57 /0.00 B:19.34 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:19.26 /0.00 B:19.22 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:19.34 /0.00 B:19.26 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:19.22 /0.00 B:19.10 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:19.53 /0.00 B:19.41 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:19.06 /0.00 B:19.26 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:19.41 /0.00 B:19.41 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:19.18 /0.00 B:19.18 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:19.18 /0.00 B:19.18 /0.00 @:0 B@:0
Recv: Bilinear Leveling Grid:
Recv:       0      1      2
Recv:  0 +0.075 +0.065 +0.115
Recv:  1 +0.028 +0.010 +0.063
Recv:  2 +0.008 +0.033 +0.108
Recv: 
Recv: X:232.00 Y:213.00 Z:11.59 E:0.00 Count X:18560 Y:17040 Z:4680
Recv: ok

### Error Message

> BED VISUALIZER ERROR
Timeout  occured before prcessing completed. Processing may still be running or  there  may be  a  fonfiguration error. Consider increasing the   timeout  value  in settings.

### Things  i tried

- Change timeout value from 60 to 300 (didn't work)

I don't know  it  worked very well before but  now  it  broke something. Probably  because of my  custom  firmware i got from TeachingTech? Anyway is  there a  way  to fix  this or  just a  way  to get  the  old version back?

fleischtorte, 2020-03-16:
Hi,
same Problem with Marlin 2.0.2 on two machines with SKR 1.0.3. With the previous it was working fine, changing the timout value has no effect :(

fleischtorte, 2020-03-16:
> 
> 
> Hi,
> same Problem with Marlin 2.0.2 on two machines with SKR 1.0.3. With the previous it was working fine, changing the timout value has no effect :(

Update:
It works with following settings : 
G28
M155 S120
@BEDLEVELVISUALIZER
G29 T
M155 S3

jneilliii, 2020-03-16:
> Anyway is there a way to fix this or just a way to get the old version back?

You can always install a previous version by manually using the URL from the releases page in Plugin Manager. Find the version you want to go back to, under assets, right-click the link for `source code (zip)` and select copy hyperlink (or similar) and paste that into the URL field of Plugin Manager.

jneilliii, 2020-03-16:
@Mitnichten, also aside from downgrading to the previous version, you could enable debug logging, restart octoprint, try the process again and provide the logs I've asked for.

Mitnichten, 2020-03-16:
> @Mitnichten, also aside from downgrading to the previous version, you could enable debug logging, restart octoprint, try the process again and provide the logs I've asked for.

Oops my bad, kinda  new to this.  How do i get  the logs when i enabled  the  option? There is nothing  in the stored  data  tab if that's where it  would be.

jneilliii, 2020-03-16:
Once you enable the option, restart octoprint, then in the logging section of OctoPrint's settings grab the file octoprint.log and plugin_bedlevelvisualizer_debug.log and you should be able to drag them into the `Write` area of your next comment here to upload and link.  You'll want to look through your octoprint.log for any personal information that you don't want to disclose to the world, if any.

sinuehT, 2020-03-16:
Please find my log files below. Any help would be greatly appreciated!

[octoprint-2.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/4340264/octoprint-2.log)
[plugin_bedlevelvisualizer_debug-2.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/4340267/plugin_bedlevelvisualizer_debug-2.log)

jneilliii, 2020-03-16:
Thanks @sinuehT, the log doesn't indicate that collection starts but seems to immediately go into a canceled state, which typically means the button was pressed while it was processing. Can you do me a favor and verify your timeout is set to a high number, ie 300 and enable debug logging is checked in the plugin's settings.  Restart your whole pi from the system menu. Force refresh your browser (shift+f5 or ctrl+f5 depending on browser) to make sure there isn't any lingering stuff in cache. And then try the process again.

sinuehT, 2020-03-16:
I’m not sure what changed as I thought I’d done the same before, but this worked. I did the following:
- disabled system logging
- enabled the logging in the plugin settings only
- set the timeout to 300
- restarted

And it worked. 

Thank you!

bonoboxita, 2020-03-19:
Same for me...not updating the data

jneilliii, 2020-03-19:
@bonoboxita, have you tried the steps listed above that have fixed this for others? If not, follow the steps above and provide logs, otherwise I can't help you.

bonoboxita, 2020-03-19:
yes...it's not updating.
Now by magic i have an updated mesh of this afternoon but it's not updating with the one i'm doing now. 
Also using 
G28
M155 S120
@BEDLEVELVISUALIZER
G29 T
M155 S3

any plan to fix the plugins?

jneilliii, 2020-03-19:
Well, it's hard for me to fix if you don't provide any logs or details other than what you have put above for what you have in settings.  For example, what is your timeout setting? Have you enabled debug logging and restarted octoprint and uploaded the log files I mentioned before in this thread? I can't fix what I can't reproduce and the plugin is working fine for me and most others in this thread have figured out one thing or another that fixed their installation.

sinuehT, 2020-03-20:
It’s worth saying that this is an extremely valuable plug-in which notably many use frequently. Thank you for creating such a great tool!

titogarrido, 2020-03-20:
Hi Guys!

My bed visualizer is not updating, here my config and logs:
Ender 3 + BLTouch v3.1
SKR 1.2 Mini
Marlin 2.0.5

GCODE for bed Visualizer:
```
G28
M155 S300
@BEDLEVELVISUALIZER
G29 T
M155 S3
```

G29 T output:
```
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 -1.055 -0.607 -0.160 +0.290 +0.740
Recv:  1 -0.845 -0.470 -0.095 +0.280 +0.645
Recv:  2 -0.730 -0.395 -0.070 +0.253 +0.580
Recv:  3 -0.620 -0.337 -0.055 +0.235 +0.535
Recv:  4 -0.445 -0.195 +0.028 +0.278 +0.508
Recv: 
Recv: X:235.00 Y:230.00 Z:12.21 E:0.00 Count X:18800 Y:18400 Z:5160
Recv: ok
```

Output from bedvisualizer logs:
```
[2020-03-20 10:58:07,237] DEBUG: Canceling mesh collection per user request
[2020-03-20 10:58:07,238] DEBUG: Mesh data collected prior to cancel:
[2020-03-20 10:58:07,239] DEBUG: []
[2020-03-20 10:58:07,240] DEBUG: Mesh data after clearing:
[2020-03-20 10:58:07,241] DEBUG: []
```

Output from Terminal after bedvisualizer logs:
```
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 -1.040 -0.600 -0.165 +0.280 +0.715
Recv:  1 -0.865 -0.490 -0.117 +0.265 +0.640
Recv:  2 -0.737 -0.408 -0.090 +0.235 +0.560
Recv:  3 -0.640 -0.342 -0.075 +0.223 +0.513
Recv:  4 -0.455 -0.215 +0.013 +0.268 +0.498
Recv: 
Recv: X:235.00 Y:230.00 Z:12.23 E:0.00 Count X:18800 Y:18400 Z:5160
Recv: ok
Send: M155 S3
Recv: ok
```

The stored values doesn't change, it is still showing old values:
```
-0.800	-0.140	+0.500
-0.533	-0.082	+0.362
-0.330	-0.010	+0.310
```

Options Selected:
 Save Mesh 
 Enable Debug Logging: 

Thanks!

Norvat, 2020-03-20:
I have the same problem.
The visualizer does not update with the new leveling information.

my plugin gcode:

> G28
M155 S120
@BEDLEVELVISUALIZER
G29 T
M155 S3

Error code:

[plugin_bedlevelvisualizer_debug.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/4360015/plugin_bedlevelvisualizer_debug.log)

> [2020-03-20 13:57:10,565] DEBUG: mesh collection started
[2020-03-20 13:57:12,882] DEBUG: echo:busy: processing
[2020-03-20 13:57:14,881] DEBUG: echo:busy: processing
[2020-03-20 13:57:16,881] DEBUG: echo:busy: processing
[2020-03-20 13:57:18,881] DEBUG: echo:busy: processing
[2020-03-20 13:57:20,880] DEBUG: echo:busy: processing
[2020-03-20 13:57:22,881] DEBUG: echo:busy: processing
[2020-03-20 13:57:25,002] DEBUG: echo:busy: processing
[2020-03-20 13:57:27,002] DEBUG: echo:busy: processing
[2020-03-20 13:57:29,002] DEBUG: echo:busy: processing
[2020-03-20 13:57:31,362] DEBUG: echo:busy: processing
[2020-03-20 13:57:33,406] DEBUG: echo:busy: processing
[2020-03-20 13:57:35,407] DEBUG: echo:busy: processing
[2020-03-20 13:57:37,713] DEBUG: echo:busy: processing
[2020-03-20 13:57:39,714] DEBUG: echo:busy: processing
[2020-03-20 13:57:40,856] DEBUG: T:50.11 /0.00 B:42.21 /0.00 @:0 B@:0
[2020-03-20 13:57:41,714] DEBUG: echo:busy: processing
[2020-03-20 13:57:44,492] DEBUG: echo:busy: processing
[2020-03-20 13:57:46,536] DEBUG: echo:busy: processing
[2020-03-20 13:57:48,536] DEBUG: echo:busy: processing
[2020-03-20 13:57:51,153] DEBUG: echo:busy: processing
[2020-03-20 13:57:53,152] DEBUG: echo:busy: processing
[2020-03-20 13:57:55,152] DEBUG: echo:busy: processing
[2020-03-20 13:57:57,797] DEBUG: echo:busy: processing
[2020-03-20 13:57:59,843] DEBUG: echo:busy: processing
[2020-03-20 13:58:01,842] DEBUG: echo:busy: processing
[2020-03-20 13:58:04,431] DEBUG: echo:busy: processing
[2020-03-20 13:58:06,476] DEBUG: echo:busy: processing
[2020-03-20 13:58:08,476] DEBUG: echo:busy: processing
[2020-03-20 13:58:11,219] DEBUG: echo:busy: processing
[2020-03-20 13:58:11,270] DEBUG: T:47.43 /0.00 B:41.54 /0.00 @:0 B@:0
[2020-03-20 13:58:13,217] DEBUG: echo:busy: processing
[2020-03-20 13:58:15,218] DEBUG: echo:busy: processing
[2020-03-20 13:58:17,559] DEBUG: echo:busy: processing
[2020-03-20 13:58:19,560] DEBUG: echo:busy: processing
[2020-03-20 13:58:21,559] DEBUG: echo:busy: processing
[2020-03-20 13:58:23,895] DEBUG: echo:busy: processing
[2020-03-20 13:58:25,939] DEBUG: echo:busy: processing
[2020-03-20 13:58:27,940] DEBUG: echo:busy: processing
[2020-03-20 13:58:30,225] DEBUG: echo:busy: processing
[2020-03-20 13:58:31,862] DEBUG: Canceling mesh collection per user request
[2020-03-20 13:58:31,863] DEBUG: Mesh data collected prior to cancel:
[2020-03-20 13:58:31,864] DEBUG: []
[2020-03-20 13:58:31,864] DEBUG: Mesh data after clearing:
[2020-03-20 13:58:31,865] DEBUG: []
[2020-03-20 14:06:55,847] DEBUG: mesh collection started
[2020-03-20 14:06:58,162] DEBUG: echo:busy: processing
[2020-03-20 14:07:00,206] DEBUG: echo:busy: processing
[2020-03-20 14:07:02,205] DEBUG: echo:busy: processing
[2020-03-20 14:07:04,205] DEBUG: echo:busy: processing
[2020-03-20 14:07:06,977] DEBUG: echo:busy: processing
[2020-03-20 14:07:09,021] DEBUG: echo:busy: processing
[2020-03-20 14:07:11,020] DEBUG: echo:busy: processing
[2020-03-20 14:07:13,363] DEBUG: echo:busy: processing
[2020-03-20 14:07:15,407] DEBUG: echo:busy: processing
[2020-03-20 14:07:17,406] DEBUG: echo:busy: processing
[2020-03-20 14:07:19,724] DEBUG: echo:busy: processing
[2020-03-20 14:07:21,723] DEBUG: echo:busy: processing
[2020-03-20 14:07:23,723] DEBUG: echo:busy: processing
[2020-03-20 14:07:26,086] DEBUG: echo:busy: processing
[2020-03-20 14:07:28,130] DEBUG: echo:busy: processing
[2020-03-20 14:07:30,129] DEBUG: echo:busy: processing
[2020-03-20 14:07:32,865] DEBUG: echo:busy: processing
[2020-03-20 14:07:34,866] DEBUG: echo:busy: processing
[2020-03-20 14:07:36,865] DEBUG: echo:busy: processing
[2020-03-20 14:07:39,510] DEBUG: echo:busy: processing
[2020-03-20 14:07:41,554] DEBUG: echo:busy: processing
[2020-03-20 14:07:43,553] DEBUG: echo:busy: processing
[2020-03-20 14:07:46,157] DEBUG: echo:busy: processing
[2020-03-20 14:07:48,156] DEBUG: echo:busy: processing
[2020-03-20 14:07:50,156] DEBUG: echo:busy: processing
[2020-03-20 14:07:52,799] DEBUG: echo:busy: processing
[2020-03-20 14:07:54,842] DEBUG: echo:busy: processing
[2020-03-20 14:07:56,555] DEBUG: T:30.31 /0.00 B:32.29 /0.00 @:0 B@:0
[2020-03-20 14:07:56,843] DEBUG: echo:busy: processing
[2020-03-20 14:07:59,584] DEBUG: echo:busy: processing
[2020-03-20 14:08:01,629] DEBUG: echo:busy: processing
[2020-03-20 14:08:03,629] DEBUG: echo:busy: processing
[2020-03-20 14:08:05,920] DEBUG: echo:busy: processing
[2020-03-20 14:08:07,964] DEBUG: echo:busy: processing
[2020-03-20 14:08:09,964] DEBUG: echo:busy: processing
[2020-03-20 14:08:12,250] DEBUG: echo:busy: processing
[2020-03-20 14:08:14,249] DEBUG: echo:busy: processing
[2020-03-20 14:08:14,530] DEBUG: Canceling mesh collection per user request
[2020-03-20 14:08:14,531] DEBUG: Mesh data collected prior to cancel:
[2020-03-20 14:08:14,532] DEBUG: []
[2020-03-20 14:08:14,533] DEBUG: Mesh data after clearing:
[2020-03-20 14:08:14,533] DEBUG: []

jneilliii, 2020-03-20:
@titogarrido and @Norvat, in both your cases your log indicates...

> Canceling mesh collection per user request

This means your timeout value isn't long enough. Increase timeout value, restart octoprint and try again. I recommend setting the value to something that is 30-60 seconds longer than how long the leveling process takes for you to complete.

titogarrido, 2020-03-20:
Hi @jneilliii 

I have increased to 180s, the bedlevel ended, I have waited more seconds and received:
[2020-03-20 11:49:21,489] DEBUG: Canceling mesh collection per user request
[2020-03-20 11:49:21,491] DEBUG: Mesh data collected prior to cancel:
[2020-03-20 11:49:21,493] DEBUG: []
[2020-03-20 11:49:21,495] DEBUG: Mesh data after clearing:
[2020-03-20 11:49:21,496] DEBUG: []

The Graph shows:
Using saved mesh data from 02/03/2020 19:10:59.

titogarrido, 2020-03-20:
My bad! I have forgot to restart octoprint. It is working now with 180s

jneilliii, 2020-03-20:
> I have forgot to restart octoprint

I'll look into why the restart seems to be necessary for this to take.  It shouldn't be that way, but will investigate further.

Borimino, 2020-03-21:
I had a similar problem as the other people, and restarting fixed it for me aswell. Just as another datapoint.

jiynn, 2020-03-23:
I'm having similar problems with the mesh not updating... before it kept displaying a previous date, now after an uninstall, restart, reinstall, and restart... it's not showing anything at all!

The last bit I see in the terminal is this:
```
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: Bilinear Leveling Grid:
Recv:       0      1      2
Recv:  0 +0.196 +0.036 -0.094
Recv:  1 +0.218 +0.059 -0.098
Recv:  2 +0.228 +0.086 -0.052
Recv: 
Recv: echo:busy: processing
Recv: echo:busy: processing
```

however the log show this:

```
[2020-03-23 15:36:08,109] DEBUG: mesh collection started
[2020-03-23 15:36:10,124] DEBUG: echo:busy: processing
[2020-03-23 15:36:12,125] DEBUG: echo:busy: processing
[2020-03-23 15:36:14,125] DEBUG: echo:busy: processing
[2020-03-23 15:36:16,126] DEBUG: echo:busy: processing
[2020-03-23 15:36:18,126] DEBUG: echo:busy: processing
[2020-03-23 15:36:20,126] DEBUG: echo:busy: processing
[2020-03-23 15:36:22,126] DEBUG: echo:busy: processing
[2020-03-23 15:36:24,126] DEBUG: echo:busy: processing
[2020-03-23 15:36:26,128] DEBUG: echo:busy: processing
[2020-03-23 15:36:28,128] DEBUG: echo:busy: processing
[2020-03-23 15:36:30,128] DEBUG: echo:busy: processing
[2020-03-23 15:36:32,127] DEBUG: echo:busy: processing
[2020-03-23 15:36:34,126] DEBUG: echo:busy: processing
[2020-03-23 15:36:36,126] DEBUG: echo:busy: processing
[2020-03-23 15:36:38,381] DEBUG: echo:busy: processing
[2020-03-23 15:36:40,380] DEBUG: echo:busy: processing
[2020-03-23 15:36:42,379] DEBUG: echo:busy: processing
[2020-03-23 15:36:44,379] DEBUG: echo:busy: processing
[2020-03-23 15:36:46,379] DEBUG: echo:busy: processing
[2020-03-23 15:36:48,588] DEBUG: echo:busy: processing
[2020-03-23 15:36:50,586] DEBUG: echo:busy: processing
[2020-03-23 15:36:52,586] DEBUG: echo:busy: processing
[2020-03-23 15:36:54,585] DEBUG: echo:busy: processing
[2020-03-23 15:36:56,585] DEBUG: echo:busy: processing
[2020-03-23 15:36:58,586] DEBUG: echo:busy: processing
[2020-03-23 15:37:01,375] DEBUG: echo:busy: processing
[2020-03-23 15:37:03,374] DEBUG: echo:busy: processing
[2020-03-23 15:37:05,696] DEBUG: echo:busy: processing
[2020-03-23 15:37:07,695] DEBUG: echo:busy: processing
[2020-03-23 15:37:09,695] DEBUG: echo:busy: processing
[2020-03-23 15:37:11,694] DEBUG: echo:busy: processing
[2020-03-23 15:37:13,697] DEBUG: echo:busy: processing
[2020-03-23 15:37:15,696] DEBUG: echo:busy: processing
[2020-03-23 15:37:18,455] DEBUG: echo:busy: processing
[2020-03-23 15:37:20,453] DEBUG: echo:busy: processing
[2020-03-23 15:37:22,751] DEBUG: echo:busy: processing
[2020-03-23 15:37:24,754] DEBUG: echo:busy: processing
[2020-03-23 15:37:26,753] DEBUG: echo:busy: processing
[2020-03-23 15:37:29,553] DEBUG: echo:busy: processing
[2020-03-23 15:37:31,553] DEBUG: echo:busy: processing
[2020-03-23 15:37:33,552] DEBUG: echo:busy: processing
[2020-03-23 15:37:35,552] DEBUG: echo:busy: processing
[2020-03-23 15:37:37,556] DEBUG: echo:busy: processing
[2020-03-23 15:37:39,759] DEBUG: echo:busy: processing
[2020-03-23 15:37:41,759] DEBUG: echo:busy: processing
[2020-03-23 15:37:43,760] DEBUG: echo:busy: processing
[2020-03-23 15:37:46,562] DEBUG: echo:busy: processing
[2020-03-23 15:37:48,561] DEBUG: echo:busy: processing
[2020-03-23 15:37:50,561] DEBUG: echo:busy: processing
[... 92 lines cut ...]
[2020-03-23 15:59:27,595] DEBUG: echo:busy: processing
[2020-03-23 15:59:29,893] DEBUG: echo:busy: processing
[2020-03-23 15:59:31,892] DEBUG: echo:busy: processing
[2020-03-23 15:59:33,892] DEBUG: echo:busy: processing
[2020-03-23 15:59:35,891] DEBUG: echo:busy: processing
[2020-03-23 15:59:37,891] DEBUG: echo:busy: processing
[2020-03-23 15:59:39,892] DEBUG: echo:busy: processing
[2020-03-23 15:59:42,685] DEBUG: echo:busy: processing
[2020-03-23 15:59:44,684] DEBUG: echo:busy: processing
[2020-03-23 15:59:46,850] DEBUG: Canceling mesh collection per user request
[2020-03-23 15:59:46,851] DEBUG: Mesh data collected prior to cancel:
[2020-03-23 15:59:46,851] DEBUG: []
[2020-03-23 15:59:46,851] DEBUG: Mesh data after clearing:
[2020-03-23 15:59:46,851] DEBUG: []
```

Currently I have this code set for the probe:

```
G28
@BEDLEVELVISUALIZER
G29 T
```

I am running and Ender 3 Pro with a timeout value of 180s. Any help would be greatly appreciated.

jneilliii, 2020-03-23:
> Canceling mesh collection per user request

As mentioned above, increase your Timeout value, restart OctoPrint and try again.

jiynn, 2020-03-23:
Tried that as well, the process finishes about 30 seconds before the timeout, also have increased to 300s as well, still same issue, no mesh.

jneilliii, 2020-03-23:
Just to confirm @jiynn, you did restart OctoPrint after setting the Timeout value higher? That seems to be the one thing that really triggers the fix from most responses I've see.

jiynn, 2020-03-24:
Tried the 300, went back to 180, then did 250, and restarted both the system and octoprint each time, now it finally works with 250... something is weird... I hope this doesn't fail again considering it already buried a nozzle into a textured glass bed...

jneilliii, 2020-03-24:
That is weird given the fact that you have G28 in your settings. Almost sounds like a failing probe.

bonoboxita, 2020-03-24:
I've increase the timeout to 180, restarted several time octoprint (anche checked that the timeout was correct) and reinstalled the plugin.
still not working.

jneilliii, 2020-03-24:
@bonoboxita, as asked several times before, if you don't provide logs I can't help. Enable debug logging in the plugin's settings, restart octoprint, and try again. Get the plugin_bedlevelvisualizer_debug.log and octoprint.log and drag/drop them into your reply to upload here for analysis.

bonoboxita, 2020-03-25:
used 250sec and now at least is plotting saved data after i do manually g28, g29 and m500
If i press update the mesh now the printer is not moving. After some times there is a pop up saying the plugins stopped 
these the logs

[plugin_bedlevelvisualizer_debug.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/4380806/plugin_bedlevelvisualizer_debug.log)
[octoprint.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/4380808/octoprint.log)

jneilliii, 2020-03-25:
@bonoboxita, not sure where those logs went, but I did get them via email. I assume you deleted the post. Looking at the debug log there appears to be an issue with the commands you are using. What firmware are you using?

> Unknown command: "g29 t"

bonoboxita, 2020-03-26:
i'm using Marlin 2.0.5
i've sorted out using the long method....doing the mesh, save the mash in marlin and then read it from the plugins...quite cumbersome but it works

jneilliii, 2020-03-26:
@bonoboxita, based on the debug log, have you tried setting your GCODE commands to be all upper case? Like I mentioned on the previous post, the log indicated an unknown command with lowercase `g29 t`, so maybe this is just a capitalization issue in the plugin's settings. If that is the case then I can account for that in code, but need to know if that is actually the problem or not. 

Also, does your leveling procedure complete in 3 minutes, still seems kind of short if you are doing a large mesh. If you do increase that Timeout please also make sure to restart OctoPrint for the changed setting to take effect.

TheGadge, 2020-03-27:
Sorry, I'm not used to having to change timeouts for a plugin update lol..
WHICH timeout? There's several?

Many thanks.

jneilliii, 2020-03-28:
I'm not really sure @TheGadge this is the timeout I was referring to. After changing it seems that restarting OctoPrint is the only thing that helps.

[image]

Also, it would help having more debug logs to identify what the issue might be, as well as what settings you're using for your gcode command and what firmware and version you are running. I can't reproduce the issue on my end so it's hard to identify where it's breaking.

TheGadge, 2020-03-28:
What is this timeout of? Each probe? Or the whole bed? I have mine triple probing (which as got me a variation of 0.03 on the bed lol so I'm sticking with it) but it takes quite a while.

> 
> Also, it would help having more debug logs to identify what the issue might be, as well as what settings you're using for your gcode command and what firmware and version you are running. I can't reproduce the issue on my end so it's hard to identify where it's breaking.

jneilliii, 2020-03-28:
@TheGadge, the value is for how long the whole probing process takes to complete, from the point in time when you press the update mesh button until an ok is received from the printer after the probing is complete.

I just pushed an updated version 0.1.13 that should help with all this confusion on the new timeout feature. It increases the default setting to 30 minutes and when it does timeout will show an error pop-up now instead of an info pop-up that has to be manually closed with instruction on what may need to be changed in settings.

IroCorp, 2020-03-29:
Problem solved by setting a timeout greater than the time needed to the bltouch to probe.
But I think the problem occurs because in the new version, settings are not loaded, they're blank.
So the timeout is blank.
I set a timeout, save, restart octoprint as recommended : Fail because the timeout is blank.
I set a timeout, save: it works because the timeout has a good value

Thanks

jneilliii, 2020-03-29:
Thanks @IroCorp. If the values are blank then there is an initial binding issue between the default settings and what was previously configured potentially. If that is the case, then there will be errors in your browser's developer console when you first open the web interface as part of the initial load that would help figure out a solution. Another possible fix, if you are running OctoPrint 1.4.0 is using the new erase feature to reset the settings back to defaults in Plugin Manager and make sure to restart OctoPrint and force a refresh of your browser.

[image]

MasterFuba, 2020-04-02:
seems like this is the active case...I indeed have the same issue on my Ender3. Last update was on 29.3. when i used it excessive. Now couple of days later its no more working.
Bed visualizer seems to be in quarantine like we all ;-) 
My timeout is 300, restarted rebooted, reinstalled it - nothing helps. 
My gcode in setting is this:
G28
M155 S120
@BEDLEVELVISUALIZER
G29 T
M155 S3

Something has changed in this plugin prevent it from working anymore.
KR
Daniel

jneilliii, 2020-04-02:
@MasterFuba please enable debug logging, restart OctoPrint and try an update mesh process. Then supply logs to debug as mentioned earlier in this post. Without this information I am unable to assist or find a solution for everyone. If anyone else on this issue thread that is still experiencing issues it would be very helpful if you did the same in order to narrow down a common denominator.

MasterFuba, 2020-04-02:
and where do i find this log?

MasterFuba, 2020-04-02:
[2020-04-02 01:49:13,724] DEBUG: Canceling mesh collection per user request
[2020-04-02 01:49:13,728] DEBUG: Mesh data collected prior to cancel:
[2020-04-02 01:49:13,729] DEBUG: []
[2020-04-02 01:49:13,729] DEBUG: Mesh data after clearing:
[2020-04-02 01:49:13,730] DEBUG: []
[2020-04-02 01:50:23,575] DEBUG: Canceling mesh collection per user request
[2020-04-02 01:50:23,575] DEBUG: Mesh data collected prior to cancel:
[2020-04-02 01:50:23,576] DEBUG: []
[2020-04-02 01:50:23,576] DEBUG: Mesh data after clearing:
[2020-04-02 01:50:23,576] DEBUG: []
[2020-04-02 01:53:33,169] DEBUG: Canceling mesh collection per user request
[2020-04-02 01:53:33,170] DEBUG: Mesh data collected prior to cancel:
[2020-04-02 01:53:33,171] DEBUG: []
[2020-04-02 01:53:33,171] DEBUG: Mesh data after clearing:
[2020-04-02 01:53:33,172] DEBUG: []
[2020-04-02 01:54:57,593] DEBUG: Canceling mesh collection per user request
[2020-04-02 01:54:57,595] DEBUG: Mesh data collected prior to cancel:
[2020-04-02 01:54:57,595] DEBUG: []
[2020-04-02 01:54:57,597] DEBUG: Mesh data after clearing:
[2020-04-02 01:54:57,597] DEBUG: []
[2020-04-02 02:01:46,014] DEBUG: Canceling mesh collection per user request
[2020-04-02 02:01:46,016] DEBUG: Mesh data collected prior to cancel:
[2020-04-02 02:01:46,017] DEBUG: []
[2020-04-02 02:01:46,017] DEBUG: Mesh data after clearing:
[2020-04-02 02:01:46,018] DEBUG: []
[2020-04-02 02:04:01,613] DEBUG: Canceling mesh collection per user request
[2020-04-02 02:04:01,614] DEBUG: Mesh data collected prior to cancel:
[2020-04-02 02:04:01,614] DEBUG: []
[2020-04-02 02:04:01,614] DEBUG: Mesh data after clearing:
[2020-04-02 02:04:01,615] DEBUG: []

MasterFuba, 2020-04-02:
funny is the message "Please wait, retrieving current mesh." because the printer does nothing at this point. If he would start probing it would make sense, but the printer stands still at this point!?
Is this as it should be? 
what does the "!" mean at the end below the small gacode input field. looks like a button without function. Should something happen at this point with the printer? With what kind of data the mesh should be generated if you don'T sent some kind of G29 to the printer?
Tried the bltouch buttons to probe bed and saved it also via button - in hope the bed visualizer hoock on this data. lets see if anything changes after a while.

jneilliii, 2020-04-03:
> Is this as it should be?

If your printer isn't doing anything when you press the button then the configured commands aren't getting to the printer.

> what does the "!" mean at the end below

I have no idea what you are talking about. Do you mean the exclamation icons like highlighted in the screenshot below? If so, those are tooltips, that when you hover over them they display additional help for what that setting does.

[image]

> With what kind of data the mesh should be generated if you don'T sent some kind of G29 to the printer?

It sends the commands that you configure in settings to run a probing process. As mentioned above if your printer isn't moving then that isn't happening. Is anyone else in this thread also experiencing this specific issue or is it just @MasterFuba?

jneilliii, 2020-04-03:
Also, @MasterFuba you may want to change the line `M155 S120` to something longer as well like `M155 S300` to match the processing timeout value. Additional information like what firmware and version you are running as well as how long it takes for the probing process to complete when you run it manually. 300 seconds is only 5 minutes, and the whole probing process may take longer than that if you like others here are doing multiple probes per point and a bunch of probe points in general. That is one reason why the last update defaulted that setting to 1800 seconds. You may want to start fresh and clear the settings.

[image]

MasterFuba, 2020-04-03:
[image]

hope pix is added here...
The "!" at the bottom i mean. This is when i enter the plugin. Does this inputbox do anything? The ! at the end could be a send button or is it just "there" doing nothing? I have also added this code in settings of app. I will wait now again if anything happens, but the last attempt i waited for 1h and nothing came. I have also changed to S300 lets see if it helps. 
Probing manual started takes about 1min definitv not longer. 
So my printer reacts on commands via terminal, but Bed Visualizer seems to get no data or is not able to send the probe command at all. 
Is it generating mesh out of last probe i stored via m500 or should it initially start probing anywhere when i enter app? I just see this screen and it never changes...

Marlin 1.1.9 i have on my Ender3

MasterFuba, 2020-04-03:
btw - i have resetted app data, entered code:
G28
M155 S300
@BEDLEVELVISUALIZER
G29 T
M155 S3
started manual probing via BLtouch buttons:
[image]
saved it also. 
then i enter app and see screen from prev. post....but nothing happens then.

MasterFuba, 2020-04-03:
these are my settings:
[image]

A big thanks in advance helping me finding the root of my issue.

jneilliii, 2020-04-03:
@MasterFuba, based on the below image your interface is completely messed up. Something isn't loading correctly, therefore not running through the process properly.

[image]

It should look like this after you enter your gcode commands and allow you to "Update Mesh" by pressing the button.

[image]

Please close your browser completely and open it back up. Once opened enable its developer console (usually f12 on the keyboard) and go to your OctoPrint web interface. After you login do you see any errors in the developer console?

MasterFuba, 2020-04-03:
[image]
after reboot i opened chrome with octo. This is my screen with f12
What do you mean with "open it back up"? I also tried edge to see if this browser reacts different, but same missing "update mesh now" button.

i can't see errors explzit but some parts are crossed in red - i am no html expert.....
I am asking myself if it is my laptop or if it is the octo installation itself blocking me to see this button.

[36 more replies]

closed 2020-05-20 by jneilliii