jneilliii/OctoPrint-BedLevelVisualizer #196: Separate options to 'read mesh data' and 'actively level bed'

diaconesq, 2020-03-16:
A lot of times, I level the print bed directly on the printer (i.e. via the LCD & dial).
It would be helpful to load this data into the BLVisualizer without having to go through the whole process again. 
I'm thinking of something like having two buttons in the UI ('Level bed' and 'Load data from printer') and separate GCode settings for the two options. E.g.

Level bed:
  G28 ;home axes
  G29 ;bed level
  M500 ;save to EEPROM

Read mesh data
  M420 V ;Print stored mesh data

(BTW, I find the current label 'Update mesh now' a bit unclear - it could mean either of the two options above)

jneilliii, 2020-03-16:
This was one of the reasons I added custom button options. Set the main settings to the M420 V command, set the leveling commands as a custom button.

closed 2020-03-16 by jneilliii

diaconesq, 2020-03-16:
Cool, wasn't aware of that.