jneilliii/OctoPrint-BedLevelVisualizer #203: No Visualize?

nicedevil007, 2020-03-20:
Heya,

tryed playing around with this plugin but can't get it to work.
I have a Waggster Mod (BL-Touch) on my Artillery Sidewinder X1 and setup the plugin like this:

[image]

The only thing I see all day long is this... What does that mean with the login? There is no login ability if I'm right?

[image]

jneilliii, 2020-03-20:
You're missing a lot of other settings in there, specifically the Timeout.  Make sure that is set for a number that is higher than the amount of time it takes to run the probing process manually by at least 30-60 seconds. Once you've set that setting and saved restart OctoPrint and try again.

jneilliii, 2020-03-20:
It's also going to be dependent on the artillery's firmware. I don't know if I've seen anyone using the plugin with that printer yet, but not sure. If you get a report that is similar to the one generated below at the end of the probing process then you should be good to go after doing the above steps.

```
Recv:  Bed Topography Report:
Recv:  -0.452	-0.319	-0.237	0.287	0.140	0.139	0.136	0.317	0.247	0.247
Recv:  -0.195	-0.273	-0.180	-0.178	0.014	0.018	0.111	0.214	0.210	0.210
Recv:  -0.270	-0.252	-0.151	-0.119	0.009	0.016	0.072	0.249	0.224	0.224
Recv:  -0.307	-0.205	-0.163	-0.124	-0.094	-0.002	0.036	0.151	0.174	0.196
Recv:  -0.186	-0.130	-0.152	-0.105	-0.144	-0.007	0.044	0.093	0.181	0.270
Recv:  -0.010	-0.077	-0.073	0.155	-0.006	-0.133	0.110	0.046	0.109	0.173
Recv:  0.059	-0.094	-0.072	-0.002	-0.006	0.037	0.050	0.065	0.124	0.184
Recv:  -0.057	-0.028	0.039	0.028	0.024	0.005	0.102	0.165	0.176	0.187
Recv:  0.067	0.015	0.096	0.117	0.001	0.079	0.138	0.346	0.185	0.185
Recv:  0.071	0.014	0.061	-0.127	0.167	0.040	0.098	0.195	0.194	0.194
```

nicedevil007, 2020-03-20:
Hey @jneilliii,

thank you for the fast response. Where should I add what kind of lines? In the window of the settings tab or inside that tab of the mainscreen under the "Login Message"?

I get a Bed Topography with M420 V :)

jneilliii, 2020-03-20:
The Timeout value is in the plugin settings. Also make sure that rotation is 0. Also moving the `@BEDLEVELVISUALIZER` flag just above the M420 V command will help.

[image]

nicedevil007, 2020-03-20:
Ok, just tested this out, but where the heck can I start a measurement or a new visualization?

Will this be trigger after I enter a normal G29? Or is there a missing start button in my environment?

[image]

[image]

jneilliii, 2020-03-20:
Once those settings are set, in the tab pressing the update mesh button should generate a graph. However, that 60 second timeout is probably too short if you don't move the flag after the G29 command line.

jneilliii, 2020-03-28:
This is related to #195, closing this one. BTW a new version was just released that should also help.

closed 2020-03-28 by jneilliii