jneilliii/OctoPrint-BedLevelVisualizer #206: Version 0.1.12 Update error

KixMan28, 2020-03-23:
Hi ! I am trying to update my plugin to version 0.1.12, but i receive the following error:

> Updating...
Updating, please wait.
+++++++++++++++++++++++++++++++++++++
Now updating Bed Visualizer to 0.1.12
+++++++++++++++++++++++++++++++++++++
/home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.12.zip --no-cache-dir
Collecting https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.12.zip
Downloading https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.12.zip
Exception:
Traceback (most recent call last):
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/basecommand.py", line 215, in main
status = self.run(options, args)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/commands/install.py", line 324, in run
requirement_set.prepare_files(finder)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/req/req_set.py", line 380, in prepare_files
ignore_dependencies=self.ignore_dependencies))
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/req/req_set.py", line 620, in _prepare_file
session=self.session, hashes=hashes)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/download.py", line 821, in unpack_url
hashes=hashes
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/download.py", line 659, in unpack_http_url
hashes)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/download.py", line 882, in _download_http_url
_download_url(resp, link, content_file, hashes)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/download.py", line 605, in _download_url
consume(downloaded_chunks)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/utils/__init__.py", line 852, in consume
deque(iterator, maxlen=0)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/download.py", line 571, in written_chunks
for chunk in chunks:
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/utils/ui.py", line 139, in iter
for x in it:
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/download.py", line 560, in resp_read
decode_content=False):
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/urllib3/response.py", line 432, in stream
for line in self.read_chunked(amt, decode_content=decode_content):
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/urllib3/response.py", line 626, in read_chunked
self._original_response.close()
File "/usr/lib/python2.7/contextlib.py", line 35, in __exit__
self.gen.throw(type, value, traceback)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/urllib3/response.py", line 316, in _error_catcher
raise ReadTimeoutError(self._pool, None, 'Read timed out.')
ReadTimeoutError: HTTPSConnectionPool(host='codeload.github.com', port=443): Read timed out.
The update did not finish successfully. Please consult <code>octoprint.log</code> and <code>plugin_softwareupdate_console.log</code> for details.

What is causing this? 
Can I update the plugin manually if I download the 0.1.12.zip to my Raspberry Pi? If yes, how should I proceed?

jneilliii, 2020-03-23:
That error I've never seen before, but it appears like the pi is having problems downloading the code from Github.

KixMan28, 2020-03-24:
OK. What should I do then?

jneilliii, 2020-03-24:
Restart the pi maybe.

jneilliii, 2020-03-25:
If you can manually transfer the zip file to your pi you can use the following command to install. Save the zip to your pi user home directory `~/` which is typically the default path when connecting via SSH.

```
cd ~/
~/oprint/bin/pip install --no-cache-dir ~/0.1.12.zip
sudo service octoprint restart
```

KixMan28, 2020-03-26:
@jneilliii  Thank you.

closed 2020-03-26 by KixMan28