jneilliii/OctoPrint-BedLevelVisualizer #207: Commands for Marlin 1.1.9

alexsvers, 2020-03-24:
Hi
I'm pretty new to coding ect. but otherwise an experienced 3D printer.
On my creality CR10 S5 i use Marlin 1.1.9 (although it is a customized version for the BondTech DD extruder and BLTouch V3.1). I have tried several suggested commands but without luck. I never get the graph, it just runs out of time eventually, and the printer stops at the last position of the level sequence. I hope someone can help me out! Thanx!

When I use these commands:
G28
M155 S30
@BEDLEVELVISUALIZER
G29 T
M155 S3

the Terminal shows:

Send: G28
Recv:  T:19.46 /0.00 B:19.65 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv:  T:19.69 /0.00 B:19.06 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:19.46 /0.00 B:19.80 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv:  T:19.20 /0.00 B:19.26 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:18.93 /0.00 B:19.45 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv:  T:19.87 /0.00 B:19.69 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:19.42 /0.00 B:19.57 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv:  T:19.42 /0.00 B:19.69 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:19.02 /0.00 B:19.30 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:19.15 /0.00 B:19.37 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv:  T:19.37 /0.00 B:19.18 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:19.69 /0.00 B:19.26 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv:  T:19.51 /0.00 B:19.41 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:19.64 /0.00 B:19.02 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv: X:218.00 Y:252.00 Z:15.60 E:0.00 Count X:17440 Y:20160 Z:6240
Recv: ok
Send: M155 S30
Recv: ok
Send: G29 T
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv:  T:19.02 /0.00 B:19.49 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv:  T:19.42 /0.00 B:19.18 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv:  T:19.24 /0.00 B:19.02 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: Bilinear Leveling Grid:
Recv:       0      1      2
Recv:  0 -0.157 -0.295 -0.092
Recv:  1 +0.108 -0.002 +0.135
Recv:  2 +0.255 +0.108 +0.113
Recv: 
Recv: X:433.00 Y:467.00 Z:15.49 E:0.00 Count X:34640 Y:37360 Z:6195
Recv: ok
Send: M155 S3
Recv: ok
Recv:  T:19.46 /0.00 B:19.30 /0.00 @:0 B@:0
Recv:  T:18.84 /0.00 B:19.37 /0.00 @:0 B@:0
Recv:  T:19.24 /0.00 B:19.34 /0.00 @:0 B@:0
Recv:  T:19.64 /0.00 B:19.34 /0.00 @:0 B@:0
Recv:  T:19.24 /0.00 B:18.91 /0.00 @:0 B@:0
Recv:  T:19.29 /0.00 B:19.49 /0.00 @:0 B@:0
Recv:  T:19.42 /0.00 B:19.37 /0.00 @:0 B@:0

jneilliii, 2020-03-24:
Quick question. After setting the timeout value did you restart octoprint? Others in this thread have mentioned that was necessary. If not try that. If it's still a problem, please follow instructions in that thread for getting the necessary log files. 

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/issues/195

alexsvers, 2020-03-24:
Yup, just bredde to restart octoprint :)
Thank you for the quick reply!!

Best

Alexander Sverstad

alexsvers, 2020-03-24:
jneilliii told me to try restarting octoprint first. I did that, and problem solved :)

closed 2020-03-24 by jneilliii