jneilliii/OctoPrint-BedLevelVisualizer #208: Bed Visualizer causes high network load

AbuMaia, 2020-03-25:
When I have this plugin enabled, there is a constant 2Mbps+ data flow on my network, which goes away only if I close the OctoPrint tab, or if I disable Bed Visualizer. I suspect this plugin is somehow keeping the webcam stream open constantly, whether the Control page is open or not. "Show webcam while processing" is disabled.

Bed Visualizer 0.1.12
OctoPrint 1.4.0 running on OctoPi 0.15.1

jneilliii, 2020-03-25:
The new webcam feature is probably too blame here. I didn't do any form of turning on/off the image when I integrated that feature. Will look into fixing this in next release.

jneilliii, 2020-03-26:
Just pushed a commit to the 0.1.13 branch that should resolve this issue. You can test by using the URL below in Plugin Manager until I've got more changes to officially release.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.13.zip

AbuMaia, 2020-03-27:
That works. Thanks for the quick fix. :)

AbuMaia, 2020-03-30:
Ok, the fix stopped the plugin from constantly running the video stream, until Update Mesh Now is pressed with the camera enabled during processing. Once probing has finished and the visualization has been updated, the camera stream is kept open, now showing as 3.5Mbps on my network. Refreshing the OctoPrint page stops it, though.

jneilliii, 2020-03-30:
I'll take a look, but that doesn't make sense to me. The img src attribute is set back to blank when processing is finished if I remember correctly.

jneilliii, 2020-04-01:
@AbuMaia how are you seeing this network utilization? I'm trying to figure out how to debug/reproduce this to see where it's happening. The once the processing of the mesh collection is completed and the graph is shown, the img displaying from the Bed Visualizer is completely removed from the UI because of the knockout if binding I'm using.

[image]

What browser are you using?

@foosel, do you know if knockout bindings need to be disposed in some way when an element is removed from DOM to also remove it from memory? That's the only other thing I can think of that would be causing this.

AbuMaia, 2020-04-02:
I use Firefox 74 on Ubuntu 18.04.04. And I see the network load with a taskbar netspeed indicator:
[image] The down speed drops to Kbps only after I refresh the OctoPrint page after updating the mesh through the Bed Visualizer plugin with the camera active.

foosel, 2020-04-02:
> @foosel, do you know if knockout bindings need to be disposed in some way when an element is removed from DOM to also remove it from memory? That's the only other thing I can think of that would be causing this.

Uff, sorry, no clue. I found this though: http://www.knockmeout.net/2014/10/knockout-cleaning-up.html

jneilliii, 2020-04-02:
> taskbar netspeed indicator.

@AbuMaia, this must be an ubuntu specific thing? I'm unable to find any add-ons to get this on my firefox in windows.

> I found this though: http://www.knockmeout.net/2014/10/knockout-cleaning-up.html

Thanks @foosel, I'll take a look.

AbuMaia, 2020-04-03:
The taskbar netspeed indicator? Yes, it's an Ubuntu thing. But there's a program called Conky that can be set up to display the same information on the desktop.

jneilliii, 2020-04-03:
Ok, thanks. I found another one called Display Info that seems to give me the info too. So in windows on Google Chrome I do see the increase to 2.5+Mbps, but as soon as the graph displays it drops down to "normal" traffic on the network. The below shows where I've done the process twice.

[image]

This leads me to believe it's either an issue with Firefox and/or Ubuntu combination. Going to do the same tests in Firefox now.

jneilliii, 2020-04-03:
I can confirm this appears to be a Firefox issue. Just did a test in it and as soon as the graph displays sure enough the network traffic remains.

[image]

I'll see if there is any way around this.

jneilliii, 2020-04-03:
Going to have to look into how @foosel is doing this because it doesn't happen on the control tab. It seems that with the control tab after you switch away from the tab firefox gets a final 200 return status to the img request, but with my plugin when the img is removed from DOM and the graph is shown the image never gets a finalized 200 status response.

[image]

closed 2020-07-26 by jneilliii

calania, 2020-08-02:
Was this ever fixed or why has the issue been closed? Have been holding off updating to newer versions of the plugin due to people reporting this issue.

jneilliii, 2020-08-02:
I think I fixed it in the last version update for good. If not, please report back.

calania, 2020-08-07:
Okay, thank you will try it!

PR #210 (2020-03-28): 0.1.13
Made webcam image not load when that option isn't enabled and only while processing, resolves #208 
Screw based adjustment degrees, thanks to @LMS0815, resolves #142