jneilliii/OctoPrint-BedLevelVisualizer #209: Copy/paste mesh results into editor

amooz, 2020-03-26:
It would be great if we could copy/paste the mesh results from the terminal logs into a text editor and update the visualization that way.  If there's a problem scraping the info from the terminal logs, we would still be able to update the visualization (just with some extra steps.)

jneilliii, 2020-03-26:
You run into some of the same issues relative to parsing because of so many of the different variants of the report from the different firmware and versions of firmware.

amooz, 2020-03-26:
As a first-pass, I would be OK if this were an advanced feature and you only accepted a CSV.  It would require some parsing on our end which is OK.

tablatronix, 2021-02-14:
How can we just get mesh data without probing to update it?

jneilliii, 2021-02-14:
I assume you mean with something like manual mesh leveling?

tablatronix, 2021-02-14:
Like G29 S0 to get mesh from printer not recreate another one.

jneilliii, 2021-02-15:
Yeah, that's pefect. You can enter that in the plugin's setting (I use `M420 V` personally with UBL because I don't probe every print). You can even use custom controls to achieve multiple slots, additional discussion [here](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/issues/375).

tablatronix, 2021-02-15:
Ahh I did not realize custom commands could link to the grapher nice gonna test now