jneilliii/OctoPrint-BedLevelVisualizer #211: Been using for an age, updated yesterday, BOOM. Results ignored!?!?

TheGadge, 2020-03-27:
I've used this just fine since it started out, -until yesterday!

I do an update of bed and it's completely ignored??
-The visualisation shows from ages ago when last used. That's all I can get!
Any advice please, many thanks.

[image]

dsiguero, 2020-03-27:
I'm also having this issue since the last update.

jneilliii, 2020-03-27:
More than likely your timeout value is too low and needs to be increased. Increase the value to a number that is past the amount of time it takes for your leveling process to complete and then restart OctoPrint (very important). This has worked for several people in the thread over here.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/issues/195

closed 2020-03-27 by jneilliii

AbuMaia, 2020-03-29:
I just installed the new release too, and I'm seeing this issue. My printer double-taps 6 points, so it takes a while. I've had my timeout value set to cover how long this takes (310s) since before this update, and I am also seeing an old probing result, not the latest.

MasterFuba, 2020-04-02:
I have the same issue on my Ender3, last update was on 29.3. and i used it excessive then. Now couple of days later Bed Visualizer seems to be in quarantine. After reinstall i see the "you must be logged in" info screen but i can't do anything there. Just wait until new update arrives? 
Also used commands shown in https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/issues/195
but no success - seems like version is broken to Ender machines or general.

Skiddz, 2020-04-10:
I'm in the same boat.  Last ABL was 3/19 and since the last update, the mesh won't update.  
I use the BLTouch plugin, heat the bed and nozzle, home everything, then use the plugin's buttons to run the probe and save.

Visualizer Gcode is M420 V then M420V1, then M501.  Mesh would update in about a minute.
Now it tell me I have to be logged in and nothing happens.  

I uninstalled the plugin and cleaned the data, then did a power off shut down and restart.  Same deal.  I tried using just the visualizer to generate the mesh with G28, then G29 T and the printer does probe, but the visualizer never updates.  

At this point, I can't even generate a log file because nothing happens.  I let it grind all night and still no joy.  

CR-10 Mini, TH3D Unified Firmware, (TH3D U1.R2.A3) BLTouch 3.1, Octoprint 1.4.0 on a 4G Raspberry Pi 4B

Using the BLTouch Plugin to generate the mesh data, what would the proper GCode be to update the visualizer?

jneilliii, 2020-04-10:
I think the clear data option messes things up to be honest. If you manually remove all the settings for the plugin in config.yaml it could potentially resolve the issue.

Skiddz, 2020-04-10:
I'm up and running on 1.1.11 right now and mesh updates are literally taking seconds.  I'll look at updating once my next few jobs are printed.  Thx!

electricJP, 2020-04-14:
This issue has been closed but I don't see any evidence it is solved. I do noticed the same issue this morning after trying to update my mesh (last update was 2020/02/20). Visualizer will just fail with a timeout error, even if I see the G29 completed in terminal.

jneilliii, 2020-04-14:
This was a duplicate issue, but your problem to me seems like you need to increase your processing timeout value to something larger. What is it set at now?

electricJP, 2020-04-14:
It was working fine at 60s. I increased up to 120s with no luck. I can see in terminal that G29 is completing correctly. It looks like plugin does not catch process end. I use the standard commands:
G28	
M155 S30	
@BEDLEVELVISUALIZER	
G29 T	
M155 S3

jneilliii, 2020-04-14:
I upped the default to 1800 for new users this was causing so much pain. Set it to 1800 and restart OctoPrint and I bet it works.

electricJP, 2020-04-14:
Yup. it did. Thank you!