jneilliii/OctoPrint-BedLevelVisualizer #212: Can't send data to plot.ly anymore

IroCorp, 2020-03-29:
I can't update data to plot.ly anymore. I just have the option to save to PNG.
Maybe because of the change of the URL?
ttps://community.plotly.com/t/url-change-for-chart-studio-cloud-https-chart-studio-plot-ly/33557

Thank you

jneilliii, 2020-03-29:
That must have been the button that was removed when setting `displayLogo` to false [here](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/blob/1e800d20799e6ab7cd214c8e8f5dab8253298696/octoprint_bedlevelvisualizer/static/js/bedlevelvisualizer.js#L215). I didn't realize anyone used that feature, will bring it back in next update.

jneilliii, 2020-04-02:
So it looks like plotly changed their config options a little and I had to manually add the button back in with a new flag. If you can test and verify for me that this works properly I can prepare it for an official release. Install the update using the URL below in plugin manager and make sure to force refresh your browser (shift+f5 or ctrl+f5 depending on your browser) after install.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.14.zip

IroCorp, 2020-04-10:
Hello,
Great, it works ! Thank you :)

closed 2020-04-11 by IroCorp