jneilliii/OctoPrint-BedLevelVisualizer #213: Missing system dependency causes BedVisualizer installation to fail silently

avolkov, 2020-04-02:
I'm running BedVisualizer with Octoprint 0.1.4 on Python 3.

One of the libraries on which numpy depends, is not present in vanilla OctoPi install and it causes BedVisualizer to silently fail installation. It seems like installation is completed but the plugin doesn't show up as installed anywhere -- this is very confusing.

 I only found this looking though the logs.

The missing package is libatlas3-base

I'm not sure what's the best way, but could you add somewhere prominent in the readme that this package needs to be install through apt in order for the plugin to work properly?

Here's plugin_pluginmanager_console.log
```
2020-04-02 01:54:15,987 > Building wheel for Bed-Visualizer (setup.py): finished with status 'done'
    2020-04-02 01:54:15,989 > Created wheel for Bed-Visualizer: filename=Bed_Visualizer-0.1.12-py3-none-any.whl size=2516034 sha256=4ee31fc02087b26ed3b5f85a13addbc6989c28e94be1a7bab1351c619532dfc6
    2020-04-02 01:54:15,989 > Stored in directory: /tmp/pip-ephem-wheel-cache-goviqro5/wheels/93/c7/cf/2acf99f698a9f1c452a6a9c3747c06b0f46ca92689388ccdd7
    2020-04-02 01:54:15,990 > Successfully built Bed-Visualizer
    2020-04-02 01:54:15,991 > Installing collected packages: Bed-Visualizer
    2020-04-02 01:54:15,991 > Attempting uninstall: Bed-Visualizer
    2020-04-02 01:54:15,992 > Found existing installation: Bed-Visualizer 0.1.13
    2020-04-02 01:54:15,992 > Uninstalling Bed-Visualizer-0.1.13:
    2020-04-02 01:54:15,993 > Successfully uninstalled Bed-Visualizer-0.1.13
    2020-04-02 01:54:16,998 > Successfully installed Bed-Visualizer-0.1.12
```

Here's octoprint log

```
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint_bedlevelvisualizer/__init__.py", line 6, in <module>
    import numpy as np
  File "/home/pi/OctoPrint/venv/lib/python3.7/site-packages/numpy/__init__.py", line 142, in <module>
    from . import core
  File "/home/pi/OctoPrint/venv/lib/python3.7/site-packages/numpy/core/__init__.py", line 54, in <module>
    raise ImportError(msg)
    ImportError:

    IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

    Importing the numpy c-extensions failed.
    - Try uninstalling and reinstalling numpy.
    - If you have already done that, then:
    1. Check that you expected to use Python3.7 from "/home/pi/OctoPrint/venv/bin/python3",
     and that you have no directories in your PATH or PYTHONPATH that can
     interfere with the Python and numpy version "1.18.2" you're trying to use.
    2. If (1) looks fine, you can open a new issue at
     https://github.com/numpy/numpy/issues.  Please include details on:
     - how you installed Python
     - how you installed numpy
     - your operating system
     - whether or not you have multiple versions of Python installed
     - if you built from source, your compiler versions and ideally a build log

    - If you're working with a numpy git repository, try `git clean -xdf`
  (removes all files not under version control) and rebuild numpy.

    Note: this error has many possible causes, so please don't comment on
    an existing issue about this - open a new one instead.

    Original error was: libf77blas.so.3: cannot open shared object file: No such file or directory

```

jneilliii, 2020-04-02:
So that's more of an issue with a numpy dependency rather than this plugin's dependency. numpy should declare that as a dependency on it's repository/build procedure I would think. Just curious if you tried the recommended step of uninstalling and reinstalling numpy manually via SSH, or is that how you found the missing dependency?

avolkov, 2020-04-02:
This happened because I upgraded the system from a really old version of raspbian Jessie to Buster. One of the dependencies got removed  along the way. I remember having issues with the dependencies a few years ago, but I was able to resolve them because I knew where to look.

This time I thought I was having issue of the plugin not being installed in the correct virtual environment, and there was no indication that anything was wrong during installation. I also tried a brand new image Octoprint and I experienced the same behaviour with BedLevelVisualizer. I just didn't know where to look so I started looking in octoprint log.

I think it would be a lot easier on first-time users who don't know about numpy is to ask in readme to check for numpy system dependencies. Something along the lines of -- 

Do you have numpy system dependencies installed, such as libatlas3-base? BedLevelVIsualizer plugin will not be available in Octoprint if numpy dependencies are missing. 
To install numpy in raspbian, please follow installation instructions for ubuntu & debian at scipy.org  -- https://www.scipy.org/install.html#install-system-wide-via-a-package-manager

jneilliii, 2020-04-02:
Thanks, I think an update to readme is a good start here. Just to confirm, this was a python 3 environment you added to an octopi image (which version) or a standard raspbian build on your pi? Which instructions did you follow for creating your python 3 virtual environment for OctoPrint?

avolkov, 2020-04-02:
I've tried both. I started with system upgrade where I updated OctoPrint in new python 3 virtual environment. Because I updated operating system as wall, one of the dependencies for numpy was removed and this it broke my bedvisualizer config.

I also tried installing fresh octopi 0.17.0 image that comes with OctoPrint  with Python 2. It also requires manually installing numpy dependencies and because  I forgot the step on installing numpy system dependencies, neither of these systems worked out of the box

Unfortunately for octopi images, pip is not installed by default, so anyone not familiar with python packages trying working with octopi plugins is going to be confused ( but that's a separate issue).

Maybe I should make a video showing how to get from brand new image of octopi to the point of bed visualizer working. That might clear up some confusing.

jneilliii, 2020-04-02:
> Unfortunately for octopi images, pip is not installed by default

That's not true, but it also doesn't have python 3, that has to be added manually. Which is why I asked what procedure you were using. I'll try to reproduce by doing a new image and follow the steps outlined on docs.octoprint.org for setting up python 3 on it ans report back.

jneilliii, 2020-08-04:
I may have resolved this issue by pinning the numpy version to successfully built versions available on [piwheels](https://www.piwheels.org/project/numpy/). This way in Python 3 environments it should install the pre-compiled binaries for numpy. Initial tests show that by doing this install time has been reduced to under a minute as well, which is awesome. If you have the ability to test this again, you can install the currect dev version by using the URL below in Plugin Manager > Get More > ...from URL and clicking the Install button.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.15.zip

kpes, 2020-08-07:
> I may have resolved this issue by pinning the numpy version to successfully built versions available on [piwheels](https://www.piwheels.org/project/numpy/). This way in Python 3 environments it [...]

Just thought I'd mention, I was having this issue, and tried the 0.1.15 dev version and everything seems better for me.

jneilliii, 2020-08-08:
Thanks for confirming @kpes

closed 2020-08-18 by jneilliii