jneilliii/OctoPrint-BedLevelVisualizer #214: Framebuffer incomplete attachment

marekmeli, 2020-04-02:
Hello, I struggle to find any answer on what's causing this error to appear:
[image]

**When it happens?**
After every single bed probing. I'm running this starting gcode:

`G28; home XYZ

G29; BLTouch mesh generation

@BEDLEVELVISUALIZER

BED_MESH_OUTPUT`

**Why it happens?**
I really have no clue, other than this can be some OpenGL issue rather than marlin/octoprint issue.

**When it all started?**
Since the first time I installed this plugin. I even tried a clean install of the whole system, still the same issue.

Is there something else I can describe to help solve this issue? I am very surprised that I'm the only one facing this problem, so I believe the error is somewhere on client side.

Thank you!

jneilliii, 2020-04-02:
That definitely looks like a webgl related issue, which is tied to your video card and what it supports.

marekmeli, 2020-04-02:
Yes, this will be the case. I am running a really old machine with outdated drivers.

Thank you for your help, the response was really quick!

closed 2020-04-02 by marekmeli