jneilliii/OctoPrint-BedLevelVisualizer #215: Support for Brave Browser.

DotNetRob, 2020-04-02:
When opening the Bed Visualizer in Brave I get a "WebGL is not supported by your browser - visit https://get.webgl.org for more info" message.  But if I go to the url https://get.webgl.org, it works just fine.

Runs just fine in chrome.

jneilliii, 2020-04-02:
What about if you go to the plotly examples page linked below, does it load for you?

https://plotly.com/javascript/3d-surface-plots/

DotNetRob, 2020-04-02:
Yep work fine in brave.

[image]

jneilliii, 2020-04-02:
Looks like the version of plotly library I have in the plugin is one version before better Brave browser detection based on the release notes [here](https://github.com/plotly/plotly.js/releases/tag/v1.52.3). I'll see about updating plotly library in a test branch and get you to verify for me.

DotNetRob, 2020-04-02:
sounds good.

jneilliii, 2020-04-02:
Ok, I've updated the library and pushed to a new branch. You can test by using the URL below in plugin manager. Make sure to force a refresh in the browser (I assume it's similar to chrome shift+f5) once installed.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.14.zip

DotNetRob, 2020-04-02:
Installed over the existing plugin then restarted octo and didn't work, so did a uninstall and clean up data.  Then installed it fresh, restarted octo but still isn't displaying.   
[image]

jneilliii, 2020-04-02:
Did you force a refresh of your browser to bypass local cache?

DotNetRob, 2020-04-02:
I did, even went into debug/dev mode(F12) and did a Empty cache and hard reload, but still no luck.  Was going to try in private mode to ensure of no caching but can't login to octoprint when in private mode in brave.

jneilliii, 2020-04-02:
Ok thanks. Not sure why this is not working then in the plugin but does on the examples page. They are in theory using the same version of the library, and I don't do anything weird to not allow it to work in Brave. I'll dig deeper into the plotly issue tracker and see if there is any additional information.

jneilliii, 2020-04-02:
Not super familiar with Brave, except that it's "more secure", but found a post that disabling shields with webgl may help?

> I use Brave as my main browser and I often encounter issues with WebGL plots. I usually solve them by turning off the Brave Shields

DotNetRob, 2020-04-02:
Will have to look thru the brave shield settings and see when I can find, I am still pretty new to brave myself.  But confirmed it not the extension, its a brave issue.  I went to the codepen link on https://plotly.com/javascript/3d-surface-plots/  and it doesn't work on codepen either.  

https://codepen.io/pen/?&editable=true

[image]

jneilliii, 2020-04-02:
It's actually an issue with plotly being able to detect the Brave browser properly I think. Saw two posts on it's issue tracker for Brave. One was closed and one was still open, which is where I got that quote from. You may want to chime in on that issue posted [here](https://github.com/plotly/plotly.js/issues/4577).

DotNetRob, 2020-04-02:
Found it.  In brave shields there is a setting for "Device recognition"  defaults to "Only block cross-site device recognition attempts", if set it to "Allow all" it works.

jneilliii, 2020-04-02:
Great, glad you were able to find it. Not sure if there is an option to "trust" a specific address to allow it only on the octoprint site or not, but if that is an option you might want to set it that way to keep it blocking other sites. That is the whole point of Brave, more secure/anonymous browser.

closed 2020-04-02 by jneilliii