jneilliii/OctoPrint-BedLevelVisualizer #216: undefined symbol: PyFPE_jbuf

Enjoy-Combi, 2020-04-03:
Hi,

I have upgraded the Pi from Stretch to Buster.
The plugin version 0.1.13 is not working since then.
Mainly two issues reported in the log:

octoprint.plugin.core - ERROR - Error loading plugin bedlevelvisualizer
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 967, in _import_plugin
    module = _load_module(module_name, spec)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 71, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_bedlevelvisualizer/__init__.py", line 6, in <module>
    import numpy as np
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/numpy/__init__.py", line 150, in <module>
    from . import random
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/numpy/random/__init__.py", line 143, in <module>
    from .mtrand import *
ImportError: /home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/numpy/random/mtrand.so: undefined symbol: PyFPE_jbuf

and 

octoprint.plugin.core - ERROR - Error loading plugin bedlevelvisualizer
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 967, in _import_plugin
    module = _load_module(module_name, spec)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 71, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_bedlevelvisualizer/__init__.py", line 6, in <module>
    import numpy as np
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/numpy/__init__.py", line 140, in <module>
    from . import _distributor_init
ImportError: cannot import name _distributor_init

I think numphy has to be compiled again or install again.
How to force this ?

Enjoy-Combi, 2020-04-03:
Testing:
sudo systemctl stop octoprint
cd OctoPrint
virtualenv venv
source venv/bin/activate
pip install pip --upgrade
pip install pip --upgrade --force-reinstall numpy

Will see the result when compilation finished

jneilliii, 2020-04-03:
May want to add --no-cache-dir to that pip command line as well in case you had a bad download. Other reports have indicated this failing because of missing dependencies in python 3 that numpy either doesn't reference correctly itself or doesn't get installed. Not sure if it might be the same, but just in case check here. 

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/issues/213

Enjoy-Combi, 2020-04-03:
It is fixed :-)

jneilliii, 2020-04-03:
Cool.

closed 2020-04-03 by jneilliii