jneilliii/OctoPrint-BedLevelVisualizer #217: Anycubic Chiron moving up in Z every probe

whospatlaw, 2020-04-03:
Chiron probes 25 spots but moves up after each one.  See attached for settings.

[image]
[image]
[image]

Send: N37 G28*39
Recv: X:-10.00 Y:0.00 Z:0.00 E:0.00 Count X: -800 Y:0 Z:0
Recv: ok
Send: N38 G29 T*93
Recv: G29 Auto Bed Leveling
Recv: Bilinear Leveling Grid:
Recv:       0     1     2     3     4
Recv:  0 +27.99 +35.99 +43.98 +51.98 +59.97
Recv:  1 +99.95 +91.95 +83.96 +75.96 +67.97
Recv:  2 +107.94 +115.94 +123.93 +131.93 +139.92
Recv:  3 +179.90 +171.90 +163.91 +155.91 +147.92
Recv:  4 +187.89 +195.89 +203.88 +211.88 +219.87
Recv: 
Recv: X:384.00 Y:393.00 Z:16.85 E:0.00 Count X: 30720 Y:39300 Z:94689
Recv: ok
Send: N39 M105*29
Recv: ok T:23.0 /0.0 B:23.3 /0.0 @:0 B@:0

whospatlaw, 2020-04-04:
Lol dangit.  Now my bed level is set to the values above...
Back to leveling again.

jneilliii, 2020-04-04:
The plugin doesn't do anything out of the ordinary when it comes to the commands, so not sure why that would be going up for you as long as the commands you entered are what does a standard probing process for your printer. You may definitely want to increase your Processing Timeout value to something higher if you're probing 25 points though. I'm sure that doesn't complete in 5 minutes and is why you got the timeout message.

whospatlaw, 2020-04-04:
Hmm weird.  Is there anywhere in the marlin file I can find what the command is?  I've uploaded the master file I use on my printer.
[ANYCUBIC_CHIRON_V1.3.0-master1.zip](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/4430491/ANYCUBIC_CHIRON_V1.3.0-master1.zip)

jneilliii, 2020-04-04:
Ok, looks like it's based on an old Marlin fork, `1.1.0-RC8` to be specific. The commands you have set should be fine, maybe try taking out the T and using just G29 instead?  What command do you run to do a probing if you were to do it manually via the terminal tab?

whospatlaw, 2020-04-04:
Its weird.  They require you to run a print called auto_leveling which runs a 25 point auto probe.  I attached the stl.zip
[Auto_Leveling.zip](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/4430529/Auto_Leveling.zip)
.

whospatlaw, 2020-04-04:
Thank you for your help as well.  You are a kind individual.  I do like learning about it.

whospatlaw, 2020-04-04:
Here is a github readme that explains the function.
https://github.com/ANYCUBIC-3D/ANYCUBIC_CHIRON/blob/master/README.md

jneilliii, 2020-04-04:
So if you go to OctoPrint's terminal tab and type the following commands one at a time, does it probe the bed when you send the last line?

```
M140 S0
M104 S0
M1001
```

whospatlaw, 2020-04-04:
They reset the on screen leveling to 0 essentially and input a z offset of -14.  You then hit probe and it starts the auto leveling. I assume when you hit probe it does a G29 and starts the auto level.  Maybe I should do that and watch the terminal to see if it shows what command is sent?
[image]
[image]

jneilliii, 2020-04-04:
Ok, so it's basically instructing the panel to be usable to PROBE. Yeah, need to find out what the underlying gcode that the touch screen is sending to the printer. When you press it does anything show up in the terminal tab that corresponds to what commands are being run?

jneilliii, 2020-04-04:
Yeah, so looking at the Configuration.h of their marlin repo on github it looks like it's configured for `AUTO_BED_LEVELING_BILINEAR`, so if you run those commands above and then follow it up with `G29` does it probe the bed properly?

whospatlaw, 2020-04-04:
Ok.  So I tried it this morning and here is what I see in terminal.  I put comments in the lines below showing when I did what.

Recv: echo:Now fresh file: auto_l~1.gco     "<-- Opened the PRINT window on Chiron LCD to access Auto_Leveling STL"
Recv: File opened: auto_l~1.gco Size: 25
Recv: File selected
Send: N12 M27*37
Recv: SD printing byte 0/25
Recv: ok
Send: N13 M27*36
Recv: SD printing byte 0/25
Recv: ok
Send: N14 M105*18
Recv: ok T:21.5 /0.0 B:22.3 /0.0 @:0 B@:0
Send: N15 M27*34
Recv: SD printing byte 0/25
Recv: ok
Send: N16 M27*33
Recv: SD printing byte 0/25
Recv: ok
Send: N17 M27*32
Recv: SD printing byte 0/25
Recv: ok
Send: N18 M27*47
Recv: SD printing byte 0/25
Recv: ok
Send: N19 M27*46
Recv: SD printing byte 0/25
Recv: ok
Send: N20 M105*21
Recv: ok T:21.3 /0.0 B:21.9 /0.0 @:0 B@:0
Send: N21 M27*37
Recv: SD printing byte 0/25
Recv: ok
Send: N22 M27*38
Recv: SD printing byte 0/25
Recv: ok
Send: N23 M27*39
Recv: SD printing byte 0/25
Recv: ok
Send: N24 M27*32
Recv: SD printing byte 0/25
Recv: ok
Send: N25 M27*33
Recv: SD printing byte 0/25
Recv: ok
Send: N26 M105*19
Recv: ok T:21.5 /0.0 B:22.5 /0.0 @:0 B@:0
Send: N27 M27*35
Recv: SD printing byte 0/25
Recv: ok
Send: N28 M27*44
Recv: SD printing byte 0/25
Recv: ok
Send: N29 M27*45
Recv: SD printing byte 0/25
Recv: ok
Recv: Done printing file     "<--Selected PRINT on file and completes immediately"
Recv: echo:enqueueing "M84 X Y Z E"
Send: N30 M27*37
Recv: Done, Auto Leveling was actived!SD printing byte 25/25  
Recv: ok
Send: N31 M27*36
Recv: SD printing byte 25/25
Recv: ok
Send: N32 M105*22
Recv: ok T:21.2 /0.0 B:21.9 /0.0 @:0 B@:0
Send: N33 M27*38
Recv: SD printing byte 25/25
Recv: ok
Send: N34 M27*33
Recv: SD printing byte 25/25
Recv: ok
Send: N35 M27*32
Recv: SD printing byte 25/25
Recv: ok
Send: N36 M27*35
Recv: SD printing byte 25/25
Recv: ok
Send: N37 M27*34
Recv: SD printing byte 25/25
Recv: ok
Send: N38 M105*28
Recv: ok T:21.5 /0.0 B:22.5 /0.0 @:0 B@:0
Send: N39 M27*44
Recv: SD printing byte 25/25
Recv: ok
Send: N40 M27*34
Recv: SD printing byte 25/25
Recv: ok
Send: N41 M27*35
Recv: SD printing byte 25/25
Recv: ok
Send: N42 M27*32
Recv: SD printing byte 25/25
Recv: ok
Send: N43 M27*33
Recv: SD printing byte 25/25
Recv: ok
Send: N44 M105*23
Recv: ok T:21.2 /0.0 B:21.9 /0.0 @:0 B@:0
Recv: echo:enqueueing "G28"     "<--I think this is where I selected PROBE"
Recv: echo:enqueueing "G29"
Send: N45 M27*39
Recv: X:-10.00 Y:0.00 Z:0.00 E:0.00 Count X: -800 Y:0 Z:0
Recv: G29 Auto Bed Leveling   "<--About half way through the leveling it errors out in terminal but continues finishing auto leveling.  After finishing it updates the level screen on the Chiron."
Communication timeout while idle, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N46 M27*36
Communication timeout while idle, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N47 M105*20
No response from printer after 3 consecutive communication timeouts, considering it dead. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Changing monitoring state from "Operational" to "Offline (Error: Too many consecutive timeouts, printer still connected and alive?)"
Connection closed, closing down monitor

When I go to the terminal and perform
M140 S0
M104 S0
M1001
G29

it does the same thing which we started with.  Going to each spot and raising.

jneilliii, 2020-04-04:
Try these in terminal command based on the above.
```
M140 S0
M104 S0
M1001
G28
G29
```

jneilliii, 2020-04-04:
If that doesn't work, and you really want this plugin I think you're going to have to look into what options are available for updating your firmware to a more "stock" Marlin version.

whospatlaw, 2020-04-04:
Same result.  Thank you for your help and quick replies!

Its odd because in one of your other threads someone using a Chiron got it to work.
https://community.octoprint.org/t/bed-level-visualizer/1383/77

[image]

jneilliii, 2020-04-04:
What we don't know is if @Descapper is using a non-stock firmware or not. Maybe he can reply here and let us know.

jneilliii, 2020-04-04:
What we don't know is if @Decapper is using a non-stock firmware or not. Maybe he can reply here and let us know.

Deapper, 2020-04-06:
Hi, what my problem turned out to be is that my sensor got pushed into my bed and the button on the sensor got stuck. As the printer moves the head up each time it touches the plate, hence every time it checked for the sensor it would constantly move up

closed 2020-04-27 by stale[bot]