jneilliii/OctoPrint-BedLevelVisualizer #218: Error: gl-fbo: Parameters are too large for FBO

Silversurfer79, 2020-04-04:
Any idea whats wrong here? 

Was working fine before the update tot he new version of BLV and OP1.4.

[image]

Thanks

Silversurfer79, 2020-04-04:
`G28
M155 S30
@BEDLEVELVISUALIZER
G29 T
M155 S3`

Silversurfer79, 2020-04-04:
No this?
[image]

jneilliii, 2020-04-04:
Please enable debug logging and restart octoprint. Press update mesh button and then download the plugin_bedlevelvisualizer_debug.log file from the logging section of octoprint's settings. Then drag and drop it into a comment here.

Silversurfer79, 2020-04-04:
[plugin_bedlevelvisualizer_debug.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/4432296/plugin_bedlevelvisualizer_debug.log)

jneilliii, 2020-04-04:
No problem. what is returned in the log appears to show "normal" results, so the graphing library is probably flaking out for some reason. This is what is being graphed for the z offsets.

[image]

Your view does look odd to me though, assuming you used the mouse to change the angle. What does it look like when it first loads? I took that debug log and virtually ran it in my OctoPrint and got this.

[image]

Silversurfer79, 2020-04-04:
Like this,
[image]

Silversurfer79, 2020-04-04:
No dice
[image]

jneilliii, 2020-04-05:
Something is getting messed up, because the dimensions of the graph in x and y axes are in the thousands.

Silversurfer79, 2020-04-06:
Any ideas?

jneilliii, 2020-04-06:
Try this. Go into settings and un-check the box to store mesh and see if it makes a difference. It looks like your settings have your bed configured as 220 x 220 x 250 in OctoPrint, is that correct?

Silversurfer79, 2020-04-06:
I found that if i switched between Bed Visualiser and Resource Monitor or refreshded the page it load like this, though I still got the same Error message.

Silversurfer79, 2020-04-06:
> 
> 
> Try this. Go into settings and un-check the box to store mesh and see if it makes a difference. It looks like your settings have your bed configured as 220 x 220 x 250 in OctoPrint, is that correct?

I have ender 3, so that is correct. Please remember this has been working up to this point (update to the new version ot BV and OP)

Silversurfer79, 2020-04-06:
> 
> 
> Try this. Go into settings and un-check the box to store mesh and see if it makes a difference. It looks like your settings have your bed configured as 220 x 220 x 250 in OctoPrint, is that correct?

Well now...

[image]

jneilliii, 2020-04-06:
Oh wait, Resource Monitor is a good point. There could be a conflicting plotly version in that plugin. Let me check.

jneilliii, 2020-04-06:
Nope, that's not it. Resource Monitor is using the built in graphing library of OctoPrint (flot). Not sure why this is happening, but one user on another issue mentioned that the graph would work once after saving settings, but then stop working. Would be interested to know if it's the same for you as well.

Silversurfer79, 2020-04-06:
Its also now auto leveling when i click on the tab

jneilliii, 2020-04-06:
Yeah, that's because we turned off the store mesh option, that is expected. In order for it to graph when that option is disabled it has to probe the bed to get the report. If it's working now with that option disabled, try enabling the option back and let's see if it sticks or reverts to a problematic situation.

Silversurfer79, 2020-04-06:
I also noticed that there is some stange behaviour happeining and I have zoomed out so that you can see, its when i hover over the graph. highlighted below,

[image]

jneilliii, 2020-04-06:
That's because of your Themify customizations I think. The width is causing weird stuff to happen.

Silversurfer79, 2020-04-06:
Same issue
[image]

Silversurfer79, 2020-04-06:
If I click any other tab and click back and its fine, the erro message still sticks around (top right - its missng as i cancelled it)

jneilliii, 2020-04-07:
Yeah, like I mentioned, I think it's your themify customizations. Turn off Themify and let me know if this still happens. If that works then I might be able to help you resolve with the correct CSS selecors to limit the size of the containing graph to bring it back in to normal width instead of the wide set-up that is currently throwing it off.

Silversurfer79, 2020-04-07:
> Yeah, like I mentioned, I think it's your themify customizations. Turn off Themify and let me know if this still happens. If that works then I might be able to help you resolve with the correct CSS [...]

@jneilliii  Your on the money, seemed to be the issue, seems to be be something witht he over scaling of the UI. Any chance you could fix this? I love your plugin and would really love to use it without and issues.

Silversurfer79, 2020-04-07:
Well after re-enabling Thermify, the issue no long is presant... So Ill monitor this over the coming days and reprot back if I have any issues.

jneilliii, 2020-04-08:
If it does come back you could try setting the width of `div#bedlevelvisualizergraph` CSS selector to be 588px, which is what it appears to stretch to in the non-themed view.

Silversurfer79, 2020-04-08:
> be 588px, which is what it appears to stretch to in the non-themed view.

Where would I do this? Sorry for the doff question ;-)

jneilliii, 2020-04-08:
In themify advanced section... the same way you set the width to make it wider...

Silversurfer79, 2020-04-11:
Sorry, Im not following, please see my advanced settings
[image]

jneilliii, 2020-04-11:
Create a new custom style and set the selector to `div#bedlevelvisualizergraph` thre css-rule to `width` and the value to `588px`.

Silversurfer79, 2020-04-11:
No dice [image]

jneilliii, 2020-04-12:
No dot (.) at the beginning of the css rule.

Silversurfer79, 2020-04-14:
Same thing... Gutted

jneilliii, 2020-04-14:
Interesting, I'll have to test this. It should work. Might require an !important at the end of the value like this, but not sure.

```
588px !important
```

Silversurfer79, 2020-04-15:
Tried that, it lools like this now. Im not sure if its related...
[image]

jneilliii, 2020-04-15:
Interesting, looks like it folded over on itself. Not sure if that's related to the sizing or not.

Silversurfer79, 2020-04-16:
Its to unrelaible to use in this state, I will have to find another solution. Anyway to revert to the  preiviuos version which work perfectly with themify?

jneilliii, 2020-04-16:
Yes, go to the releases tab up top and use the url for the zip file in plugin manager for thre previous version.

Silversurfer79, 2020-04-17:
With BV 0.1.12 install, first off it never presents a table and worst of all it uses 50% of my nic, sending 50MB-60MB constantly and making my Pi run like a dog (CPU 25%) (data cleared before testing). So, I updated back to 0.1.13, above issues no longer appear, as I have upgraded to OctoPrint 1.4 I am now stuck in this limbo :-(

Silversurfer79, 2020-04-17:
Well now, after doing the downgrade 0.1.12 and clearing the data and cache and then doing the update (i guess the cache clearing maybe the fix) it seems to be responisive and alot faster and loads correctly. Watch this space, will report back. 
[image]

Silversurfer79, 2020-04-17:
Scratch that, on just clicking the tab after running previously, for the first time, the error appears again (see first post). Its likely something to do the excisitng cache mesh and sizing as you suggested.

closed 2020-05-09 by stale[bot]

PretzelVector, 2024-03-06:
For anyone new coming across this bug, the error can also be caused by Librewolf's "Resist Fingerprinting".

Thank you for this plugin.