jneilliii/OctoPrint-BedLevelVisualizer #219: Adjustment Angle Table Y ordering

luudvkeulen, 2020-04-07:
I am using the new Adjustment Angle tool. I found one thing that might be a bug.
Shouldn't the Y start at 220 and then go down to 0 instead of the way around?
It is like this now:
Y
0
110
220
X 0 110 220
Wouldn't it be better to have it like this?
Y
220
110
0
X 0 110 220
Because the Front/Bottom left corner is 0.0.

luudvkeulen, 2020-04-07:
[image]

jneilliii, 2020-04-07:
I believe there's am option in settings to flip that y axis. Those changes were added by another user.

luudvkeulen, 2020-04-07:
> I believe there's an option in settings to flip that y axis. Those changes were added by another user.

I tried that option but it doesn't flip that table. Only the values in it.

jneilliii, 2020-04-07:
Sorry, not the flip option in case you misinterpreted, but the option on the cureent mesh data section in settings.

[image]

luudvkeulen, 2020-04-09:
Thanks! I didn't see that button at first.

closed 2020-04-09 by luudvkeulen