jneilliii/OctoPrint-BedLevelVisualizer #220: Incorrect date/time showing

KenLucke, 2020-04-08:
When I do a bed level via the control box, or another plugin, then use Bed Visualizer to display it (using only a M420V), it uses the date/time of the display request rather than the actual date/time of the actual bed level.  I am not familiar enough with the data stored to know if stores the actual time of probing with the mesh.  If so, Bed Visualizer is not using it.  I can't recall for sure, but I thought it used to do so correctly before the update that shows the webcam during probing/analyzing. 

If that data is not stored.... nevermind.

jneilliii, 2020-04-08:
The timestamp that is stored has always been the time in which you pressed the Update Mesh button in the interface. I just checked and it doesn't appear there is any information returned via M420 V that does show a time stamp, so it wouldn't be possible to display it from there.

closed 2020-04-08 by jneilliii

KenLucke, 2020-04-09:
OK, thanks for checking, I must have misremembered.  I thought it might have been stored as a standard time coded integer.  Thanks for all the work you do!