jneilliii/OctoPrint-BedLevelVisualizer #222: Regarding GCODE profile for Ender 3 Pro

minnixtx, 2020-04-10:
Machine: Ender 3 Pro
Firmware: Marlin 2.0.5.3
Bed leveling enabled.
Octoprint: 1.4.0
Bed-visualizer: 0.1.13

This isn't an issue per se but I just need a GCODE configuration for my particular setup. Using the same one from the read me I used:

<pre><code>G28
 M155 S30
 @BEDLEVELVISUALIZER
 G29 T
 M155 S3</code></pre>

With this GCODE when I choose to update mesh within the plugin window, the ender 3 just homes and nothing else while I get a message that bed visualizer is waiting. I am able to level the bed through physical printer interface though. I know this is not the plugins fault and I am just not setting up the GCODE correctly but I am so new to this that I could really use a helping hand and had nowhere else to turn. Thanks in advance

jneilliii, 2020-04-11:
If you open the terminal tab of octoprint (suppress temperature, sd status, wait, etc), can you tell what commands is used to level with?

minnixtx, 2020-04-11:
ok let me level it now and i will report back

minnixtx, 2020-04-11:
<detials>
<summary>terminal output</summary>

<pre><code>Recv: echo:busy: processing
Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
Send: M113 S2
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0
Recv: ok
Recv: ok
[...]
Recv: echo:busy: processing
Recv: MBL G29 point 1 of 9
Recv: X:10.00 Y:10.00 Z:0.00 E:0.00 Count X:800 Y:800 Z:0
Recv: ok
[...]
Recv: echo:busy: processing
[...]
Recv: MBL G29 point 2 of 9
Recv: X:117.50 Y:10.00 Z:0.00 E:0.00 Count X:9400 Y:800 Z:0
Recv: ok
[...]
Recv: echo:busy: processing
[...]
Recv: MBL G29 point 3 of 9
Recv: X:225.00 Y:10.00 Z:0.20 E:0.00 Count X:18000 Y:800 Z:80
Recv: ok
[...]
Recv: echo:busy: processing
[...]
Recv: MBL G29 point 4 of 9
Recv: X:225.00 Y:117.50 Z:0.33 E:0.00 Count X:18000 Y:9400 Z:130
Recv: ok
[...]
Recv: echo:busy: processing
[...]
Recv: MBL G29 point 5 of 9
Recv: X:117.50 Y:117.50 Z:0.25 E:0.00 Count X:9400 Y:9400 Z:100
Recv: ok
[...]
Recv: echo:busy: processing
[...]
Recv: MBL G29 point 6 of 9
Recv: X:10.00 Y:117.50 Z:0.20 E:0.00 Count X:800 Y:9400 Z:80
Recv: ok
[...]
Recv: echo:busy: processing
[...]
Recv: MBL G29 point 7 of 9
Recv: X:10.00 Y:225.00 Z:0.10 E:0.00 Count X:800 Y:18000 Z:40
Recv: ok
[...]
Recv: echo:busy: processing
[...]
Recv: MBL G29 point 8 of 9
Recv: X:117.50 Y:225.00 Z:0.38 E:0.00 Count X:9400 Y:18000 Z:150
Recv: ok
[...]
Recv: echo:busy: processing
[...]
Recv: MBL G29 point 9 of 9
Recv: X:225.00 Y:225.00 Z:0.27 E:0.00 Count X:18000 Y:18000 Z:110
Recv: ok
[...]
Recv: Mesh probing done.
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
Recv: X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0
Recv: MBL G29 point -1 of 9
Recv: X:0.00 Y:0.00 Z:0.03 E:0.00 Count X:0 Y:0 Z:0
Recv: ok</pre></code></details>

jneilliii, 2020-04-11:
Ok, just to verify...your gcode lines don't have a space at the front of them do they? What do you get if you run an `M420 V` command in the terminal tab?

minnixtx, 2020-04-11:
No spaces, that was just an error in how I formatted the test  when I posted.

<code>M420 V</code>

<code>Send: M420 V
Recv: echo:Invalid mesh.
Recv: echo:Bed Leveling OFF
Recv: echo:Fade Height OFF
Recv: ok</code>

minnixtx, 2020-04-11:
That's what it was, I didn't store my settings. It works now. 
[image]

closed 2020-04-11 by minnixtx