jneilliii/OctoPrint-BedLevelVisualizer #223: Feature request: read and display the mesh leveling of the eeprom

devildant, 2020-04-13:
hello, already for started very good plugin, thank you for that :)

I have a small request (or not), I have a magnetic bed, to be always in agreement with the sufrace of the bed I launch a g29 at the start of each print.
I would like the surface of the bed to be updated in the plugin.

start gcode possible:
```
G28
G29
M500
M420 S1

M118 E1 BEDLEVELVISUALIZER // for send a text BEDLEVELVISUALIZER  to octoprint, you can catch this text
M420 V1
```
the M420 V1 send the stat to serial

result : 
```
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 -0.056 +0.022 -0.003 +0.078 +0.080
Recv:  1 -0.049 +0.007 +0.000 +0.084 +0.095
Recv:  2 -0.061 +0.002 -0.000 +0.084 +0.100
Recv:  3 -0.101 -0.018 -0.031 +0.042 +0.055
Recv:  4 -0.133 -0.067 -0.077 +0.005 +0.013
Recv: 
Recv: echo:Bed Leveling Marche
```

jneilliii, 2020-04-13:
In theory if you change your start gcode to this it should collect and graph.

```
G28
G29
M500
M420 S1

@BEDLEVELVISUALIZER
M420 V1
```

devildant, 2020-04-13:
Hello, 
I do print via SD, the @BEDLEVELVISUALIZER will not be recognized by marlin and therefore not transmitted to the serial.
that's why I gave you the command M118 E1 BEDLEVELVISUALIZER, this allows you to send a message to the octoprint, then just catch it :) 

when it is about an SD impression, only certain commands can be recovered by octoprint (I used M118 in one of my plugins because I needed to trigger an action)

devildant, 2020-04-13:
exemple with hook :octoprint.comm.protocol.gcode.received 
```
def receivedGCODE(self, comm_instance, line, *args, **kwargs):
		if self.printing and "echo:BEDLEVELVISUALIZER" == line.strip():
			def process():
				#do something
			thread = threading.Thread(target=process)
			thread.daemon = True
			thread.start()
			
		return line
```

jneilliii, 2020-04-13:
Good point, didn't think of printing from SD. I'll look into incorporating this on the next update.

jneilliii, 2020-07-31:
I've started looking at implementing this for the next release @devildant and was curious if you get mesh data in the terminal of OctoPrint when you run the leveling command from SD?

devildant, 2020-07-31:
hello @jneilliii , yes everything is in my first message :)

`M118 E1 BEDLEVELVISUALIZER`

the preceding command sends an event to octoprint (serial) and can therefore be caught (it is to replace @BEDLEVELVISUALIZER)

`M420 V1`
the preceding command is the bed leveling state command in verbose mode and during my tests the mesh was successfully send to octoprint with a sd print (visible in the terminal)

result : 
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 -0.056 +0.022 -0.003 +0.078 +0.080
Recv:  1 -0.049 +0.007 +0.000 +0.084 +0.095
Recv:  2 -0.061 +0.002 -0.000 +0.084 +0.100
Recv:  3 -0.101 -0.018 -0.031 +0.042 +0.055
Recv:  4 -0.133 -0.067 -0.077 +0.005 +0.013
Recv: 
Recv: echo:Bed Leveling Marche

exemple code for catch M118 E1 BEDLEVELVISUALIZER : 

> exemple avec hook: octoprint.comm.protocol.gcode.received
> 
> ```
> def receivedGCODE(self, comm_instance, line, *args, **kwargs):
> 		if self.printing and "echo:BEDLEVELVISUALIZER" == line.strip():
> 			def process():
> 				#do something
> 			thread = threading.Thread(target=process)
> 			thread.daemon = True
> 			thread.start()
> 			
> 		return line
> ```

jneilliii, 2020-07-31:
implemented in above commit. If you want to give it a try and verify for me that would be great. I just don't have a SD card printer to test with. You can install by using this URL in Plugin Manager > Get More > ...from URL.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.15.zip

devildant, 2020-07-31:
I'll test this for you this weekend (I'm currently printing: p)

devildant, 2020-08-01:
One shot 👍

commit f63eb413 (2020-07-31): first stab at SD card support, resolves #223
commit 3fb614e6 (2020-10-03): 0.1.15 (#304)

fix axis label colors, resolves #278
make colorscale limits allow option auto for the old method of coloring, resolves #275
add responsive option to plotly config, resolves #274
first stab at SD card support, resolves #223
fix little issue with reactive option where graph was getting overlaid on top of correction table
change home button press to a different strategy, resolves #265
update plotly to version 1.54.7
add option to download snapshots on graph rendering completion, #239
remove custom home button in preference of resetCameraLastSave3d, resolves #265
numpy versioning fo [... cut]
commit 7565c066 (2020-10-03): 0.1.15 (#304)

fix axis label colors, resolves #278
make colorscale limits allow option auto for the old method of coloring, resolves #275
add responsive option to plotly config, resolves #274
first stab at SD card support, resolves #223
fix little issue with reactive option where graph was getting overlaid on top of correction table
change home button press to a different strategy, resolves #265
update plotly to version 1.54.7
add option to download snapshots on graph rendering completion, #239
remove custom home button in preference of resetCameraLastSave3d, resolves #265
numpy versioning fo [... cut]
closed 2020-08-18 by jneilliii