jneilliii/OctoPrint-BedLevelVisualizer #224: Install Successfully but cant display in Octoprint

technik-dude, 2020-04-15:
Hello,

I installed a fresh Ocotprint 1.4 and wanted to install this Plugin. The Pluginmanager tell me after finish the install "Successfully installed Bed-Visualizer-0.1.13".I restarted the prozess and the Pi (3). But nothing happen. So i read the Log from Octoprint.

```
2020-04-15 00:44:58,069 - octoprint.plugin.core - ERROR - Error loading plugin bedlevelvisualizer
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv3/lib/python3.7/site-packages/numpy/core/__init__.py", line 24, in <module>
    from . import multiarray
  File "/home/pi/OctoPrint/venv3/lib/python3.7/site-packages/numpy/core/multiarray.py", line 14, in <module>
    from . import overrides
  File "/home/pi/OctoPrint/venv3/lib/python3.7/site-packages/numpy/core/overrides.py", line 7, in <module>
    from numpy.core._multiarray_umath import (
ImportError: libf77blas.so.3: cannot open shared object file: No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv3/lib/python3.7/site-packages/octoprint/plugin/core.py", line 967, in _import_plugin
    module = _load_module(module_name, spec)
  File "/home/pi/OctoPrint/venv3/lib/python3.7/site-packages/octoprint/plugin/core.py", line 71, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/home/pi/OctoPrint/venv3/lib/python3.7/site-packages/octoprint/vendor/imp.py", line 241, in load_module
    return load_package(name, filename)
  File "/home/pi/OctoPrint/venv3/lib/python3.7/site-packages/octoprint/vendor/imp.py", line 215, in load_package
    return _load(spec)
  File "<frozen importlib._bootstrap>", line 696, in _load
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/pi/OctoPrint/venv3/lib/python3.7/site-packages/octoprint_bedlevelvisualizer/__init__.py", line 6, in <module>
    import numpy as np
  File "/home/pi/OctoPrint/venv3/lib/python3.7/site-packages/numpy/__init__.py", line 142, in <module>
    from . import core
  File "/home/pi/OctoPrint/venv3/lib/python3.7/site-packages/numpy/core/__init__.py", line 54, in <module>
    raise ImportError(msg)
ImportError: 

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the numpy c-extensions failed.
- Try uninstalling and reinstalling numpy.
- If you have already done that, then:
  1. Check that you expected to use Python3.7 from "/home/pi/OctoPrint/venv3/bin/python3",
     and that you have no directories in your PATH or PYTHONPATH that can
     interfere with the Python and numpy version "1.18.2" you're trying to use.
  2. If (1) looks fine, you can open a new issue at
     https://github.com/numpy/numpy/issues.  Please include details on:
     - how you installed Python
     - how you installed numpy
     - your operating system
     - whether or not you have multiple versions of Python installed
     - if you built from source, your compiler versions and ideally a build log

- If you're working with a numpy git repository, try `git clean -xdf`
  (removes all files not under version control) and rebuild numpy.

Note: this error has many possible causes, so please don't comment on
an existing issue about this - open a new one instead.

Original error was: libf77blas.so.3: cannot open shared object file: No such file or directory
```
I tryed uninstalling and reinstalling numpy and check the path to python and numpy. Everything look fine but nothing happend. 
I have no idea what else I can do. Where am I doing something wrong?
Other plugins work perfectly and can be installed without any problems.

jneilliii, 2020-04-15:
Looks like numpy install/build failed. The only thing I can suggest is making sure that when you manually try to install numpy you are active in your venv3 virtual environment when running the pip commands. Something like this via SSH.

```
source /home/pi/OctoPrint/venv3/bin/activate
pip install numpy
```

There may also be missing setuptools and compilers, etc. I'm not really sure what exactly is required.

technik-dude, 2020-04-15:
I tryed exactly this commands. The uninstall and reinstall of numpy was successful. I restart the PI and after that i try again to install bedlevlevisualizer. Same message again. The pluginmanager said everything is fine but the plugin wont start. The logs tell me the same story. : (

jneilliii, 2020-04-15:
Someone else was having a similar issue [here](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/issues/108#issuecomment-466799905) and it seems updating setuptools fixed it for them.

technik-dude, 2020-04-15:
pip install --upgrade setuptools
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already up-to-date: setuptools in ./venv3/lib/python3.7/site-packages (46.1.3)

So setuptool is up to date. After that i reinstall numpy and give a new try to BedLevelVisualizer. :/ with same result. Did not work.
Numpy 1.18.2 is installed. Is this one to new? In the link you post they talk about 1.15.4

jneilliii, 2020-04-17:
Just remembered I knew I had saw this before...this is a duplicate of https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/issues/213 and is discussed further over [here](https://github.com/numpy/numpy/issues/11110) and [here](https://github.com/numpy/numpy/issues/15890).

```
sudo apt-get install libatlas-base-dev
```
or
```
pip3 uninstall numpy #remove previously installed package
apt install python3-numpy
```

technik-dude, 2020-04-18:
Thx dude!
This command work for me. Sorry that i didnt see the other issue befor :(

closed 2020-04-18 by jneilliii