jneilliii/OctoPrint-BedLevelVisualizer #225: Failed to execute 'uniformMatrix4fv' on 'WebGLRenderingContext

vov32, 2020-04-16:
Bed Visualizer (0.1.13), Klipper (don't know exact version, but it is fresh)
`TypeError: Failed to execute 'uniformMatrix4fv' on 'WebGLRenderingContext': No function was found that matched the signature provided.`
Output of ~/.octoprint/logs/plugin_bedlevelvisualizer_debug.log
```
[2020-04-16 09:01:24,372] DEBUG: mesh collection started
[2020-04-16 09:01:24,376] DEBUG: // Mesh Leveling Probed Z positions:
[2020-04-16 09:01:24,389] DEBUG: // -9.709671 -9.709671 -9.709671 -9.709671 -9.709671
[2020-04-16 09:01:24,392] DEBUG: // -8.882791 -8.882791 -10.053788 -11.430901 -11.430901
[2020-04-16 09:01:24,394] DEBUG: // -8.750643 -9.474328 -10.602038 -11.780672 -12.702485
[2020-04-16 09:01:24,396] DEBUG: // -10.316536 -10.316536 -11.288482 -12.200490 -12.200490
[2020-04-16 09:01:24,397] DEBUG: // -11.619917 -11.619917 -11.619917 -11.619917 -11.619917
[2020-04-16 09:01:24,399] DEBUG: Mesh X,Y: 13,13
[2020-04-16 09:01:24,403] DEBUG: Search Height: 5
[2020-04-16 09:01:24,406] DEBUG: Mesh Average: -10.69
[2020-04-16 09:01:24,408] DEBUG: Mesh Range: min=-12.7106 max=-8.5946
[2020-04-16 09:01:24,410] DEBUG: Interpolation Algorithm: lagrange
[2020-04-16 09:01:24,411] DEBUG: Measured points:
[2020-04-16 09:01:24,413] DEBUG: -11.619917  -11.619917  -11.619917  -11.619917  -11.619917  -11.619917  -11.619917  -11.619917  -11.619917  -11.619917  -11.619917  -11.619917  -11.619917
[2020-04-16 09:01:24,415] DEBUG: -11.457617  -11.164509  -11.045310  -11.064503  -11.186547  -11.375878  -11.596911  -11.814040  -11.991635  -12.094045  -12.085594  -11.930588  -11.593307
[2020-04-16 09:01:24,417] DEBUG: -10.962361  -10.654357  -10.565938  -10.649657  -10.859128  -11.149021  -11.475069  -11.794062  -12.063850  -12.243342  -12.292507  -12.172373  -11.845028
[2020-04-16 09:01:24,418] DEBUG: -10.316536  -10.137446  -10.152406  -10.316536  -10.587181  -10.923902  -11.288482  -11.644921  -11.959444  -12.200490  -12.338722  -12.347022  -12.200490
[2020-04-16 09:01:24,420] DEBUG: -9.665937  -9.654916  -9.785757  -10.023313  -10.335183  -10.691705  -11.065965  -11.433790  -11.773749  -12.067158  -12.298074  -12.453298  -12.522375
[2020-04-16 09:01:24,426] DEBUG: -9.119768  -9.241057  -9.457471  -9.745176  -10.082564  -10.450247  -10.831062  -11.210068  -11.574547  -11.914003  -12.220166  -12.486984  -12.710632
[2020-04-16 09:01:24,452] DEBUG: -8.750643  -8.923317  -9.169465  -9.474328  -9.823710  -10.203973  -10.602038  -11.005388  -11.402065  -11.780672  -12.130369  -12.440880  -12.702485
[2020-04-16 09:01:24,702] DEBUG: -8.594582  -8.722294  -8.934092  -9.219990  -9.567965  -9.963960  -10.391885  -10.833614  -11.268983  -11.675798  -12.029827  -12.304803  -12.472427
[2020-04-16 09:01:25,878] DEBUG: -8.651018  -8.651742  -8.774143  -9.008394  -9.339623  -9.747917  -10.208322  -10.690838  -11.160426  -11.577004  -11.895447  -12.065589  -12.032220
[2020-04-16 09:01:27,030] DEBUG: -8.882791  -8.718566  -8.722844  -8.882791  -9.177938  -9.580183  -10.053788  -10.555385  -11.033969  -11.430901  -11.679910  -11.707090  -11.430901
[2020-04-16 09:01:28,165] DEBUG: -9.216150  -8.922828  -8.823860  -8.903447  -9.137117  -9.491725  -9.925453  -10.387811  -10.819634  -11.153088  -11.311662  -11.210176  -10.754774
[2020-04-16 09:01:36,082] DEBUG: -9.540754  -9.257740  -9.131289  -9.147641  -9.286322  -9.520144  -9.815208  -10.130900  -10.419895  -10.628153  -10.694921  -10.552734  -10.127415
[2020-04-16 09:01:43,988] DEBUG: -9.709671  -9.709671  -9.709671  -9.709671  -9.709671  -9.709671  -9.709671  -9.709671  -9.709671  -9.709671  -9.709671  -9.709671  -9.709671
[2020-04-16 09:01:44,013] DEBUG: [u'-9.709671', u'-9.709671', u'-9.709671', u'-9.709671', u'-9.709671', u'-9.709671', u'-9.709671', u'-9.709671', u'-9.709671', u'-9.709671', u'-9.709671', u'-9.709671', u'-9.709671']
[2020-04-16 09:01:44,022] DEBUG: 
[2020-04-16 09:01:44,040] DEBUG: ok
[2020-04-16 09:01:44,043] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 340.0, 'x_min': 0, 'type': 'circular', 'y_max': 340.0, 'z_max': 500.0}
[2020-04-16 09:01:44,044] DEBUG: stopping mesh collection
[2020-04-16 09:01:44,048] DEBUG: [[u'-9.709671', u'-9.709671', u'-9.709671', u'-9.709671', u'-9.709671', u'-9.709671', u'-9.709671', u'-9.709671', u'-9.709671', u'-9.709671', u'-9.709671', u'-9.709671', u'-9.709671']]

```

jneilliii, 2020-04-16:
What's in your debug log is exactly correct tp be able to map the mesh. The problem I think is the numbers are out of visual range. Try enabling the option to use relative z offsets andsee restart octoprint and try again.

vov32, 2020-04-16:
Thank you! It done, but no success. Same error occured.
```
[2020-04-16 13:10:20,914] DEBUG: mesh collection started
[2020-04-16 13:10:20,951] DEBUG: // Mesh Leveling Probed Z positions:
[2020-04-16 13:10:20,979] DEBUG: // -9.709671 -9.709671 -9.709671 -9.709671 -9.709671
[2020-04-16 13:10:20,981] DEBUG: // -8.882791 -8.882791 -10.053788 -11.430901 -11.430901
[2020-04-16 13:10:20,983] DEBUG: // -8.750643 -9.474328 -10.602038 -11.780672 -12.702485
[2020-04-16 13:10:20,985] DEBUG: // -10.316536 -10.316536 -11.288482 -12.200490 -12.200490
[2020-04-16 13:10:20,987] DEBUG: // -11.619917 -11.619917 -11.619917 -11.619917 -11.619917
[2020-04-16 13:10:20,989] DEBUG: Mesh X,Y: 13,13
[2020-04-16 13:10:20,992] DEBUG: Search Height: 5
[2020-04-16 13:10:20,995] DEBUG: Mesh Average: -10.69
[2020-04-16 13:10:20,997] DEBUG: Mesh Range: min=-12.7106 max=-8.5946
[2020-04-16 13:10:20,999] DEBUG: Interpolation Algorithm: lagrange
[2020-04-16 13:10:21,001] DEBUG: Measured points:
[2020-04-16 13:10:21,003] DEBUG: -11.619917  -11.619917  -11.619917  -11.619917  -11.619917  -11.619917  -11.619917  -11.619917  -11.619917  -11.619917  -11.619917  -11.619917  -11.619917
[2020-04-16 13:10:21,005] DEBUG: -11.457617  -11.164509  -11.045310  -11.064503  -11.186547  -11.375878  -11.596911  -11.814040  -11.991635  -12.094045  -12.085594  -11.930588  -11.593307
[2020-04-16 13:10:21,006] DEBUG: -10.962361  -10.654357  -10.565938  -10.649657  -10.859128  -11.149021  -11.475069  -11.794062  -12.063850  -12.243342  -12.292507  -12.172373  -11.845028
[2020-04-16 13:10:21,008] DEBUG: -10.316536  -10.137446  -10.152406  -10.316536  -10.587181  -10.923902  -11.288482  -11.644921  -11.959444  -12.200490  -12.338722  -12.347022  -12.200490
[2020-04-16 13:10:21,010] DEBUG: -9.665937  -9.654916  -9.785757  -10.023313  -10.335183  -10.691705  -11.065965  -11.433790  -11.773749  -12.067158  -12.298074  -12.453298  -12.522375
[2020-04-16 13:10:21,015] DEBUG: -9.119768  -9.241057  -9.457471  -9.745176  -10.082564  -10.450247  -10.831062  -11.210068  -11.574547  -11.914003  -12.220166  -12.486984  -12.710632
[2020-04-16 13:10:21,040] DEBUG: -8.750643  -8.923317  -9.169465  -9.474328  -9.823710  -10.203973  -10.602038  -11.005388  -11.402065  -11.780672  -12.130369  -12.440880  -12.702485
[2020-04-16 13:10:21,206] DEBUG: -8.594582  -8.722294  -8.934092  -9.219990  -9.567965  -9.963960  -10.391885  -10.833614  -11.268983  -11.675798  -12.029827  -12.304803  -12.472427
[2020-04-16 13:10:22,339] DEBUG: -8.651018  -8.651742  -8.774143  -9.008394  -9.339623  -9.747917  -10.208322  -10.690838  -11.160426  -11.577004  -11.895447  -12.065589  -12.032220
[2020-04-16 13:10:23,491] DEBUG: -8.882791  -8.718566  -8.722844  -8.882791  -9.177938  -9.580183  -10.053788  -10.555385  -11.033969  -11.430901  -11.679910  -11.707090  -11.430901
[2020-04-16 13:10:24,645] DEBUG: -9.216150  -8.922828  -8.823860  -8.903447  -9.137117  -9.491725  -9.925453  -10.387811  -10.819634  -11.153088  -11.311662  -11.210176  -10.754774
[2020-04-16 13:10:32,577] DEBUG: -9.540754  -9.257740  -9.131289  -9.147641  -9.286322  -9.520144  -9.815208  -10.130900  -10.419895  -10.628153  -10.694921  -10.552734  -10.127415
[2020-04-16 13:10:40,523] DEBUG: -9.709671  -9.709671  -9.709671  -9.709671  -9.709671  -9.709671  -9.709671  -9.709671  -9.709671  -9.709671  -9.709671  -9.709671  -9.709671
[2020-04-16 13:10:40,526] DEBUG: [u'-9.709671', u'-9.709671', u'-9.709671', u'-9.709671', u'-9.709671', u'-9.709671', u'-9.709671', u'-9.709671', u'-9.709671', u'-9.709671', u'-9.709671', u'-9.709671', u'-9.709671']
[2020-04-16 13:10:40,531] DEBUG: 
[2020-04-16 13:10:40,534] DEBUG: ok
[2020-04-16 13:10:40,537] DEBUG: {'z_min': 0, 'y_min': 0, 'x_max': 340.0, 'x_min': 0, 'type': 'circular', 'y_max': 340.0, 'z_max': 500.0}
[2020-04-16 13:10:40,538] DEBUG: stopping mesh collection
[2020-04-16 13:10:40,541] DEBUG: using relative offsets
[2020-04-16 13:10:40,544] DEBUG: [[0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]]
```

jneilliii, 2020-04-16:
I think you're going to have to get you bed more flat manually maybe. 

> Range: min=-12.7106 max=-8.5946

That's a large variance and the plugin only maps 2mm in the z.

vov32, 2020-04-17:
> I think you're going to have to get you bed more flat manually maybe.
> 
> > Range: min=-12.7106 max=-8.5946
> 
> That's a large variance and the plugin only maps 2mm in the z.
You are right. I tensed belts and re-calibrate everything. Now min/max range is common. And error gone.

closed 2020-04-17 by vov32

randellhodges, 2020-04-26:
I was getting this error as well today and it was because I turned on gzip encoding inside nginx (not using octopi).

I disabled gzip and it started working.

```
        gzip on;

        gzip_vary on;
        gzip_proxied any;
        gzip_comp_level 6;
        gzip_buffers 16 8k;
        gzip_http_version 1.1;
        gzip_types text/plain text/css application/json application/javascript text/xml application/xml application/xml+rss text/javascript;
```
I wasn't seeing any response in the developer tools. I'm sure it is just a misconfiguration here, but I thought I'd add this comment in the event someone else runs into the issue.