jneilliii/OctoPrint-BedLevelVisualizer #227: Error:!! STOP called because of BLTouch error - restart with M999

tbrogdon, 2020-04-17:
I have tried multiple times to run a bed level through Bed Visulaizer and always get this error.  It worked at one point, then started doing this.  If I run a Bed Level from LCD it works without any issues.  I went through firmware settings thinking maybe I did something wrong and no changes affect it.  It will work for a period of time and then stop, the BLTouch flashes red, printer reports error with BLTouch and logs from Visualizer so the error listed.  Sometimes this happens on first touch, sometimes it will make it halfway through, completely random.  I've do have the recommended gcode settings on Visualizer as well.

jneilliii, 2020-04-17:
The plugin doesn't really do anything specific to the GCODE, it just sends them as entered. What you are describing sounds similar to an issue I was having with my BLTouch that ended up being a bad connection on one of the pins and the unit was not getting consistent power.

tbrogdon, 2020-04-17:
Your plugin requires gcode, that is what I was talking about.  As for connection, I would agree with you IF it didn't work perfectly outside of Bed Visualizer.

jneilliii, 2020-04-17:
When you do it from the printer control panel it's the same as entering gcode from this plugin or the terminal tab. If you enable debug logging in the plugin's settings and restart OctoPrint then try an update mesh again we can possibly track down the problem. When you do the leveling from the printer control panel what do you see in the terminal tab of OctoPrint?

tbrogdon, 2020-04-17:
Sorry, I think it is bad.  For some reason it fails every single time through Bed Visualizer but basically never through LCD.  However, I did just have a failure through LCD.  I'm going to get replacement and see how it goes.

jneilliii, 2020-04-17:
Like I said, it might just be a bad connection or supply power issue to the probe. It really depends on your board. I think the newer version BLTouch has some command that will flip the power logic between 5v and 3.3v as well if I remember correctly.

closed 2020-04-17 by jneilliii