jneilliii/OctoPrint-BedLevelVisualizer #231: "nan" values for grid causes plugin to get stuck in a processing state

kantlivelong, 2020-04-22:
OctoPrint Ver : 1.4.0
Plugin Ver: 0.1.13
FW: Smoothieware edge

When a probe has not yet been done Smoothieware will report **nan** for the grid levels. BLV will get stuck in the processing state until good values are reported. I took a look at the code and I seems like https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/blob/1e800d20799e6ab7cd214c8e8f5dab8253298696/octoprint_bedlevelvisualizer/__init__.py#L144 would handle **NAN** in uppercase but am not 100% sure as that RegEx melts my brain.

Working Sample:
```
Send: M375.1
Recv: 0.00 0.00 0.00
Recv: 0.00 1.00 0.00
Recv: 0.00 0.00 0.00
```

Bad Sample:
```
Send: M375.1
Recv: nan nan nan
Recv: nan nan nan
Recv: nan nan nan
```

Can replicate by adding the following to `virtual.py` in the OctoPrint Virtual Printer
```
	#Drop the .1 part of the command and just use M375 for testing.
	def _gcode_M375(self, data):
		self._send("nan nan nan")
		self._send("nan nan nan")
		self._send("nan nan nan")
```

jneilliii, 2020-04-22:
> but am not 100% sure as that RegEx melts my brain.

Yeah, it melts mine too. One reason I wish firmware vendors would standardize the protocol. In theory it would just be a matter of adding another grouping there. I tend to use regex101.com to test the whole string.

`|(nan\,?))+(\s+\],?)`

jneilliii, 2020-04-23:
You want to test this out with your current fork and update the PR? I think this would work.

```
 if self.processing and "ok" not in line and re.match(r"^((G33.+)|(Bed.+)|(\d+\s)|(\|\s*)|(\s*\[\s+)|(\[?\s?\+?\-?\d?\.\d+\]?\s*\,?)|(\s?\.\s*)|(NAN\,?)|(nan\,?\s?))+(\s+\],?)?$", line.strip()): 
```

kantlivelong, 2020-04-23:
Is there a reason for the difference between `(NAN\,?)` and `(nan\,?\s?)`?

Results with **nan** returned `True` and **NAN** `False`. I honestly didn't test **NAN** before as I wasn't sure if that was the right block. So with **nan** returning `True` it still doesn't work but I think the problem is at https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/blob/1e800d20799e6ab7cd214c8e8f5dab8253298696/octoprint_bedlevelvisualizer/__init__.py#L145 as `new_line` isn't set and that regex doesn't account for any variation of **nan**.

If i manually set `new_line` to `['nan', 'nan', 'nan']` in the debuger it will run through but then the js side complains because of NaN values. Seems like this will take more than a simple patch?

kantlivelong, 2020-04-23:
.. Ideally the UI would just say something to the order of "No information available at the moment."

jneilliii, 2020-04-23:
No worries. I'll dig into it when I can. Some of the recent changes introduced by another contributor is probably impacting this some on the js side.

kantlivelong, 2020-04-23:
All good. Figured I'd at least report it. Might optimize the hook a bit more if I can figure it out. Breaking up that regex would be really helpful.

jneilliii, 2020-04-23:
So that regex is basically to say "here comes the real data you're interested in" start grabbing the points/offsets and stuff into the mesh array line by line. All the virtual gcode files that are in the repo in theory should cover most examples I've collected over time and can be printed while using the virtual printer to verify/debug bed collection functionality.

jneilliii, 2020-04-23:
I'm always open to improvements in code to make it easier to update or less error prone for the end users. I'd love to be able to say "I'm only supporting Marlin with UBL" as that's what I have on my setup and originally developed for. That regex was way easier back then...lol.

kantlivelong, 2020-04-23:
Ah. I'll submit a sample as well once the time comes.

>  I'd love to be able to say "I'm only supporting Marlin with UBL" as that's what I have on my setup and originally developed for. That regex was way easier back then...lol.

That's why I tend to stay away from firmware specific features. Gets ugly O_O

jneilliii, 2020-05-25:
@kantlivelong what settings are you using in the plugin? It seems after looking at your serial.log that G30 is the probe command and that wouldn't return the nan values. So something like this in the plugins settings should work.
```
; do whatever prep work gcode here
@BEDLEVELVISUALIZER
G30
; do whatever post work gcode here
```

kantlivelong, 2020-05-25:
Right, I don't have BLV actually doing the probe and am just using it to visualize the data. This is mainly due to the fact that BLV would try to probe immediately when switching to the tab instead of having a dedicated "Probe Now" button. Ideally the tab should just say "No data" instead of infinitely hanging.

**GCODE Commands for Mesh Update Process**
```
M375.1
```
**Save Mesh**: off

jneilliii, 2020-05-25:
Yeah, Save Mesh on would bypass the automatic probing of the bed, but I understand your use case now. You are running leveling outside of the plugin and just want it to automatically load the last stored in EEPROM rather than storing it within the plugin and require to use it to run the probing.

If I understand correctly this also only happens after a controller restart?

kantlivelong, 2020-05-26:
In my case I don't even store the data in the EEPROM but instead just run probing before each print. I could store the data but I figure it's a waste of writes being I'll run it again anyway. Perhaps I'm just doing something non-standard but personally I'd rather always having fresh data rather than relying on past tests.

jneilliii, 2020-05-26:
Ok, I got is parsing now and replacing with zeros. Still a little javascript work to do. Try this one....

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.14.zip

closed 2020-07-26 by jneilliii