jneilliii/OctoPrint-BedLevelVisualizer #232: Suggestion...Modify Z axis Scale

cmontgo, 2020-04-23:
I have a large bed and lots of points, could the z scale be set by the user? 2mm off seems catastrophic, I’d like to set it to something like 0.5mm so I could better see small variation

jneilliii, 2020-04-23:
2mm is just what ended up looking the best visually and anything lower could cause some graphs to not render at all and they were getting cut off because they may be that far off. Whatever the offset, the surface itself will still scale, so not even sure if it would make a difference.

cmontgo, 2020-04-23:
That’s why I was thinking user adjustable. I have a 500mm bed, and I’m hoping the variation won’t be too much🤞 So I’m looking for the variations to be more pronounced.

Cameron Redsell-Montgomerie

- Math is Awesome; Question Everything

slim0287, 2020-05-29:
I recently installed Bed Level Visualizer.  I too have a problem with the Z axis range.  The max my bed is out is +0.052, so on the plot it basically looks flat because the range is +/-2 (screenshot below).  @jneilliii writes that the surface itself scales but that doesn't appear to be the case for me.

[image]

jneilliii, 2020-05-29:
@slim0287 what I meant in regards to scaling is the color mapping, that currently scales to the retrieved data, not the overall surface itself. I'm working on changing that.

jneilliii, 2020-05-29:
Here's where I'm at so far...

[image]

slim0287, 2020-05-29:
I am printing on glass so I'm not surprised it is quite flat.  If I understand properly you are working on changing the way the 3D plot is displayed for those of us with high quality surfaces.  So, I'll wait until the next version to see how it looks.   I really like the plugin by the way-- after going through the trouble to add a sensor to my system and modify the firmware to account for the measurement it is very satisfying to visualize the surface,

jneilliii, 2020-05-29:
Yeah, just really need to figure out at what point it should be red I think. I think if I make it red at +/-0.25 it might make more sense than the above example. But then the extremely level/flat beds still end up being all blue. Thoughts?

[image]

jneilliii, 2020-05-29:
The idea is to eliminate some of the confusion as it relates to the auto coloring before.

slim0287, 2020-05-30:
For me, the color works very well.  -0.02 is blue and +0.05 is red.  Your plots don't seem to select the color scale like mine does.  What I'd like to do is see some shape to the surface-- the plot has Z going from -2 to +2, so despite my colors looking good and reflecting the surface distortion, the plot shows something which looks very much like a plane.  So, it appears we have different problems, perhaps?

jneilliii, 2020-05-30:
I see what you're saying. Yes, slightly different issues. I think maybe to resolve both needs converting the z axis to be dynamic instead of fixed at +/-2 and the color scaling as shown above would potentially resolve both.

cmontgo, 2020-05-30:
That could work. Even if there was a manual override in the back end, and maybe colour based on the selection?
On a related not, it would be sweet if you could select and recalibrate specific points, but that’s probably not possible. Just a thought.

slim0287, 2020-05-30:
Yes, I agree.  The z-axis scaling should be dynamic.  At least from my perspective, I'd like an image of what my distorted print surface actually looks like.  And if you'd like to go the extra mile, you could put in a setting which allows the user to chose whether the Z axis is dynamic or is set to some fixed value.

jneilliii, 2020-05-30:
@cmontgo you can toggle the little arrows button in the hover bar and that switches to go to point mode. When you click a point it will go that point on your printer. Keep in mind, the points might not be exact because of offsets, and estimations, but will be close I think.

cmontgo, 2020-05-30:
Oh wow, very cool, I hadn’t noticed as I run octoprint through an iPad. Since I’m doing manual mesh leveling with the display (yes, working on the upgrade) is there an easy way to modify that point once I select it? I guess I could send commands for a new point then test?

jneilliii, 2020-05-31:
That's one of the reasons for the custom buttons in the plugin's settings. You can create a custom button for the gcode commands necessary to move/set/store your points, for example with UBL you go to the point and enter the command `G29 P4 T` and it allows you to adjust, then a button to save with `G29 S1` and/or `M500` to save the EEPROM. That's dependent on what your firmware supports and what type of leveling you use.

slim0287, 2020-06-09:
@jneilliii What did you end up deciding to do with this request to modify the Z axis scale?  My colors are chosen well, but it'd be nicer to see how the surface actually looks and the  +/- 2mm on the Z axis makes it look flat.

jneilliii, 2020-06-09:
I haven't yet @slim0287. More than likely will make it a configurable option in settings, but have been a little overwhelmed recently with work and other plugin bug fixes that are a higher priority right now.

jneilliii, 2020-06-15:
So I've pushed some changes to the current 0.1.4 development branch that makes the Z limits configurable. To get a closer offset change the settings to something like `-0.2,0.2` for a closer detailed scale. 

[image]

Hopefully this satisfies this request, if you want to try it out you can install using this URL below in plugin manager.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.14.zip

Would love to get some additional testers on this version as it has some significant performance improvements.

slim0287, 2020-06-15:
I'm sorry for the "noob" question, but do I need to uninstall 0.1.13 prior to installing 0.1.14?  Whenever I've updated a plugin it's been through the normal OctoPrint update process, so this case is a little different.

jneilliii, 2020-06-15:
No not necessary, you can just install on top of. I haven't bumped the version yet so it will still show as 0.1.13 once installed.

tablatronix, 2020-07-03:
ooh I need this, I am going to check this out.

jneilliii, 2020-07-04:
This will be included in the next release. Uses an comma separated list for z-min and z-max of the rendered mesh with the default values set to `-2,2`.

[image]

tablatronix, 2020-07-04:
Working good so far.

cmontgo, 2020-07-13:
Beautiful work!

closed 2020-07-26 by jneilliii

dewi-ny-je, 2020-12-29:
I use klipper and, as recommended, my printer "homes away", the endstop are at the end of the Z axis.
This means that the bed mesh is not centered around zero, but around a value that doesn't need to be zero, and depending on the surface I use on that print may change.

As of now, colours are useless to me because they assume zero is the best.

I think that one of the options should be to automatically center the colour scale to whatever the average value or the centre value (but meshes could be 4*4 and there is no centre value) are.

If you think about it, once bed mesh is used, there is no reason why the mesh should be ideally zero, what matters is that the min and max values are as close as possible.

Auto-centering the scale around the average would do that.

Proposal to combine the improvement you already implemented with my suggestion: instead of setting z min and z max in the settings, offer two text boxes: reference Z value and max Z deviation. Put a checkbox next to each one of them labelled "auto".
Reference Z (0 as default) and Z max deviation (1 mm as default, as it is now) both manual are equivalent to what you implemented, just a different way of setting them.
Z reference with auto would auto-centre the scale.
Max Z deviation auto would scale the amplitude for users with very warped beds or for those who want to zoom in if they have a very flat bed.