jneilliii/OctoPrint-BedLevelVisualizer #233: You haven't yet decided on whether to enable or disable anonymous usage tracking.

MrBill1234, 2020-04-24:
I managed to install this - but when i reload the UI i see the following:

` You haven't yet decided on whether to enable or disable anonymous usage tracking. You need to either enable or disable it before continuing.`

The plugin is enabled bug usage reporting is disabled - and there is no way out of this screen except to edit config.yaml and disable this plugin manually and restart octoprint.

I'm on OctoPrint 1.4.0

Is usage reporting a requirement?  I enabled it and the setup went through OK.

Thanks,

jneilliii, 2020-04-24:
I've never heard of this issue before. Anonymous usage tracking is a separate plugin, and not related to this one.

MrBill1234, 2020-04-24:
Yes, very odd.   In case anyone sees it in future:

Edit config.yaml

Under:

plugins:
    _disabled:

Add this:
     - bedlevelvisualizer

Resart octoprint:   sudo service octoprint restart

Enable anonymous usage tracking

Restart octoprint

Enable bedlevelvisualizer

Restart octoprint

Add your GCODE into the wizard for bedlevelvisualizer

Disable anonymous usage tracking (if you want).

Restart octoprint.

Everything should work.

jneilliii, 2020-04-24:
What version of python are you running OctoPrint with? What other plugins do you have installed?

MrBill1234, 2020-04-24:
Python 2.7.13

The only non standard ones are PortLister (0.1.8) - Cost Estimation (2.1.3) - and Bed Visualizer (0.1.13)

jneilliii, 2020-04-24:
Just to be clear on the steps to reproduce the issue, the following will cause the error to happen?

Uninstall this plugin with uninstall and clean up data button, disable anonymous usage tracking (if enabled), restart OctoPrint for good measure and make sure everything is working normally. Install Bed Visualizer and upon restart/reload process you get that wizard dialog for anonymous usage tracking?

MrBill1234, 2020-04-24:
To be honest - I cannot give a 100% reproduction instruction.

I can tell you that I had just upgraded from Octoprint 1.3.9 to 1.4.0.  Looked through my plugins, and installed one (the Cost Estimation), disabled the anonymous usage at the plugin level too (it was already soft-disabled in the options menu).

Something else - perhaps not related, but i should mention - i had a few failed installs of the bed visualizer due to it hanging while trying to compile numpy.  I eventually installed this manually on the command line and it was ok - after which the plugin installation worked fine via the octoprint UI.

I suspect this could Octoprint issue.  The wizard popped up asking me about the mesh GCODE - I entered that info - and when I pressed the Finished button - it popped up this error.  A quick search shows that this is indeed an error from the OC code tree.  The annoying thing was there was no way to quit out of the wizard.  I had to kill octoprint manually from the command line.

Hope this helps - let me know if you need anything further.

jneilliii, 2020-04-24:
Thanks for the information. This could be a combination of things it appears that made this random issue occur. The failed installs could possibly be related and that is one of the bad things about having a numpy dependency unfortunately. I have successfully installed via the OctoPrint UI on multiple pi models, including the pi zero, it just takes a really long time and you can't do anything while the process is ongoing.

MrBill1234, 2020-04-24:
I don't know if this is something under a plug-in's control - but after installing and restarting it takes you straight into the wizard - no choice to do anything else.   Can you change that, and/or perhaps put a <cancel> button in the wizard to allow you to enter this config data at a later time?

jneilliii, 2020-04-24:
So the only one I have control over is the one that is related to the GCODE commands being entered to update the mesh. I've been thinking of removing that wizard and a couple of other tweaks to the tab for initial installs in a future release already.

MrBill1234, 2020-04-24:
I think that would mitigate this problem in future.  Be sure to have a 'cancel' button on that input too if you don't want to proceed.  As it is now - you can do absolutely nothing until you get past that screen.  Before I encountered the error i logged this issue for, I actually wanted to get to the Terminal screen to test the various gcodes and see what the output looked like.  Couldn't do it.  Was stuck there.

jneilliii, 2020-04-24:
My thought was to remove this plugin's wizard, replace the gcode input on the tab with instructions, links to examples, and a button to open the plugin's settings to configure.

MrBill1234, 2020-04-24:
Yeah, that works :)

cplagz, 2020-05-10:
Mine exhibited a similar behaviour (docker on ubuntu).  Fresh container with backup restored, install Bed Visualizer and reboot and it pops up the anonymous tracking box.  Redeploy container, restore, disable anonymous, restart container, install bed visualizer, restart. All ok.

jneilliii, 2020-05-10:
Super weird. I'll have to test that workflow, although I'm already in the process of removing the wizard from the plugin entirely anyway for the next release.

jneilliii, 2020-07-04:
wizard will be removed in next release, so this issue will be irrelevant.

closed 2020-07-26 by jneilliii