jneilliii/OctoPrint-BedLevelVisualizer #234: Bad colors for theme

tablatronix, 2020-04-24:
Any suggestions on how to fix this bad contrast ?

[image]

jneilliii, 2020-04-25:
@LMS0815, since this was part of your changes, you got any idea on how to resolve this? You're way more familiar with this new table layout than me and might be able to answer the question faster.

jneilliii, 2020-04-25:
@LMS0815 I think is we were to flip your table to usilize bootstrap css classes it might resolve this but will need to verify.

https://getbootstrap.com/2.3.2/base-css.html#tables

tablatronix, 2020-07-03:
Anyone have tips on how to accomplish this ? It it using hardcoded css? Can I just add some bootstrap classes?

jneilliii, 2020-07-03:
I've actually starting to do that in my latest dev branch.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/tree/0.1.14

jneilliii, 2020-07-03:
Honestly, it probably just needs to be documented how to use Themeify's advanced selectors to change those colors.

jneilliii, 2020-07-04:
I just pushed updates to my 0.1.4 branch that should help with this. Convert hard-coded styling to use css class names instead. Relative css selectors for Themeify are as follows:

```
.table thead tr.info .raise, .table tbody tr .raise, .table tfoot tr.info .raise, .table tfoot tr td div span.raise {
    color: blue;
}

.table thead tr.info .lower, .table tbody tr .lower, .table tfoot tr.info .lower, .table tfoot tr td div span.lower {
    color: red;
}

.table thead tr.info .x-axis-label, .table tbody tr .x-axis-label, .table tfoot tr.info .x-axis-label, .table tfoot tr .x-axis-label {
    color: black;
}

.table thead tr.info .y-axis-label, .table tbody tr .y-axis-label, .table tfoot tr.info .y-axis-label, .table tfoot tr .y-axis-label {
    color: green;
}
```

closed 2020-07-26 by jneilliii