jneilliii/OctoPrint-BedLevelVisualizer #236: Bedlevel Bug

joloman2, 2020-04-26:
i have changed my headed bed and manuely leveled it but the plugin stil says it is not level how ca i fix this 

[image]

jneilliii, 2020-04-26:
This drastic of a skew +2/-1.5 is to much outof alignment, adjust to get those  closer and update mesh again.

joloman2, 2020-04-26:
i know that is looks lik if its out off alinement but it is not but is says it is

jneilliii, 2020-04-26:
Then please enable debug logging in the plugin's settings, restart octoprint and try again. If it still does it, upload your plugin_bedlevelvisualizer_debug.log file by dragging/dropping it into a comment here,

joloman2, 2020-04-26:
Ty for helping me it works now i reinstaled it and it works ty

closed 2020-04-26 by joloman2