jneilliii/OctoPrint-BedLevelVisualizer #238: Graph data does not align with visual bed representation

Silventar, 2020-04-28:
Please see the attached image. The graph data under the visual data does not align with the bed.

My printer has a center homeing point, so x-150,y-150 is lower left. I have the visualization oriented that way. However, the graph shows it in the upper left. This is visually confusing.

[image]

jneilliii, 2020-04-28:
If you go to the current mesh tab in the plugin's settings and enable the option `Descending y axis` does it change it to what you were expecting?

Silventar, 2020-04-28:
Yes! That is perfect. Thank you!

Shoot well!
Andrew Robertson
RobertsonArchery.com

closed 2020-04-28 by jneilliii