jneilliii/OctoPrint-BedLevelVisualizer #239: future request:

brandonscholet, 2020-04-28:
Is it possible to have it auto save a picture of each bed level. I'd like to compile a gif to see what my drift looks like over time.

Or, is historical data stored somewhere that's easily accessible

jneilliii, 2020-04-28:
There is no historical data, but the graph does have an option to save a snapshot. If you hover over the  graph and press the camera icon in the pop-up menu, it will download a png.

brandonscholet, 2020-04-28:
I am aware. Would you consider having an option to save the snapshot automatically to the system running octoprint?

jneilliii, 2020-04-28:
I'll have to think about that one, because all the graphing is done on the client side (web interface), the server side just provides the data to the client side to visualize.

jneilliii, 2020-07-26:
Would it be acceptable to automatically download the image to the browser rather than saving to the octoprint server? I think it might be possible to hook into the on rendering complete callback of plotly to spawn the download.

jneilliii, 2020-08-01:
implemented in above commit, and will be included in the next release 0.1.15. If you want to give it a try ahead of release you can install by using the URL below in Plugin Manager > Get More > ...from URL. Once the option is enabled in settings every graph rendering will generate the snapshot with a filename based on the format `YYYY-MM-DD_HH-mm-ss.png`.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.15.zip

DodgeDeBoulet, 2020-08-01:
While I very much like the idea of auto-saving, I don't think having it done as a local render then prompt-to-save is a "streamlined" experience. I do see the value of some sort of history, though. How about just auto-logging the mesh data as a time-stamped file to a user-defined location on the OctoPrint server, along with a modification to the BLV tab that allows selection of a historical mesh file for rendering?

jneilliii, 2020-08-01:
This was just the easiest way to implement the original feature request, which does exactly as requested. A historical mesh database is not a bad idea, and could coincide with changes required for #154.

DodgeDeBoulet, 2020-08-02:
> This was just the easiest way to implement the original feature request, which does exactly as requested.

Depending on the browser settings, the save may not be completely automatic. That was my reasoning behind the "streamlined" comment.

> A historical mesh database is not a bad idea, and could coincide with changes required for #154.

Thank you 😁. I think this really addresses the spirit of the original request by providing a fully-automatic mechanism for bed leveling history, assuming generation of a chart based on a historical selection is part of the solution 😉

closed 2020-08-18 by jneilliii

jneilliii, 2020-10-18:
For anyone interested I have created an auxiliary plugin that will log mesh data and allow for swapping between previous mesh data. You can install by copying/pasting the URL below into Plugin Manager > Get More > ...fromURL and clicking Install. 

```
https://github.com/jneilliii/OctoPrint-BLVLogger/archive/master.zip
```

It's still early in development and be aware that when you switch the mesh with the pull down it will store that as the last run mesh (if save mesh option is enabled in BLV). You can always select the most recent date in the drop-down and it will switch it back.