jneilliii/OctoPrint-BedLevelVisualizer #242: Provide percentages for Bed Level Correction in Prusa MK2/3

ayourk, 2020-05-09:
Any chance you could provide the percentages to use after running your plugin to set in the Mesh Level Correction function of the Prusa firmware?:

Left side [um]?
Right side [um]?
Front side [um]?
Rear side [um]?

jneilliii, 2020-05-09:
Sorry, don't use prusa. Thre correction adjustment view in the newer version though does give adjustment screw angles in degrees, is that what you mean?

jneilliii, 2020-05-09:
> Mesh Level Correction function of the Prusa firmware

Do you have a link or docs that explain how this process works?

ayourk, 2020-05-10:
Here are the official docs on what I am suggesting:
https://help.prusa3d.com/en/article/bed-level-correction_2267

I'm pretty sure that the mesh values can be used to calculate these calculation values.  I personally use a Prusa i3 MK2 with v3.0.12 firmware.

Some other links that may be useful:
https://forum.prusaprinters.org/forum/original-prusa-i3-mk2-s-others-archive/how-to-use-bed-level-correct/
https://www.thingiverse.com/thing:3008633

jneilliii, 2020-05-26:
lol there are other bugs/requests ahead of you.

ayourk, 2020-05-26:
I forgot to comment because of the stale tag, that's what the bump was for.

jneilliii, 2020-05-26:
No worries, I had already tagged the enhancement label, which removes the stale...

jneilliii, 2020-06-15:
Have you guys seen the Prusa Leveling Guide plugin? I ran across this over the weekend and seems to fit your exact needs.

https://plugins.octoprint.org/plugins/PrusaLevelingGuide/

ayourk, 2020-06-15:
It appears to be more for the MK3 users than the MK2 users.

dbrouwers, 2020-06-27:
The prusalevelinguide is only valid from mk3 onwards with the Nylock mod. 
I was asking the same question in discord, it would be great if any of these tools could add this feature for mk2 users.

jneilliii, 2020-06-27:
Understood. Would make more sense for the other plugin IMO to incorporate this since it's prusa specific mod but I will keep this feature request on the list.

jneilliii, 2020-07-04:
Do you think this would work in your situation for getting those numbers or is there some other math involved other than just the screw turns?

[image]

ayourk, 2020-07-05:
The value that is calculated is more quadrant based and is expressed in Z-axis nozzle height in microns.
[image]
I believe the units used are smaller than what is used for the Live Z adjustment.  But what you have above is a good start.  The value can be positive or negative.  The red region is the Left zone for the calculation, the green region is the back zone for the calculation.  These regions are just for illustration purposes.

jneilliii, 2020-08-02:
After going back and looking at the Thingiverse link, I think I need to clarify the feature request. Did you want to add the whole process of printing the piece, taking the measurements and enter those values or purely base the values off of the measured probe points like described below @ayourk?

Given the following set of mesh points.

```
[["-0.452","-0.319","-0.237","0.287","0.140","0.139","0.136","0.317","0.247","0.247"],
["-0.195","-0.273","-0.180","-0.178","0.014","0.018","0.111","0.214","0.210","0.210"],
["-0.270","-0.252","-0.151","-0.119","0.009","0.016","0.072","0.249","0.224","0.224"],
["-0.307","-0.205","-0.163","-0.124","-0.094","-0.002","0.036","0.151","0.174","0.196"],
["-0.186","-0.130","-0.152","-0.105","-0.144","-0.007","0.044","0.093","0.181","0.270"],
["-0.010","-0.077","-0.073","0.155","-0.006","-0.133","0.110","0.046","0.109","0.173"],
["0.059","-0.094","-0.072","-0.002","-0.006","0.037","0.050","0.065","0.124","0.184"],
["-0.057","-0.028","0.039","0.028","0.024","0.005","0.102","0.165","0.176","0.187"],
["0.067","0.015","0.096","0.117","0.001","0.079","0.138","0.346","0.185","0.185"],
["0.071","0.014","0.061","-0.127","0.167","0.040","0.098","0.195","0.194","0.194"]]
```

[image]

I'm calculating the average of each of the `zones` and getting the following.

back_half_avg:-0.0003200000000000036
front_half_avg:0.06831999999999998
left_half_avg:0.006693877551020405
right_half_avg:0.06279591836734692

I would think that those numbers would be multiplied by 1000 for the micrometer value? Resulting in the following rounded numbers.

back [um]: 0
front [um]: 68
left [um]: 7
right [um]: 63

Does that seem right or should the values be inverted?

ayourk, 2020-08-02:
That pretty much is what I'm looking for.  Front = front edge of the Y-axis=0; X-axis=0 along the left side.  But there is a small catch.  Some versions of the firmware need that value either halved or doubled (I can't remember).  The 0.0000 value is considered the be the exact center of the bed.

jneilliii, 2020-08-02:
Ok, so the math logic in theory is sound and doesn't need to be inverted? If so it's just a matter of including a multiplier option for different firmware versions.

ayourk, 2020-08-02:
Positive values mean the nozzle needs to be moved farther away from the bed.  Negative values means that the nozzle needs to be moved closer to the bed (as per the article).  I guess I'll have to test the new changes to see how it compares to my printer.

jneilliii, 2020-08-02:
I'll try to integrate the math into the plugin and upload tomorrow. I'm a little concerned the orientations of front and left might be wrong based on your description. Seems rotated to the way the mesh is reported.

dbrouwers, 2020-08-02:
This sounds promising! I will test right away!

jneilliii, 2020-08-02:
Ok, just for some quick tests I've gotten the math integrated and writing out the corrections into the developer console of the browser. I can work on integrating the information in the UI once we know this is working right. If you want to give it a test for me you can use the URL below in Plugin Manager > Get More > ...from URL and click the Install button. Let me know how off it is and hopefully it doesn't damage your bed/nozzle.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.15.zip

jneilliii, 2020-09-28:
Were any of you able to validate the numbers I was getting in the developer console for these corrections?

jonohayon, 2021-02-19:
Hi! Is there any progress update on this feature? Is there any way to help in development? I'm a MK2.5S user and would love something like this to work on my machine, was just hacking a solution with the Prusa Level Guide extension when I saw this thread

jneilliii, 2021-02-19:
I don't think the math is quite right, but if you open your browsers developer console you'll see the um values listed out when the mesh is graphed on the console tab.

jonohayon, 2021-02-20:
It should be noted that without setting the center to the origin point and the z offset checkbox to on the values are very off (got >200 while prusa allows for <100 values). After setting both I got results that even could be entered to the machine - which makes sense to me, since I think these values don't apply to the middle (that's how I see it anyway).

In any case, after the problem was resolved, I used the values given in the dev console to print [this calibration model](https://www.prusaprinters.org/prints/36706-bed-level-test) from Prusa themselves, and it turned out _great_ - one of the best first layers I've ever had on my machine.

jneilliii, 2021-02-20:
Just to confirm, you set use these two options to enabled and the um values actually worked as expected?

[image]

ayourk, 2021-02-20:
> It should be noted that without setting the center to the origin point and the z offset checkbox to on the values are very off (got >200 while prusa allows for <100 values). After setting both I got [...]

What firmware version are you using?

jonohayon, 2021-02-22:
> Just to confirm, you set use these two options to enabled and the um values actually worked as expected?
> 
> [image]

Yes, I used these values and the calibration model went as expected (at least way better than the calibration I tried to do myself).

> What firmware version are you using?

I have a Prusa i3 MK2.5S with firmware 3.9.3-3556, which is the latest one afaik.

PR #497 (2022-01-01): 1.1.1
* add ability to configure graph height in visualization settings, resolves #462
* add min, max, and variance values of graphed mesh, resolves #286
* add OctoDash support, create Custom Action with `[!WEB]http://127.0.0.1:5000/plugin/bedlevelvisualizer/bedlevelvisualizer` as the command.
* adjust the tables for better theme support, resolves #479
* add option for showing mesh data on tab, resolves #496
* remove tooltip hover on tables, resolves #490
* add Prusa adjustment values to graph (still needs verification), resolves #242
* add appearance > title setting to automatically downloade [... cut]
closed 2022-01-01 by jneilliii