jneilliii/OctoPrint-BedLevelVisualizer #243: Requesting new data hangs OctoPrint

Taomyn, 2020-05-10:
For some reason whenever I now try to retrieve a new mesh (Ender 3 Pro with SKR mini E3) the plugin never completes and just hangs OctoPrint, forcing me to restart the service manually - the processing timeout has no effect. I have also reset the plugin and it behaves the same.

The plugin (v0.1.3) debug log shows very little, the OctoPrint (v1.4.0) log shows nothing:

```
[2020-05-10 19:46:54,780] DEBUG: mesh collection started
[2020-05-10 19:46:54,790] DEBUG: Bilinear Leveling Grid:
[2020-05-10 19:46:54,811] DEBUG: 0      1      2      3      4
[2020-05-10 19:46:54,815] DEBUG: 0 -0.059 +0.004 +0.034 +0.052 +0.100
[2020-05-10 19:46:54,818] DEBUG: [u'-0.059', u'+0.004', u'+0.034', u'+0.052', u'+0.100']
[2020-05-10 19:46:54,822] DEBUG: 1 -0.095 -0.048 -0.002 +0.008 +0.060
[2020-05-10 19:46:54,824] DEBUG: [u'-0.095', u'-0.048', u'-0.002', u'+0.008', u'+0.060']
[2020-05-10 19:46:54,828] DEBUG: 2 -0.118 -0.050 -0.004 +0.020 +0.080
[2020-05-10 19:46:54,830] DEBUG: [u'-0.118', u'-0.050', u'-0.004', u'+0.020', u'+0.080']
[2020-05-10 19:46:54,835] DEBUG: 3 -0.082 -0.011 +0.032 +0.043 +0.075
[2020-05-10 19:46:54,837] DEBUG: [u'-0.082', u'-0.011', u'+0.032', u'+0.043', u'+0.075']
[2020-05-10 19:46:54,840] DEBUG: 4 -0.041 +0.012 +0.051 +0.054 +0.088
[2020-05-10 19:46:54,842] DEBUG: [u'-0.041', u'+0.012', u'+0.051', u'+0.054', u'+0.088']
[2020-05-10 19:46:54,846] DEBUG:
[2020-05-10 19:46:54,848] DEBUG: Subdivided with CATMULL ROM Leveling Grid:
[2020-05-10 19:46:54,850] DEBUG: resetting mesh to blank because of CATMULL subdivision
[2020-05-10 19:46:54,852] DEBUG: 0        1        2        3        4        5        6        7        8        9       10       11       12
[2020-05-10 19:46:54,855] DEBUG: 0 -0.05900 -0.03698 -0.01496 +0.00350 +0.01650 +0.02595 +0.03400 +0.03983 +0.04428 +0.05200 +0.06565 +0.08257  =======
```

I've set my commands for the plugin as, it used to be just `M420 V`

```
M155 S30
@BEDLEVELVISUALIZER
M420 V
M155 S3
```

The `M420 V` command returns

```
Send: M420 V
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 -0.059 +0.004 +0.034 +0.052 +0.100
Recv:  1 -0.095 -0.048 -0.002 +0.008 +0.060
Recv:  2 -0.118 -0.050 -0.004 +0.020 +0.080
Recv:  3 -0.082 -0.011 +0.032 +0.043 +0.075
Recv:  4 -0.041 +0.012 +0.051 +0.054 +0.088
Recv: 
Recv: Subdivided with CATMULL ROM Leveling Grid:
Recv:         0        1        2        3        4        5        6        7        8        9       10       11       12
Recv:  0 -0.05900 -0.03698 -0.01496 +0.00350 +0.01650 +0.02595 +0.03400 +0.03983 +0.04428 +0.05200 +0.06565 +0.08257  =======
Recv:  1 -0.07168 -0.05220 -0.03272 -0.01548 -0.00118 +0.01089 +0.02056 +0.02571 +0.02847 +0.03506 +0.04880 +0.06636  =======
Recv:  2 -0.08437 -0.06743 -0.05048 -0.03446 -0.01885 -0.00417 +0.00711 +0.01159 +0.01267 +0.01811 +0.03195 +0.05015  =======
Recv:  3 -0.09550 -0.07959 -0.06368 -0.04800 -0.03137 -0.01496 -0.00250 +0.00191 +0.00237 +0.00750 +0.02172 +0.04061  =======
Recv:  4 -0.10620 -0.08860 -0.07100 -0.05400 -0.03678 -0.02015 -0.00715 -0.00153 +0.00047 +0.00707 +0.02258 +0.04270  =======
Recv:  5 -0.11535 -0.09456 -0.07377 -0.05455 -0.03702 -0.02107 -0.00796 -0.00052 +0.00407 +0.01298 +0.03006 +0.05145  =======
Recv:  6 -0.11800 -0.09470 -0.07141 -0.05050 -0.03278 -0.01744 -0.00450 +0.00394 +0.01000 +0.02000 +0.03737 +0.05869  =======
Recv:  7 -0.11052 -0.08608 -0.06163 -0.03998 -0.02209 -0.00700 +0.00546 +0.01333 +0.01855 +0.02717 +0.04231 +0.06085  =======
Recv:  8 -0.09654 -0.07164 -0.04674 -0.02485 -0.00692 +0.00799 +0.01970 +0.02617 +0.02944 +0.03544 +0.04710 +0.06149  =======
Recv:  9 -0.08200 -0.05750 -0.03300 -0.01150 +0.00622 +0.02095 +0.03200 +0.03715 +0.03863 +0.04250 +0.05157 +0.06304 +0.07450
Recv: 10 -0.06870 -0.04580 -0.02290 -0.00265 +0.01449 +0.02899 +0.03963 +0.04386 +0.04423 +0.04715 +0.05560 +0.06660 +0.07759
Recv: 11 -0.05485 -0.03441 -0.01397 +0.00443 +0.02074 +0.03501 +0.04532 +0.04871 +0.04814 +0.05057 +0.05932 +0.07106 +0.08280
Recv: 12 -0.04100 -0.02302 -0.00504 +0.01150 +0.02698 +0.04102 +0.05100 +0.05356 +0.05206 +0.05400 +0.06304 +0.07552 +0.08800
Recv: 
Recv: echo:Bed Leveling ON
Recv: echo:Fade Height OFF
Recv: ok P31 B31

```

jneilliii, 2020-05-10:
Those bad data points are probably causing a problem...

```
=======
```

missed probe points, or didn't extend the mesh maybe? Not sure if Bilinear Leveling has that option.

kmobs, 2020-05-13:
I'm having the same issue. Same board. Freezes octoprint completely

kmobs, 2020-05-13:
Disabling CATMULL subdivision fixed it for me. Odd. Just accelerated my move to UBL and no longer having issues.

jneilliii, 2020-05-13:
@kmobs, thanks for that additional information. Glad UBL is working out, I have been using that since starting developing this plugin, and all additional firmware/leveling support has been kind of "figure it out" from terminal responses, etc. Were you seeing the same thing, where some CATMULL points were `=======`?

Taomyn, 2020-05-13:
Seems after a new build of Marlin bugfix-2.0.x (the previous was working fine until I reported the issue) and it's producing a proper mesh now, so it was hopefully just a bug. But this isn't the first time I've had the plug-in hang up and crash OctoPrint like this when it was fine for many days/weeks but I didn't look into it so closely that time.

Either way, receiving "=======" or any kind of non-compliant input shouldn't be causing a plug-in to hang up an entire system. I think that's the real issue.

I'm going to look into moving to UBL from ABL myself, but I want to fix the current mesh not including the whole bed first, due to the offset BLTouch and the nozzle not being allowed to go past the end of the bed when probing.

jneilliii, 2020-05-13:
> Either way, receiving "=======" or any kind of non-compliant input shouldn't be causing a plug-in to hang up an entire system. I think that's the real issue.

Agreed, that's why I'm trying to figure out if that was the issue. So for the missing points I know UBL has a gcode command to fill in missing points I think it's G29 P3 or something like that, does bilinear subdivided have something similar or no?

Taomyn, 2020-05-13:
Never knew about codes to fill in the missing data as I didn't know that could happen.

I did try looking through the Marlin code for anything that set the value to "=======" but came up empty.

kmobs, 2020-05-13:
I was also getting ======= CATMULL points. 

My UBL experiment ended early, and I'm not getting any issues with CATMULL enabled and set to 1. I haven't tried set to 3 once again since Taomyn reports that it's now working. Might have just been a marlin bug. 

Agree that bad data shouldn't crash octoprint though.

kmobs, 2020-05-13:
Taomyn - How large is your firmware? I know these boards have secret > 256kb storage, but can be prone to bugs. Maybe this is one of the bugs? I'm running passed the 256kb size (~224 is actual "maximum" storage size due to bootloader and EEPROM.

jneilliii, 2020-05-14:
What do you guys get if you run the command `M420 V`?

Taomyn, 2020-05-17:
@kmobs it's 290Kb, but it's been pretty much ok since I last reported it, so I'm hoping it was just a Marlin bug.

@jneilliii I re-ran an M420 V and this is what I am seeing

```
Send: M420 V*82
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 -0.102 -0.048 +0.032 +0.137 +0.254
Recv:  1 -0.124 -0.078 +0.001 +0.100 +0.221
Recv:  2 -0.136 -0.084 -0.015 +0.075 +0.193
Recv:  3 -0.157 -0.130 -0.081 -0.005 +0.100
Recv:  4 -0.166 -0.144 -0.090 -0.018 +0.083
Recv: 
Recv: Subdivided with CATMULL ROM Leveling Grid:
Recv:         0        1        2        3        4        5        6        7        8        9       10       11       12
Recv:  0 -0.10200 -0.08496 -0.06793 -0.04800 -0.02418 +0.00252 +0.03200 +0.06470 +0.10019 +0.13700 +0.17511 +0.21456 +0.25400
Recv:  1 -0.10968 -0.09383 -0.07797 -0.05889 -0.03530 -0.00849 +0.02091 +0.05298 +0.08765 +0.12404 +0.16242 +0.20252 +0.24261
Recv:  2 -0.11737 -0.10269 -0.08801 -0.06978 -0.04642 -0.01951 +0.00982 +0.04126 +0.07512 +0.11107 +0.14972 +0.19047 +0.23122
Recv:  3 -0.12400 -0.10987 -0.09574 -0.07800 -0.05500 -0.02839 +0.00050 +0.03117 +0.06411 +0.09950 +0.13820 +0.17935 +0.22050
Recv:  4 -0.12855 -0.11351 -0.09846 -0.08030 -0.05767 -0.03193 -0.00398 +0.02570 +0.05760 +0.09226 +0.13077 +0.17204 +0.21332
Recv:  5 -0.13205 -0.11546 -0.09887 -0.07993 -0.05781 -0.03335 -0.00668 +0.02186 +0.05262 +0.08641 +0.12452 +0.16566 +0.20680
Recv:  6 -0.13650 -0.11961 -0.10272 -0.08400 -0.06298 -0.04013 -0.01500 +0.01224 +0.04176 +0.07450 +0.11185 +0.15243 +0.19300
Recv:  7 -0.14331 -0.12866 -0.11401 -0.09754 -0.07887 -0.05837 -0.03535 -0.00982 +0.01824 +0.04957 +0.08554 +0.12479 +0.16404
Recv:  8 -0.15107 -0.13991 -0.12875 -0.11552 -0.09978 -0.08198 -0.06131 -0.03767 -0.01116 +0.01859 +0.05282 +0.09031 +0.12780
Recv:  9 -0.15750 -0.14913 -0.14076 -0.13000 -0.11626 -0.10013 -0.08100 -0.05874 -0.03348 -0.00500 +0.02785 +0.06393 +0.10000
Recv: 10 -0.16139 -0.15423 -0.14706 -0.13717 -0.12365 -0.10741 -0.08835 -0.06660 -0.04203 -0.01426 +0.01792 +0.05331 +0.08870
Recv: 11 -0.16394 -0.15729 -0.15063 -0.14083 -0.12660 -0.10923 -0.08943 -0.06768 -0.04350 -0.01613 +0.01574 +0.05080 +0.08585
Recv: 12 -0.16650 -0.16035 -0.15420 -0.14450 -0.12955 -0.11105 -0.09050 -0.06876 -0.04496 -0.01800 +0.01356 +0.04828 +0.08300
```
I've also figured out how to get the mesh out further on the X-axis to better cover the bed, so this week I will be looking at switching to UBL and giving that a go.

kmobs, 2020-05-17:
@Taomyn I'll try increasing subdivisions and trying again. No issues with 5x5 and 1 subdivision.

sshockley, 2020-05-25:
I'm trying to enable bilinear subdivision, and I am having the same problem, even with 5x5 and 1 subdivision:

```
Send: M420 V
Recv: echo:M420 V
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 +0.243 +0.025 +0.062 +0.098 +0.142
Recv:  1 +0.120 +0.020 +0.027 +0.033 +0.040
Recv:  2 -1.293 +0.040 -0.003 -0.017 -0.030
Recv:  3 +0.195 +0.053 +0.010 +0.027 -0.048
Recv:  4 +0.237 +0.070 +0.027 +0.007 -0.013
Recv: 
Recv: Subdivided with CATMULL ROM Leveling Grid:
Recv:         0        1        2        3        4
Recv:  0 +0.24250 +0.02500 +0.06250 +0.09750  =======
Recv:  1 +0.12000 +0.02000 +0.02750 +0.03250  =======
Recv:  2 -1.29250 +0.04000 -0.00250 -0.01750  =======
Recv:  3 +0.19500 +0.05250 +0.01000 +0.02750  =======
Recv:  4 +0.23750 +0.07000 +0.02750 +0.00750 -0.01250
Recv: 
Recv: echo:Bed Leveling ON
Recv: echo:Fade Height OFF
Recv: X:235.00 Y:235.00 Z:0.00 E:0.00 Count X:18800 Y:18800 Z:0
Recv: ok
```

So even with this setup it's still printing the =s.  In my case it doesn't hang Octoprint, it just prints:
`TypeError: WebGLRenderingContext.uniformMatrix4fv: Argument 3 could not be converted to any of: Float32Array, sequence.`

For reference, here's the output of 5x5 with the default 3 subdivisions:
```
Send: M420 V
Recv: echo:M420 V
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 +1.600 +1.370 +1.415 +1.447 +1.490
Recv:  1 +1.475 +1.377 +1.370 +1.377 +1.397
Recv:  2 +1.493 +1.393 +1.348 +1.340 +1.320
Recv:  3 +1.547 +1.408 +1.352 +1.368 +1.312
Recv:  4 +1.572 +1.445 +1.375 +1.352 +1.335
Recv: 
Recv: Subdivided with CATMULL ROM Leveling Grid:
Recv:         0        1        2        3        4        5        6        7        8        9       10       11       12
Recv:  0 +1.60000 +1.51315 +1.42630 +1.37000 +1.36509 +1.39074 +1.41500 +1.42639 +1.43639 +1.44750 +1.46093 +1.47546  =======
Recv:  1 +1.55306 +1.48508 +1.41711 +1.37222 +1.36593 +1.38272 +1.39917 +1.40689 +1.41427 +1.42296 +1.43397 +1.44629  =======
Recv:  2 +1.50611 +1.45702 +1.40792 +1.37444 +1.36677 +1.37470 +1.38333 +1.38740 +1.39215 +1.39843 +1.40701 +1.41712  =======
Recv:  3 +1.47500 +1.43917 +1.40333 +1.37750 +1.36778 +1.36806 +1.37000 +1.37093 +1.37352 +1.37750 +1.38324 +1.39037  =======
Recv:  4 +1.46889 +1.43751 +1.40612 +1.38194 +1.36893 +1.36312 +1.35982 +1.35800 +1.35868 +1.36019 +1.36223 +1.36510  =======
Recv:  5 +1.47861 +1.44606 +1.41352 +1.38722 +1.37026 +1.35956 +1.35213 +1.34809 +1.34733 +1.34648 +1.34441 +1.34225  =======
Recv:  6 +1.49250 +1.45713 +1.42176 +1.39250 +1.37204 +1.35769 +1.34750 +1.34269 +1.34204 +1.34000 +1.33426 +1.32713  =======
Recv:  7 +1.50917 +1.46936 +1.42955 +1.39667 +1.37348 +1.35723 +1.34648 +1.34389 +1.34680 +1.34593 +1.33688 +1.32404  =======
Recv:  8 +1.53000 +1.48410 +1.43820 +1.40083 +1.37537 +1.35845 +1.34852 +1.34958 +1.35763 +1.35907 +1.34718 +1.32868  =======
Recv:  9 +1.54750 +1.49769 +1.44787 +1.40750 +1.38028 +1.36250 +1.35250 +1.35491 +1.36509 +1.36750 +1.35435 +1.33343 +1.31250
Recv: 10 +1.55806 +1.50852 +1.45897 +1.41833 +1.39006 +1.37068 +1.35870 +1.35812 +1.36493 +1.36565 +1.35375 +1.33576 +1.31778
Recv: 11 +1.56528 +1.51819 +1.47111 +1.43167 +1.40285 +1.38168 +1.36685 +1.36096 +1.36140 +1.35907 +1.35002 +1.33821 +1.32639
Recv: 12 +1.57250 +1.52787 +1.48324 +1.44500 +1.41565 +1.39268 +1.37500 +1.36380 +1.35787 +1.35250 +1.34630 +1.34065 +1.33500
Recv: 
Recv: echo:Bed Leveling ON
Recv: echo:Fade Height OFF
Recv: ok
```

With this config I get no error, but Bed Visualizer never returns when I do "update mesh now".  Octoprint still behaves normally otherwise.

It looks like the Marlin code is in Marlin/src/feature/bedlevel/bedlevel.cpp at line 205 (in 2.0.x):
'''
          #ifdef SCAD_MESH_OUTPUT
            for (uint8_t i = 3; i < precision + 3; i++)
              SERIAL_CHAR(' ');
            SERIAL_ECHOPGM("NAN");
          #else
            LOOP_L_N(i, precision + 3)
              SERIAL_CHAR(i ? '=' : ' ');
          #endif
'''
So it looks like the =s should possibly be treated the same as NAN.

sshockley, 2020-05-25:
This may be related to [this bug](https://github.com/MarlinFirmware/Marlin/issues/16678.)

jneilliii, 2020-05-25:
Yeah, the `=======` values is definitely what's causing the parsing issue within the plugin. Thanks for that link, I'll take a look.

sshockley, 2020-05-25:
Confirmed; went back to 5x5 with 3 subdivisions and verified M501/M500 removed the =s.  However, now Bed Visualizer gives the error `TypeError: t is null` despite having a 13x13 CATMULL grid with all valid values.

jneilliii, 2020-05-25:
Can you share your output of the `M420 V` please @sshockley?

sshockley, 2020-05-26:
Hm, I just tried retrieving the current data with the plugin and it was successful.

Here's the data from when it showed the error:
```
Send: M420 V
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 +0.257 +0.025 +0.070 +0.112 +0.148
Recv:  1 +0.125 +0.040 +0.027 +0.035 +0.045
Recv:  2 +0.150 +0.040 +0.013 -0.015 -0.030
Recv:  3 +0.205 +0.060 +0.007 +0.025 -0.045
Recv:  4 +0.233 +0.075 +0.033 +0.005 -0.010
Recv: 
Recv: Subdivided with CATMULL ROM Leveling Grid:
Recv:         0        1        2        3        4        5        6        7        8        9       10       11       12
Recv:  0 +0.25750 +0.16972 +0.08194 +0.02500 +0.01954 +0.04491 +0.07000 +0.08463 +0.09898 +0.11250 +0.12472 +0.13611 +0.14750
Recv:  1 +0.20750 +0.14107 +0.07463 +0.03056 +0.02349 +0.03879 +0.05481 +0.06476 +0.07544 +0.08565 +0.09485 +0.10358 +0.11231
Recv:  2 +0.15750 +0.11241 +0.06732 +0.03611 +0.02745 +0.03267 +0.03963 +0.04489 +0.05189 +0.05880 +0.06497 +0.07105 +0.07713
Recv:  3 +0.12500 +0.09398 +0.06296 +0.04000 +0.02972 +0.02750 +0.02750 +0.02843 +0.03157 +0.03500 +0.03815 +0.04157 +0.04500
Recv:  4 +0.12056 +0.09161 +0.06266 +0.04037 +0.02869 +0.02366 +0.02009 +0.01638 +0.01412 +0.01296 +0.01316 +0.01445 +0.01574
Recv:  5 +0.13361 +0.09946 +0.06531 +0.03907 +0.02596 +0.02076 +0.01574 +0.00773 -0.00011 -0.00602 -0.00878 -0.00962 -0.01046
Recv:  6 +0.15000 +0.11028 +0.07056 +0.04000 +0.02472 +0.01861 +0.01250 +0.00287 -0.00676 -0.01500 -0.02093 -0.02546 -0.03000
Recv:  7 +0.16713 +0.12338 +0.07963 +0.04537 +0.02613 +0.01637 +0.00898 +0.00312 -0.00038 -0.00611 -0.01635 -0.02882 -0.04130
Recv:  8 +0.18759 +0.13945 +0.09130 +0.05296 +0.02902 +0.01487 +0.00657 +0.00715 +0.01357 +0.01278 -0.00198 -0.02395 -0.04593
Recv:  9 +0.20500 +0.15324 +0.10148 +0.06000 +0.03306 +0.01639 +0.00750 +0.01139 +0.02306 +0.02500 +0.00815 -0.01843 -0.04500
Recv: 10 +0.21620 +0.16226 +0.10831 +0.06537 +0.03852 +0.02268 +0.01361 +0.01471 +0.02258 +0.02278 +0.00795 -0.01454 -0.03704
Recv: 11 +0.22435 +0.16900 +0.11365 +0.07019 +0.04514 +0.03199 +0.02306 +0.01823 +0.01763 +0.01389 +0.00351 -0.01000 -0.02352
Recv: 12 +0.23250 +0.17574 +0.11898 +0.07500 +0.05176 +0.04130 +0.03250 +0.02176 +0.01269 +0.00500 -0.00093 -0.00546 -0.01000
Recv: 
Recv: echo:Bed Leveling OFF
Recv: echo:Fade Height OFF
Recv: ok
```

Here's the current data that succeeds:
```
Send: M420 V
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 +0.162 +0.015 +0.090 +0.122 +0.122
Recv:  1 +0.092 +0.077 +0.102 +0.130 +0.102
Recv:  2 +0.140 +0.102 +0.108 +0.112 +0.082
Recv:  3 +0.170 +0.115 +0.090 +0.128 +0.047
Recv:  4 +0.148 +0.072 +0.080 +0.053 +0.015
Recv: 
Recv: Subdivided with CATMULL ROM Leveling Grid:
Recv:         0        1        2        3        4        5        6        7        8        9       10       11       12
Recv:  0 +0.16250 +0.10509 +0.04769 +0.01500 +0.02509 +0.05991 +0.09000 +0.10519 +0.11565 +0.12250 +0.12491 +0.12370 +0.12250
Recv:  1 +0.13481 +0.09655 +0.05829 +0.03722 +0.04578 +0.07154 +0.09444 +0.10838 +0.11946 +0.12593 +0.12564 +0.12074 +0.11583
Recv:  2 +0.10713 +0.08801 +0.06889 +0.05944 +0.06647 +0.08318 +0.09889 +0.11158 +0.12328 +0.12935 +0.12638 +0.11777 +0.10917
Recv:  3 +0.09250 +0.08602 +0.07954 +0.07750 +0.08278 +0.09250 +0.10250 +0.11352 +0.12481 +0.13000 +0.12491 +0.11370 +0.10250
Recv:  4 +0.10028 +0.09552 +0.09076 +0.08907 +0.09241 +0.09883 +0.10556 +0.11354 +0.12182 +0.12481 +0.11887 +0.10763 +0.09639
Recv:  5 +0.12111 +0.11157 +0.10203 +0.09648 +0.09767 +0.10285 +0.10778 +0.11229 +0.11655 +0.11685 +0.11063 +0.10046 +0.09028
Recv:  6 +0.14000 +0.12593 +0.11185 +0.10250 +0.10102 +0.10426 +0.10750 +0.11046 +0.11343 +0.11250 +0.10509 +0.09380 +0.08250
Recv:  7 +0.15324 +0.13733 +0.12142 +0.10963 +0.10388 +0.10225 +0.10306 +0.10885 +0.11708 +0.11843 +0.10749 +0.08967 +0.07185
Recv:  8 +0.16454 +0.14704 +0.12954 +0.11537 +0.10484 +0.09764 +0.09611 +0.10666 +0.12288 +0.12796 +0.11258 +0.08606 +0.05954
Recv:  9 +0.17000 +0.15056 +0.13111 +0.11500 +0.10213 +0.09259 +0.09000 +0.10222 +0.12139 +0.12750 +0.10954 +0.07852 +0.04750
Recv: 10 +0.16639 +0.14431 +0.12224 +0.10491 +0.09393 +0.08770 +0.08611 +0.09424 +0.10702 +0.10917 +0.09203 +0.06426 +0.03648
Recv: 11 +0.15694 +0.13188 +0.10681 +0.08870 +0.08206 +0.08237 +0.08306 +0.08402 +0.08536 +0.08083 +0.06639 +0.04606 +0.02574
Recv: 12 +0.14750 +0.11944 +0.09139 +0.07250 +0.07019 +0.07704 +0.08000 +0.07380 +0.06370 +0.05250 +0.04074 +0.02787 +0.01500
Recv: 
Recv: echo:Bed Leveling ON
Recv: echo:Fade Height OFF
Recv: ok
```

jneilliii, 2020-05-26:
Yeah, at quick glance I don't see anything in the first one there that would cause the plugin to not graph.

sshockley, 2020-05-28:
Fair enough, I can no longer repro.  Cosmic rays, maybe.

jneilliii, 2020-05-28:
So I've made some progress on this `======` issue, just need to get a couple of more tweaks done before release. I have to say @Taomyn your bed is pretty even from a flatness perspective.

[image]

I hope to have the update ready this weekend.

Taomyn, 2020-06-01:
@jneilliii look forward to it, although I've never seen it look like that, this is the current plot I have and they've always looked similar:

[image]

I'm sure that tilting just one side should lessen the red areas, but it never seem to do it no matter what and simply makes the bed on that side too low. But prints are mostly fine these days so I don't worry so much.

jneilliii, 2020-06-01:
Yeah, the current release uses that auto generated color scale, so it always looks like your bed is off even though the difference between the lowest point and highest point might only be .1 mm. It caused confusion for some users.

maciej-rutkowski-intive, 2020-06-21:
Hi! I'm getting the same issue currently. After having increased the number of points for bilinear leveling to 80 and enabling catmull-rom subdivision, the plugin will hang octoprint, keeping it at 100% CPU until it times out after a loooong try.

The host is a Raspberry Pi 3 with some other services running alongside Octoprint, like Mozilla IoT gateway and Homebridge, so pretty heavy stuff for sure.

The offending conversation with the printer seemed to go well though.

```

Send: M155 S30
Recv: ok
Send: M420 V
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4      5      6      7
Recv:  0 +0.693 +0.704 +0.592 +0.550 +0.461 +0.365 +0.262 +0.185
Recv:  1 +0.492 +0.517 +0.463 +0.427 +0.398 +0.338 +0.249 +0.233
Recv:  2 +0.284 +0.270 +0.220 +0.199 +0.130 +0.071 -0.019 -0.059
Recv:  3 +0.336 +0.329 +0.278 +0.235 +0.179 +0.112 +0.026 -0.033
Recv:  4 +0.229 +0.258 +0.235 +0.234 +0.176 +0.153 +0.103 +0.098
Recv:  5 +0.180 +0.205 +0.171 +0.154 +0.138 +0.103 +0.047 +0.038
Recv:  6 +0.218 +0.260 +0.199 +0.194 +0.140 +0.113 +0.058 +0.035
Recv:  7 +0.057 +0.102 +0.105 +0.121 +0.126 +0.109 +0.087 +0.109
Recv:  8 +0.133 +0.166 +0.137 +0.166 +0.132 +0.132 +0.100 +0.110
Recv:  9 +0.165 +0.205 +0.188 +0.185 +0.180 +0.159 +0.104 +0.118
Recv: 
Recv: Subdivided with CATMULL ROM Leveling Grid:
Recv:         0        1        2        3        4        5        6        7        8        9       10       11       12       13       14       15       16       17       18       19       20       21
Recv:  0 +0.69350 +0.70153 +0.70956 +0.70400 +0.67322 +0.62886 +0.59225 +0.57484 +0.56519 +0.55025 +0.52415 +0.49277 +0.46075 +0.42975 +0.39811 +0.36550 +0.33045 +0.29444 +0.26150 +0.23403 +0.20964 +0.18525
Recv:  1 +0.62669 +0.63649 +0.64629 +0.64408 +0.62002 +0.58395 +0.55346 +0.53723 +0.52659 +0.51313 +0.49373 +0.47152 +0.44745 +0.42182 +0.39433 +0.36509 +0.33190 +0.29696 +0.26701 +0.24597 +0.22992 +0.21387
Recv:  2 +0.55987 +0.57144 +0.58301 +0.58417 +0.56681 +0.53903 +0.51468 +0.49962 +0.48799 +0.47601 +0.46332 +0.45027 +0.43416 +0.41389 +0.39055 +0.36469 +0.33334 +0.29948 +0.27252 +0.25791 +0.25020 +0.24249
Recv:  3 +0.49225 +0.50361 +0.51497 +0.51750 +0.50463 +0.48293 +0.46325 +0.44965 +0.43807 +0.42725 +0.41815 +0.40980 +0.39800 +0.38125 +0.36106 +0.33775 +0.30774 +0.27462 +0.24950 +0.23875 +0.23600 +0.23325
Recv:  4 +0.41365 +0.42086 +0.42808 +0.42827 +0.41607 +0.39683 +0.37964 +0.36844 +0.35927 +0.34932 +0.33811 +0.32612 +0.31197 +0.29565 +0.27717 +0.25522 +0.22565 +0.19261 +0.16731 +0.15608 +0.15259 +0.14909
Recv:  5 +0.33424 +0.33533 +0.33642 +0.33229 +0.31853 +0.29956 +0.28339 +0.27486 +0.26915 +0.25976 +0.24332 +0.22322 +0.20308 +0.18534 +0.16757 +0.14617 +0.11651 +0.08322 +0.05660 +0.04231 +0.03469 +0.02707
Recv:  6 +0.28375 +0.28059 +0.27744 +0.27025 +0.25529 +0.23630 +0.22050 +0.21319 +0.20906 +0.19950 +0.17936 +0.15378 +0.12950 +0.11041 +0.09262 +0.07125 +0.04161 +0.00839 -0.01925 -0.03639 -0.04794 -0.05950
Recv:  7 +0.28757 +0.28361 +0.27964 +0.27180 +0.25665 +0.23763 +0.22115 +0.21171 +0.20481 +0.19333 +0.17313 +0.14834 +0.12431 +0.10411 +0.08468 +0.06207 +0.03241 -0.00043 -0.02869 -0.04826 -0.06327 -0.07827
Recv:  8 +0.32031 +0.31742 +0.31453 +0.30729 +0.29231 +0.27299 +0.25516 +0.24213 +0.23060 +0.21647 +0.19772 +0.17636 +0.15442 +0.13320 +0.11139 +0.08704 +0.05745 +0.02532 -0.00326 -0.02507 -0.04333 -0.06159
Recv:  9 +0.33550 +0.33478 +0.33406 +0.32850 +0.31459 +0.29585 +0.27800 +0.26377 +0.25043 +0.23550 +0.21801 +0.19894 +0.17875 +0.15796 +0.13606 +0.11200 +0.08369 +0.05320 +0.02575 +0.00406 -0.01459 -0.03325
Recv: 10 +0.30968 +0.31283 +0.31597 +0.31397 +0.30298 +0.28683 +0.27191 +0.26173 +0.25278 +0.24085 +0.22342 +0.20300 +0.18301 +0.16547 +0.14835 +0.12893 +0.10438 +0.07752 +0.05406 +0.03705 +0.02343 +0.00981
Recv: 11 +0.26630 +0.27442 +0.28253 +0.28517 +0.27799 +0.26532 +0.25465 +0.25091 +0.24917 +0.24209 +0.22453 +0.20162 +0.18149 +0.16866 +0.15860 +0.14585 +0.12626 +0.10397 +0.08596 +0.07597 +0.07026 +0.06455
Recv: 12 +0.22925 +0.24085 +0.25245 +0.25825 +0.25353 +0.24300 +0.23500 +0.23530 +0.23812 +0.23450 +0.21799 +0.19504 +0.17600 +0.16662 +0.16116 +0.15275 +0.13647 +0.11725 +0.10300 +0.09802 +0.09801 +0.09800
Recv: 13 +0.20538 +0.21731 +0.22924 +0.23523 +0.23055 +0.21992 +0.21164 +0.21079 +0.21228 +0.20908 +0.19650 +0.17922 +0.16430 +0.15582 +0.14971 +0.14074 +0.12448 +0.10536 +0.09154 +0.08748 +0.08872 +0.08995
Recv: 14 +0.18784 +0.19861 +0.20938 +0.21410 +0.20810 +0.19604 +0.18589 +0.18149 +0.17901 +0.17483 +0.16736 +0.15819 +0.14862 +0.13980 +0.13056 +0.11870 +0.10081 +0.08029 +0.06521 +0.06006 +0.06035 +0.06063
Recv: 15 +0.18000 +0.19054 +0.20107 +0.20500 +0.19724 +0.18287 +0.17050 +0.16367 +0.15883 +0.15400 +0.14916 +0.14431 +0.13750 +0.12823 +0.11699 +0.10325 +0.08422 +0.06269 +0.04650 +0.04003 +0.03889 +0.03775
Recv: 16 +0.19358 +0.20624 +0.21890 +0.22321 +0.21249 +0.19343 +0.17774 +0.17090 +0.16744 +0.16268 +0.15486 +0.14574 +0.13590 +0.12619 +0.11576 +0.10261 +0.08358 +0.06182 +0.04482 +0.03668 +0.03329 +0.02991
Recv: 17 +0.21686 +0.23257 +0.24827 +0.25345 +0.23934 +0.21470 +0.19534 +0.18961 +0.18915 +0.18477 +0.17150 +0.15431 +0.13885 +0.12861 +0.12011 +0.10856 +0.09001 +0.06841 +0.05076 +0.04071 +0.03461 +0.02851
Recv: 18 +0.21800 +0.23581 +0.25361 +0.26000 +0.24530 +0.21918 +0.19925 +0.19528 +0.19750 +0.19425 +0.17881 +0.15789 +0.14000 +0.13000 +0.12303 +0.11300 +0.09570 +0.07535 +0.05850 +0.04850 +0.04200 +0.03550
[... cut ...]
Recv: 26 +0.15581 +0.17035 +0.18488 +0.19282 +0.18863 +0.17785 +0.17056 +0.17227 +0.17747 +0.17973 +0.17576 +0.16886 +0.16248 +0.15924 +0.15653 +0.15000 +0.13472 +0.11562 +0.10311 +0.10288 +0.10924 +0.11560
Recv: 27 +0.16475 +0.18007 +0.19540 +0.20450 +0.20275 +0.19478 +0.18825 +0.18628 +0.18575 +0.18500 +0.18396 +0.18270 +0.17975 +0.17533 +0.16922 +0.15925 +0.14082 +0.11854 +0.10400 +0.10367 +0.11108 +0.11850
Recv: 
Recv: echo:Bed Leveling OFF
Recv: echo:Fade Height OFF
Recv: ok
Send: M155 S3
Recv: ok

```

Update: this time it worked immediately. 🤷‍♂️😀

zanovis, 2020-06-30:
I'm also having this issue. It's stuck on a data acquisition loop, "Please wait, retrieving current mesh". This is on an Ender 3 and RPi 4, after mesh bed leveling. The gcode used is G29 T. The terminal's outputting a 3x3 matrix, but BLV doesn't seem to be receiving it.

kmobs, 2020-06-30:
@zanovis Try running an m500 and run it again.

zanovis, 2020-06-30:
Tried that, no luck. 

This is the output after mesh bed leveling:

```
Recv:  T:27.16 /0.00 B:30.99 /0.00 @:0 B@:0
Send: M500
Recv: echo:Settings Stored (640 bytes; crc 24259)
Recv: ok
Send: G29 T
Recv: Mesh Bed Leveling ON
Recv: 3x3 mesh. Z offset: 0.00000
Recv: Measured points:
Recv:         0        1        2
Recv:  0 +0.12500 +0.07500 +0.22500
Recv:  1 +0.27500 +0.05000 +0.05000
Recv:  2 +0.27500 +0.15000 -0.02500
Recv: 
Recv: X:0.00 Y:0.00 Z:-0.11 E:0.00 Count X:0 Y:0 Z:1
Recv: ok
```

jneilliii, 2020-06-30:
For anybody that is still experiencing issues, please enable debug logging, restart octoprint, try the process again and if it still isn't working as expected upload your `plugin_bedlevelvisualizer_debug.log` file from OctoPrint's logging section in settings. Along with that share a screenshot of your settings. Until this happens, I'm kind of flying blind so to say and can't really assist you with your issue.

zanovis, 2020-07-01:
> For anybody that is still experiencing issues, please enable debug logging, restart octoprint, try the process again and if it still isn't working as expected upload your [...]

I enabled the debug logging, tried the process and it worked. Not sure what was different this time. Thanks!

closed 2020-07-25 by stale[bot]