jneilliii/OctoPrint-BedLevelVisualizer #245: Stuck on "Please wait, retrieving current mesh."

ClaudioIbarra, 2020-05-15:
When I press "Update Mesh Now", the printer (CR10s Pro V2, Marlin installed from Tiny Machines) goes through a 5x5 point probe process. It finishes, and I see "Please wait, retrieving current mesh" until it times out after 1800 seconds, but I never get a graph.

Please help? And I apologize if this is not the best place to ask.

Here's the GCODE for the Mesh Update Process in the settings:

G28
M155 S30
@BEDLEVELVISUALIZER
G29 T
M155 S3

Here's what I see in Terminal after sending "G29 T" manually:

Recv: Not SD printing
Recv:  T:21.41 /0.00 B:22.47 /0.00 @:0 B@:0
Recv: wait
Recv: Not SD printing
Recv: wait
Recv: Not SD printing
Recv: wait
Recv: Not SD printing
Recv:  T:21.52 /0.00 B:22.53 /0.00 @:0 B@:0
Recv: wait
Recv: Not SD printing
Recv: wait
Recv: Not SD printing
Recv: wait
Recv: Not SD printing
Recv:  T:21.52 /0.00 B:22.50 /0.00 @:0 B@:0
Recv: wait
Recv: Not SD printing
Send: G29 T
Recv: Not SD printing
Recv: Not SD printing
Recv: echo:busy: processing
Recv:  T:21.64 /0.00 B:22.50 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: Not SD printing
Recv: echo:busy: processing
Recv: Not SD printing
Recv:  T:21.76 /0.00 B:22.50 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: echo:busy: processing
Recv: Not SD printing
Recv: Not SD printing
Recv: echo:busy: processing
Recv:  T:21.64 /0.00 B:22.50 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: Not SD printing
Recv: echo:busy: processing
Recv: Not SD printing
Recv:  T:21.68 /0.00 B:22.50 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: echo:busy: processing
Recv: Not SD printing
Recv: Not SD printing
Recv: echo:busy: processing
Recv:  T:21.64 /0.00 B:22.53 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: Not SD printing
Recv: echo:busy: processing
Recv: Not SD printing
Recv:  T:21.60 /0.00 B:22.56 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: echo:busy: processing
Recv: Not SD printing
Recv: Not SD printing
Recv: echo:busy: processing
Recv:  T:21.64 /0.00 B:22.50 /0.00 @:0 B@:0
[... 350 lines cut ...]
Recv: X:312.00 Y:294.00 Z:5.00 E:3976.69 Count X:24960 Y:23520 Z:2000
Recv: //action:prompt_end
Recv: //action:notification old
Recv: 
Recv: ok P15 B3
Recv: wait
Recv: Not SD printing
Recv: wait
Recv:  T:21.60 /0.00 B:22.50 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: wait
Recv: Not SD printing
Recv: wait
Recv: Not SD printing
Recv: wait
Recv:  T:21.60 /0.00 B:22.50 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: wait
Recv: Not SD printing
Recv: wait
Recv: Not SD printing
Recv: wait
Recv:  T:21.56 /0.00 B:22.47 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: wait

jneilliii, 2020-05-15:
What is your processing timeout value set at?

ClaudioIbarra, 2020-05-15:
1800 secs

jneilliii, 2020-05-15:
Can you try restarting OctoPrint for me and see if it works afterwards. Sometimes the settings don't take for some reason when you first put them in.

ClaudioIbarra, 2020-05-15:
Restarting now, and I'll try updating the visualizer mesh again afterwards. Would it help to copy what appears in Terminal again when I do?

ClaudioIbarra, 2020-05-15:
SUCCESS! Thank you! In the end all I had to do was turn it off and on? That's the most basic fix that I should have thought of first. Thank you so much.

[image]

Is this the kind of mesh I should see with a glass bed?

closed 2020-05-15 by ClaudioIbarra

jneilliii, 2020-05-15:
Yeah, fairly common. There are inconsistencies relative to the probes themselves, but your variance is roughly .2mm across the bed based on the heat bar on the right. That is pretty darn good.