jneilliii/OctoPrint-BedLevelVisualizer #246: Adjustment Angle axis causes confusion

b4545, 2020-05-15:
[image]

It took me forever to realize that your adjustment angle table does not start with 0,0 on the bottom left like one might assume, but instead has the Y origin on the top row.

Can you please please put the Y origin on the bottom row?

b4545, 2020-05-15:
Also what would be really nice is an option to display number of turns instead of degrees. So instead of 400 degrees it would be 400/360 = 1.11 turns

jneilliii, 2020-05-15:
There's an option for that in settings...

[image]

And there is an option that kind of shows that math, but it's was buggy and currently hidden from view until I can figure it out.

closed 2020-05-15 by jneilliii