jneilliii/OctoPrint-BedLevelVisualizer #251: Prusa MKS3 and "Please wait, retrieving current mesh."

JeffB42, 2020-05-23:
I have a Prusa MKS3 and I'm getting stuck with the "Please wait, retrieving current mesh." message.  I read the thread at https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/issues/4, but it's was closed a while ago.

I'm running OctoPrint 1.4.0 running on OctoPi 0.17.0
Bed Visualizer is 0.1.13

GCode commands are:
[image]

...and Terminal Auto Uppercase Blacklist are:
[image]

In Bed Visualizer, when I click the Update Mesh Now button, I get this in the terminal:
Send: G80
Recv: echo:Enqueing to the front: "G28 W0"
Recv: tmc2130_home_enter(axes_mask=0x01)
Recv:   0 step=61 mscnt= 989
Recv: tmc2130_goto_step 0 61 2 1000
Recv: tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x01
Recv: tmc2130_home_enter(axes_mask=0x02)
Recv: echo:busy: processing
Recv:   0 step=17 mscnt= 287
Recv: tmc2130_goto_step 1 1 2 1000
Recv: tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x02
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: ok
Send: G81
Recv: Num X,Y: 7,7
Recv: Z search height: 5.00
Recv: Measured points:
Recv:   0.10000  0.11176  0.10731  0.08667  0.04981  -0.00324  -0.07250
Recv:   0.07778  0.09391  0.09613  0.08444  0.05885  0.01934  -0.03407
Recv:   0.06639  0.08521  0.09243  0.08806  0.07209  0.04453  0.00537
Recv:   0.06583  0.08565  0.09620  0.09750  0.08954  0.07231  0.04583
Recv:   0.07611  0.09524  0.10746  0.11278  0.11119  0.10271  0.08731
Recv:   0.09722  0.11397  0.12619  0.13389  0.13706  0.13570  0.12981
Recv:   0.12917  0.14185  0.15241  0.16083  0.16713  0.17130  0.17333
Recv: ok

I've read through some previous discussion on this, and I think I've applies all of the changes needed, yet I'm still getting stuck with the "Please wait, retrieving current mesh" message.  Any other suggestions to try, or is there any other additional information about my set up I can provide?

Thanks,

Jeff

ctgreybeard, 2020-05-24:
I am interested in this also. I have a Prusa i3 MK2S and see the same in the terminal window. It does look like the G81 is responding correctly. I will probably dig  into it a bit on my own. I'll let you know if I find anything.

jneilliii, 2020-05-24:
You guys might want tti check out the examples. 

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/wiki/Firmware-Specific-GCODE-Examples

JeffB42, 2020-05-25:
Thanks @jneilliii, I was (only) using G80 and G81.  I just changed it to:
G80
@BEDLEVELVISUALIZER
G81

...and it worked.

closed 2020-05-25 by JeffB42