jneilliii/OctoPrint-BedLevelVisualizer #253: Plugin removes the fontawesome trash icon in octoprint and other plugins

lciscon, 2020-05-24:
Ok this is a bit bizarre.  I believe this issue relates to the fact that several octoprint plugins (and octoprint itself) all use the same fontawesome icons, and they use the same class names (e.g. "fa-trash-o") to refer to them.  I think this is causing conflicts if one plugin overrides the class definition with something different than the others.   And I think the Bed Level Visualizer is doing just that.  Here is what happens:

Issue Description:

When the Bed Level Visualizer is installed and enabled, the fontawesome icons in other octoprint plugins are modified or removed.  Specifically the trash icon disappears.  

How to reproduce:

1. Install and set up TouchUI and reboot/start as specified in their instructions.
2. Install Bed Level Visualizer
3. Reboot and get everything running in TouchUI mode.
4. Go to the Temperature screen.   Look at the bottom right corner.  The icons are missing, but the buttons still function correctly.  
5. Disable or remove the Bed Level Visualizer plugin
6. Return to the Temperature screen.  The icons are now visible.

The same behavior holds true for ALL trash icons located anywhere in octoprint, TouchUI, or other plugins.

lciscon, 2020-05-24:
BTW if TouchUI is not active it does not affect the default octoprint trash icon.  I believe this is because of the loading order of the built-in octoprint resources versus the plugins.

jneilliii, 2020-05-24:
Pretty sure you don't have this issue with just my plugin installed do you? If not then the issue would be with touch ui. I've included the fontawesome v4 shim file, which is supposed to handle this if I'm not mistaken.

jneilliii, 2020-05-24:
Just double checked and with just BLV plugin installed I get this, does the issue only show in the TouchUI interface when it's enabled or in the normal OctoPrint UI as well? 

[image]

lciscon, 2020-05-24:
I have seen it with other plugins that are not TouchUI as well.  I'll get you another sample case.   As I mentioned it appears to be load-order-dependent, so it does not happen in all cases.

Is there anything special you're doing with fontawesome icons in general or the trash icon specifically?

jneilliii, 2020-05-24:
I'm checking into the shim file to see if that icon is there or not. If not, it's just a matter of adding it I think and pointing to the correct icon.

jneilliii, 2020-05-24:
This is in the shim file...
```
.fa.fa-trash-o{font-family:"Font Awesome 5 Free";font-weight:400}.fa.fa-trash-o:before{content:"\f2ed"}
```

jneilliii, 2020-05-24:
and it appears the font awesome 5 library is defining two like this, so seems it's using the same as the alt version.
```
.fa-trash:before{content:"\f1f8"}
.fa-trash-alt:before{content:"\f2ed"}
```

jneilliii, 2020-05-24:
Doing a quick debugging before heading out and it looks like TouchUI's css file is specifying the font-family [here](https://github.com/BillyBlaze/OctoPrint-TouchUI/blob/c43482205ab38903264261dcde88822b8d09991f/octoprint_touchui/static/less/touchui.bundled.less#L559) which is causing the icon to not show if I'm not mistaken. That is causing `fontawesome 5 free` assignment to be overwritten when loaded.

jneilliii, 2020-05-24:
I think I have a possible solution by adding `!important` to my v4 shim file for this icon. My local testing seems to work.

lciscon, 2020-05-24:
To me it looks like the old conflicting-DLLs-problem.  Some of the plugins are using Version 4 and some are using Version 5.  And since everything is using the same class namespace for these, whichever loads last wins. I'm not sure what the right answer is.   Hopefully there is an easy fix.

lciscon, 2020-05-24:
> I think I have a possible solution by adding `!important` to my v4 shim file for this icon. My local testing seems to work.

Cool.  Fingers crossed!

jneilliii, 2020-05-24:
You are more than happy to try it by installing the plugin using this URL in plugin manager.

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/0.1.14.zip

lciscon, 2020-05-24:
Will do! Thanks!!!!

jneilliii, 2020-05-24:
Cool, let me know and maybe you'd consider a tip or patreon sponsorship for my extremely quick response. At least "this guy" is trying to help... lol.

lciscon, 2020-05-24:
It works! Thanks!

lciscon, 2020-05-24:
How do I do the tip/sponsorship thing?  I'd like to contribute.

jneilliii, 2020-05-24:
Great news, that change is queued up now and go out in the next official release. There's a PayPal link in the readme, I haven't yet updated it with my Patreon info, but that's [here](https://www.patreon.com/jneilliii).

lciscon, 2020-05-26:
Paypal sent!  Thanks again!

closed 2020-07-26 by jneilliii