jneilliii/OctoPrint-BedLevelVisualizer #255: Issue installing

nick-harper, 2020-05-28:
When I try to install this to use with my Ender 3, it just restarts Octoprint in Safe Mode.

It doesn't show anything in the logs that make it obvious either.

`2020-05-27 23:33:48,000 - octoprint.plugins.pluginmanager - INFO - Installing plugin from https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zip
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                 2020-05-27 23:34:26,945 - octoprint.startup - INFO - ******************************************************************************
2020-05-27 23:34:26,946 - octoprint.startup - INFO - Starting OctoPrint 1.4.0`

nick-harper, 2020-05-28:
It seems to get as far as this:

Building wheel for numpy (setup.py): started

jneilliii, 2020-05-28:
Yeah, start the install and don't touch it until it's done. The process can take more than an hour on lower end devices, less on 3b and 4 but will takes time to complete.

nick-harper, 2020-05-28:
It is using a 4.

It seems to popup saying "reload" halfway through. Do I just leave it without reloading?

nick-harper, 2020-05-28:
After about 3 minutes I get:

Server is offline
The server appears to be offline, at least I'm not getting any response from it. I'll try to reconnect automatically over the next couple of minutes, however you are welcome to try a manual reconnect anytime using the button below.

Do I just leave it to carry on?

nick-harper, 2020-05-28:
`2020-05-28 08:17:17,964 - octoprint.util.comm - INFO - Printer reports firmware name "Marlin TH3D U1.R2.B2"
2020-05-28 08:17:19,656 - octoprint.plugins.pluginmanager - INFO - Installing plugin from https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zip
2020-05-28 08:17:22,533 - octoprint.server.preemptive_cache - INFO - ... done in 4.66s
2020-05-28 08:17:22,534 - octoprint.server.preemptive_cache - INFO - Preemptively caching / (ui _default) for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'http://nick-pi.ddns.net:8000/'}
2020-05-28 08:17:23,513 - octoprint.server.preemptive_cache - INFO - ... done in 0.98s
2020-05-28 08:17:23,513 - octoprint.server.preemptive_cache - INFO - Preemptively caching / (ui _default) for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'https://octopi.local/'}
2020-05-28 08:17:24,534 - octoprint.server.preemptive_cache - INFO - ... done in 1.02s
2020-05-28 08:21:53,200 - octoprint.startup - INFO - ******************************************************************************
2020-05-28 08:21:53,201 - octoprint.startup - INFO - Starting OctoPrint 1.4.0
2020-05-28 08:21:53,201 - octoprint.startup - INFO - ******************************************************************************`

jneilliii, 2020-05-28:
Ok you're probably going to have to try installing numpy outside of octoprint to see if there are any errors during the build. SSH to your pi and run the following command. 

```
~/oprint/bin/pip install --upgrade --no-cache-dir numpy 
```
And hopefully that works. If not we'll get a usable error to work with.

nick-harper, 2020-05-28:
Thanks, I will give it a go later. I forgot to open the port for SSH remotely.

jneilliii, 2020-05-28:
Also @nick-harper what version of Python are you running OctoPrint under. There was a related issue with Python 3 where numpy wasn't automatically installing its dependencies and therefore would fail. Relevant ticket on that issue can be found [here](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/issues/213).

nick-harper, 2020-05-28:
Trying to run it manually, it seems to just time out the pi 4.

Disconnects me from SSH.

nick-harper, 2020-05-28:
I think it is making the Pi reboot as it restarts the printer too.

nick-harper, 2020-05-28:
> Also @nick-harper what version of Python are you running OctoPrint under. There was a related issue with Python 3 where numpy wasn't automatically installing its dependencies and therefore would [...]

How would I tell which version?

nick-harper, 2020-05-28:
Sorry, after some testing I changed the power supply for the Pi and taped up the 5v connection to the printer and it worked fine.

It looked like it was the Pi rebooting under any sort of load which caused it.

Thanks for your time and help!

closed 2020-05-28 by nick-harper