jneilliii/OctoPrint-BedLevelVisualizer #258: cannot read property 'canvas' of undefined

sacy80, 2020-06-03:
[image]

hi i got this error. after updating Bed Visuailzer.

Bed Visualizer version: 0.1.13
clear data from plugin and reprobe got the same error. 

GCODE Commands for Mesh Updae Process:
G1 X20 Y20 Z20 F3000 ; Free Axis
G28      ; aome all axes
M155 S30 ; reduce temperature reporting rate to reduce output pollution
@BEDLEVELVISUALIZER	; tell the plugin to watch for reported mesh
G29 T	   ; run bilinear probing
M155 S3  ; reset temperature reporting

enable: Save Mesh
Processing Timeout: 3000sec

jneilliii, 2020-06-03:
What browser are you using and on what device?

sacy80, 2020-06-03:
Google Chrome version: 83.0.4103.61 <- windows 10 desktop
Octoprint version: 1.4.0
OS: armbian ubuntu bionic 4.19.62

jneilliii, 2020-06-04:
I've seen this error before when there was a video driver issue or your video card doesn't support openGL. If you go to the link below does the rotating box load for you?

https://get.webgl.org/

sacy80, 2020-06-04:
the box load and rotate . :)
[image]

jneilliii, 2020-06-05:
Ok, that's strange. Enable debug logging in the plugin's settings, restart OctoPrint and force a browser refresh (shift + F5), and try again when you get a chance. If it doesn't work then, please upload your `plugin_beflevelvisualizer_debug.log` file from the logging section of OctoPrint.

sacy80, 2020-06-05:
Hi .. guess it my desktop issue.. after the standard SOP of deskside support, if windows dont work.. reboot the computer.. And it works. O.o"

jneilliii, 2020-06-05:
lol...85% of the time, yes, that typically does resolve windows issues.

closed 2020-06-05 by jneilliii