jneilliii/OctoPrint-BedLevelVisualizer #262: Does this plugin support AUTO_BED_LEVELING_BILINEAR?

ralphhughes, 2020-06-07:
I see in your wiki there is example gcode for Marlin 1.1.9 with UBL, but none for bilinear, is it supported?

I can't put UBL on my TronXY P802MA as there is not enough progmem so I'm stuck with Marlin 1.1.9 with AUTO_BED_LEVELING_BILINEAR enabled. The output of G29 T on my machine looks like this:
`Recv: Bilinear Leveling Grid:`
`Recv:       0      1      2`
`Recv:  0 -0.210 -0.043 -0.055`
`Recv:  1 -0.290 -0.055 -0.020`
`Recv:  2 -0.313 -0.030 +0.080`
`Recv: `
`Recv: X:181.00 Y:203.00 Z:7.29 E:0.00 Count X:18100 Y:20300 Z:2992`
`Recv: ok`

When I try Bed Level Visualizer I just get "Bed Visualizer Error. Timeout occured before prcessing completed. Processing may still be running or there may be a configuration error."

ralphhughes, 2020-06-07:
My timeout seemed really high (1800s = 30min?) to I dropped it down to 3min, restarted octoprint and now it works :-)

After restarting octoprint and it started working, I saw this message in the debug log pretty much as soon as I clicked the button to update mesh "DEBUG: mesh collection started"

Before I restarted octoprint I was NOT getting that in the debug log, no matter how long I waited, even after 30 mins. Don't know if that's relevant.

Anyway, this can be closed as its working for me now.

jneilliii, 2020-06-07:
Yeah there's a weird issue when first installed that the wizard doesn't actually save right for some reason and it doesn't really work until octoprint restart or possibly just opening thr settings and saving. The next release will remove the wizard all together.

closed 2020-06-14 by jneilliii